pschatzmann/ESP32-A2DP #3: A2DP source using ADC(DMA) to sink

rujha, 2020-09-17:
thanks for the article/code  "Sending Sound from an ESP32 to a Bluetooth Sink".
I wanted to use your code to send data flow like below:

mobile-->microphone---ADC (DMA, using I2S ) source esp32 ----bluetooth speaker(sink)

So i defined the I2S function as below 
`{
    i2s_config_t i2s_config = {
      .mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_RX | I2S_MODE_ADC_BUILT_IN),
      .sample_rate = 44100,
      .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT,
      .channel_format = I2S_CHANNEL_FMT_ALL_LEFT,
      .communication_format =  I2S_COMM_FORMAT_I2S_MSB,
      .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1, 
      .dma_buf_count = 8, 
      .dma_buf_len = 1024,  
      .use_apll = false 
    };

 //install and start i2s driver
   i2s_driver_install(i2s_num, &i2s_config, 0, NULL);
   i2s_set_adc_mode(ADC_UNIT_1, ADC1_CHANNEL_0);
   analogReadResolution(12);
   adc1_config_width(ADC_WIDTH_BIT_12);
   adc1_config_channel_atten(ADC1_CHANNEL_0, ADC_ATTEN_DB_0);

   vTaskDelay(5000/portTICK_RATE_MS);
   i2s_adc_enable(I2S_NUM_0);
  
}
`
changed the get_data callback function as below:
`size_t bytes_read;
    int16_t audio[len/2];
    i2s_read(I2S_NUM_0, (void*) audio, len, &bytes_read, portMAX_DELAY); 

    int countt = 0;
    int16_t raw = 0;
    for(int i=0; i<len/2; i++){
        raw = (audio[i] & 0xfff)*16 ;  
        audio[i] = raw;
    }
    memcpy(data, audio, len);`

Getting data with noise. 
can you suggest some pointers or any other better approach for this.
thanks a lot.

pschatzmann, 2020-09-17:
Hallo,
I am not sure, I double checked the code from my Blog and indeed found an error: The data needs to be two-channel 16-bit sample data.
I am currently also trying to figure out how to use this functionality for some other scenarios… 

As far as your project is concearned, I would propose to split up the topic in two pieces that can be tested separatly
- the receiving 
- the sending (e.g. with prerecorded data)
so that you can better identify the problem.

Kind regards
Phil

rujha, 2020-09-17:
Hi Phil,
thanks for the info. Tested sink and source separately.
- the receiving is working fine
- sending is having some noise
will try to find out more on sending part.

Just want to confirm , have you disabled the source example code to disconnect/reconnect the sink looping.

thanks,
Rupak

pschatzmann, 2020-09-17:
I think the disconnect/reconnect the sink loop is not in the repository yet.
I am planning to push an new version tomorrow with corrections…

rujha, 2020-09-17:
disconnect/reconnect loop code can be found from this repo:

https://github.com/admiralmaggie/ESP32_BT_SRC_ADC/blob/master/main/main.c

Thanks a lot !!

pschatzmann, 2020-09-20:
I pushed some corrections to master. 
Hopefully this helps....

rujha, 2020-09-21:
Thanks for the code. I tried your new commit with my I2S example. Its not working might be due to "music_data_channels_cb_t callback".
 Same code works fine with your old commit.

pschatzmann, 2020-09-21:
Yes, you can will use the ‚old‘ logic by calling

    void startRaw(char* name, music_data_cb_t callback = NULL, bool is_ssp_enabled = false);

Kind regards
Phil

rujha, 2020-09-28:
Hi Phil,

Thanks for the code. Now source remains connected to the sink. 
one observation:
when Sink is disconnected, it is not re-connecting to source till I poweroff/on the sink device again. 

`20:02:43.539 -> [I][BluetoothA2DPSource.cpp:386] filter_inquiry_scan_result(): Device discovery found: X31
20:02:43.539 -> [I][BluetoothA2DPSource.cpp:390] filter_inquiry_scan_result(): Checking name: MyMusic
20:02:43.539 -> [I][BluetoothA2DPSource.cpp:390] filter_inquiry_scan_result(): Checking name: Philips PH805
20:02:43.573 -> [I][BluetoothA2DPSource.cpp:390] filter_inquiry_scan_result(): Checking name: MusicPlayer
20:02:43.573 -> [I][BluetoothA2DPSource.cpp:356] filter_inquiry_scan_result(): Scanned device: 41:42:fe:d1:99:90
20:02:43.573 -> [I][BluetoothA2DPSource.cpp:362] filter_inquiry_scan_result(): --Class of Device: 0x240418
20:02:43.573 -> [I][BluetoothA2DPSource.cpp:366] filter_inquiry_scan_result(): --RSSI: -68
20:02:43.607 -> [I][BluetoothA2DPSource.cpp:386] filter_inquiry_scan_result(): Device discovery found: X31
20:02:43.607 -> [I][BluetoothA2DPSource.cpp:390] filter_inquiry_scan_result(): Checking name: MyMusic
20:02:43.607 -> [I][BluetoothA2DPSource.cpp:390] filter_inquiry_scan_result(): Checking name: Philips PH805
20:02:43.607 -> [I][BluetoothA2DPSource.cpp:390] filter_inquiry_scan_result(): Checking name: MusicPlayer
20:02:44.807 -> [I][BluetoothA2DPSource.cpp:356] filter_inquiry_scan_result(): Scanned device: 41:42:fe:d1:99:90
20:02:44.807 -> [I][BluetoothA2DPSource.cpp:362] filter_inquiry_scan_result(): --Class of Device: 0x240418
20:02:44.807 -> [I][BluetoothA2DPSource.cpp:366] filter_inquiry_scan_result(): --RSSI: -67
`
thanks,
Rupak

closed 2020-12-03 by pschatzmann