pschatzmann/ESP32-A2DP #6: Metadata + more

Mael19, 2020-10-14:
Hello, first of all thank you so much for your library which is so simple and nice !

I'd like to know if it is possible to retrieve the metadata (ie. title and artist) from the bluetooth ?

Also, at least on my iPhone I usually get a "headphones" symbol in the status bar when I connect a bluetooth speaker. Here I got nothing, would it be possible to have it ? I guess I would have to tell the phone what type of bluetooth device is connected.

Finally, but maybe that's too much to ask, with some bluetooth devices, my iPhone is capable of showing the battery level. Since my project is battery powered and I already know the level, would it be possible to send that as well ?

Best regards,
Mael

JohnyMielony, 2020-12-08:
Simple answer is yes. And it's already implemented in this library.
Here (https://github.com/pschatzmann/ESP32-A2DP/blob/master/src/BluetoothA2DPSink.cpp#L397) you're telling what you want to get.
And here (https://github.com/pschatzmann/ESP32-A2DP/blob/master/src/BluetoothA2DPSink.cpp#L434) you're getting that.
More information about that you can be read in ESP32 documentation (https://docs.espressif.com/projects/esp-idf/en/v3.3/api-reference/bluetooth/esp_avrc.html).

Mael19, 2020-12-09:
Thank you so much for your answer! I went trough the code and it seems that everything I asked is indeed possible. Now I have to find the time and courage to go deeper in the code to understand how it works, it'a maybe a bit over my skills!
Cheers.

pschatzmann, 2020-12-12:
I  think it would be cool to have some callbacks to make it easy to use.

DrNeuroSurg, 2020-12-20:
> I think it would be cool to have some callbacks to make it easy to use.

Yes, it should. But I would also like a callback for connection state changes, so I could notify the user about this, e.g. let a LED blink or so ..

pschatzmann, 2021-03-25:
Support for a metadata callback has been added with the latest commit

closed 2021-03-25 by pschatzmann