pschatzmann/ESP32-A2DP #8: Getting SPDIF output

JohnyMielony, 2020-10-18:
Hi!
I'm trying to get audio from bluetooth and output it as SPDIF signal.
I found a library that uses I2S to do that.
My issue is that I need int16_t samples for each channel but I found in BluetoothA2DPSink.cpp file that function audio_data_callback gets uint8_t stream as input.
How to derive from it 16 bit samples for each stereo channels?

pschatzmann, 2020-10-18:
Hallo,

It is my understanding, that if you use the default i2s_config this is happening automatically.
As you can see it uses 16 bits I2S_CHANNEL_FMT_RIGHT_LEFT

  // setup default i2s config
  i2s_config = {
      .mode = (i2s_mode_t) (I2S_MODE_MASTER | I2S_MODE_TX),
      .sample_rate = 44100,
      .bits_per_sample = (i2s_bits_per_sample_t)16,
      .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,
      .communication_format = (i2s_comm_format_t) (I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_MSB),
      .intr_alloc_flags = 0, // default interrupt priority
      .dma_buf_count = 8,
      .dma_buf_len = 64,
      .use_apll = false
  };
Kind regards
Phil

JohnyMielony, 2020-10-18:
I2S itself uses 16 bits, but SPDIF is some kind of hack and samples for it needs some preparation before writing to it.
That's are settings for SPDIF, taken from that library:
```
i2s_config_t i2s_config = {
      .mode = (i2s_mode_t) (I2S_MODE_MASTER | I2S_MODE_TX),
      .sample_rate = 88200,
      .bits_per_sample = I2S_BITS_PER_SAMPLE_32BIT,
      .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,
      .communication_format = (i2s_comm_format_t)(I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_MSB),
      .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1,
      .dma_buf_count = 8,
      .dma_buf_len = 256,
      .use_apll = true
  };
```

And that is the function that is preparing sample and sends it to I2S:
```
bool AudioOutputSPDIF::ConsumeSample(int16_t sample[2])
{
  if (!i2sOn) return true; // Sink the data
  int16_t ms[2];
  uint16_t hi, lo, aux;
  uint32_t buf[4];

  ms[0] = sample[0];
  ms[1] = sample[1];
  MakeSampleStereo16(ms);

  // S/PDIF encoding: 
  //   http://www.hardwarebook.info/S/PDIF
  // Original sources: Teensy Audio Library 
  //   https://github.com/PaulStoffregen/Audio/blob/master/output_spdif2.cpp
  // 
  // Order of bits, before BMC encoding, from the definition of SPDIF format
  //   PPPP AAAA  SSSS SSSS  SSSS SSSS  SSSS VUCP
  // are sent rearanged as
  //   VUCP PPPP  AAAA 0000  SSSS SSSS  SSSS SSSS
  // This requires a bit less shifting as 16 sample bits align and can be 
  // BMC encoded with two table lookups (and at the same time flipped to LSB first).
  // There is no separate word-clock, so hopefully the receiver won't notice.

  uint16_t sample_left = Amplify(ms[LEFTCHANNEL]);
  // BMC encode and flip left channel bits
  hi = pgm_read_word(&spdif_bmclookup[(uint8_t)(sample_left >> 8)]);
  lo = pgm_read_word(&spdif_bmclookup[(uint8_t)sample_left]);
  // Low word is inverted depending on first bit of high word
  lo ^= (~((int16_t)hi) >> 16);
  buf[0] = ((uint32_t)lo << 16) | hi;
  // Fixed 4 bits auxillary-audio-databits, the first used as parity
  // Depending on first bit of low word, invert the bits
  aux = 0xb333 ^ (((uint32_t)((int16_t)lo)) >> 17);
  // Send 'B' preamble only for the first frame of data-block
  if (frame_num == 0) {
    buf[1] = VUCP_PREAMBLE_B | aux;
  } else {
    buf[1] = VUCP_PREAMBLE_M | aux;
  }

  uint16_t sample_right = Amplify(ms[RIGHTCHANNEL]); 
  // BMC encode right channel, similar as above
  hi = pgm_read_word(&spdif_bmclookup[(uint8_t)(sample_right >> 8)]);
  lo = pgm_read_word(&spdif_bmclookup[(uint8_t)sample_right]);
  lo ^= (~((int16_t)hi) >> 16);
  buf[2] = ((uint32_t)lo << 16) | hi;
  aux = 0xb333 ^ (((uint32_t)((int16_t)lo)) >> 17);
  buf[3] = VUCP_PREAMBLE_W | aux;

#if defined(ESP32)
  // Assume DMA buffers are multiples of 16 bytes. Either we write all bytes or none.
  uint32_t bytes_written;
  esp_err_t ret = i2s_write((i2s_port_t)portNo, (const char*)&buf, 8 * channels, &bytes_written, 0);
  // If we didn't write all bytes, return false early and do not increment frame_num
  if ((ret != ESP_OK) || (bytes_written != (8 * channels))) return false;  
#elif defined(ESP8266)
  if (!I2SDriver.writeInterleaved(buf)) return false;
#endif
  // Increment and rotate frame number
  if (++frame_num > 191) frame_num = 0;
  return true;
}
```

I'm wondering if I could just take four bytes from data stream and concatenate two for left, and other two for right channel.
Could it be just that simple?

pschatzmann, 2020-10-18:
If the endianness is matching it should work. 
Just give it a try and if it is not working I would try to switch bytes of the uint16_t
 ….

JohnyMielony, 2020-10-31:
So getting sample was that simple afterall.
`
while (len>0){
memcpy(&sample[0],data,2);
data += 2;
memcpy(&sample[1],data,2);
data += 2;
len -= 4;
output->ConsumeSample(sample);
}
`
It's crude but works, somewhat.
I commented out everything related to I2S in your library, then added SPDIF output derived directly from ESP8266Audio library, which has support for ESP32. Now SPDIF output is working, but not without issues. There is some jitter and sound is interrupted.
[https://youtu.be/O5iFoXNrhmQ](url)
Code na be found here: [https://github.com/JohnyMielony/ESP32-A2DP/tree/master/src](url).
I'm new to ESP32, could you point me where to look for issues, what is wrong?

pschatzmann, 2020-10-31:
My gut feeling is that the data is not provided quick enough:

- You can double check that the CPU Frequency is set to the fastest value int -> Tools -> CPU Frequency : 240 MHZ 
- I would try to replace the memcpy with simple assignments. I think the following should do the same thing:

void  BluetoothA2DPSink::audio_data_callback(const uint8_t *data, uint32_t len) {
   	uint16_t *data16 = (uint16_t *) data
	for (int j=0;j<len/2;j+=2) {
		sample[0] = data16[j];
		sample[1] = data16[j+1];
  	   	output->ConsumeSample(sample);
	}
}

JohnyMielony, 2020-10-31:
Yes, it's 240 MHz, I even put a log message with cpu speed to be sure.
It's maybe slightly better with that for loop.
I noticed that when I even disconnect bluetooth and nothing is playes that 'Din unlock' message still appears.

pschatzmann, 2020-10-31:
I had a quick look at your output->ConsumeSample(sample) method and noticed that you do a i2s_write operation.  
I would recommend to change the  ConsumeSample method so that it can process the whole array of data in one shot, 
so that you just have 1 i2s_write for each audio_data_callback call.

pschatzmann, 2020-11-01:
I double checked your i2s config:
- You could also try to increase the dma_buf_count or dma_buf_len.
- Are you sure that your sample rate of 88200  is correct - usually for Audio CD Quality this is 44100 ? 

I also doubt that you can process that many samples per second…

JohnyMielony, 2020-11-01:
I was changing dma buffers, that values in code I found most optimal so far.
I think that double sample rate is because there are additional bits (VUCP PPPP  AAAA 0000  SSSS SSSS  SSSS SSSS) where 'S' is actual sound bits.
I will look into that i2w_write and will try to buffer some frames before sending it to i2s.
Maybe there is some way to use second core to speed up whole process?

pschatzmann, 2020-11-01:
Try to keep things simple and efficient: I would recommend to consider to use the second core only
- if you have some major other (non music related) processing 
- if you do not use wifi or bluetooth

JohnyMielony, 2020-11-01:
I putted a log message into i2s write error.
if ((ret != ESP_OK) || (bytes_written != (8 * channels))){
    ESP_LOGE(BT_AV_TAG,"I2S Write error. Bytes written: %d, frame: %d", bytes_written, frame_num);
    return false;
}

And that's the result:
>[E][SPDIFOut.cpp:201] ConsumeSample(): I2S Write error. Bytes written: 0, frame: 0
[E][SPDIFOut.cpp:201] ConsumeSample(): I2S Write error. Bytes written: 0, frame: 0
[E][SPDIFOut.cpp:201] ConsumeSample(): I2S Write error. Bytes written: 0, frame: 64
[E][SPDIFOut.cpp:201] ConsumeSample(): I2S Write error. Bytes written: 0, frame: 64
[E][SPDIFOut.cpp:201] ConsumeSample(): I2S Write error. Bytes written: 0, frame: 128
[E][SPDIFOut.cpp:201] ConsumeSample(): I2S Write error. Bytes written: 0, frame: 128
[E][SPDIFOut.cpp:201] ConsumeSample(): I2S Write error. Bytes written: 0, frame: 0
[E][SPDIFOut.cpp:201] ConsumeSample(): I2S Write error. Bytes written: 0, frame: 64
[E][SPDIFOut.cpp:201] ConsumeSample(): I2S Write error. Bytes written: 0, frame: 64
[E][SPDIFOut.cpp:201] ConsumeSample(): I2S Write error. Bytes written: 0, frame: 128
[E][SPDIFOut.cpp:201] ConsumeSample(): I2S Write error. Bytes written: 0, frame: 128
[E][SPDIFOut.cpp:201] ConsumeSample(): I2S Write error. Bytes written: 0, frame: 0
[E][SPDIFOut.cpp:201] ConsumeSample(): I2S Write error. Bytes written: 0, frame: 0
[E][SPDIFOut.cpp:201] ConsumeSample(): I2S Write error. Bytes written: 0, frame: 0
[E][SPDIFOut.cpp:201] ConsumeSample(): I2S Write error. Bytes written: 0, frame: 128
[E][SPDIFOut.cpp:201] ConsumeSample(): I2S Write error. Bytes written: 0, frame: 128
[E][SPDIFOut.cpp:201] ConsumeSample(): I2S Write error. Bytes written: 0, frame: 0
[E][SPDIFOut.cpp:201] ConsumeSample(): I2S Write error. Bytes written: 0, frame: 0

Any clue?

pschatzmann, 2020-11-01:
I think you found the major issue. If the buffer is full, you should wait for the buffer to be available again

Try to change

  esp_err_t ret = i2s_write((i2s_port_t)portNo, (const char*)&buf, 8 * channels, &bytes_written, 0);

to

  esp_err_t ret = i2s_write((i2s_port_t)portNo, (const char*)&buf, 8 * channels, &bytes_written, portMAX_DELAY);

JohnyMielony, 2020-11-01:
This helped with that, after change there is no single write error.
But the interruption still occurs. Little less, but still.
I was playing with code and found something weird/curious. When I disable PLL I can hear some jitter but the interruptions occurs much less, and there is no 'Din unlock' (that is showing when signal is interrupted) on my md/dac when sound is paused. Which was happening on paused sound with PLL set on.
I'm starting to think that it won't be stable and reliable, and I should give it up.

pschatzmann, 2020-11-01:
I am sorry that it’s still not working. Maybe one last thought: you could try to skip every other entry in the array and set the sample_rate to half of what it currently is.
If that will stop the interruptions it would be an indication that the current implementation is just too slow to process the currently provided amount of data.
Just out of curiosity: what kind of DAC are you using exactly ?

JohnyMielony, 2020-11-01:
> set the sample_rate to half of what it currently is

Allready tried this. No sound at all.

I'm using Sony Minidisc Deck MDS-JE510 in a role of DAC. With other devices it works with no issues.
You think that it could be a hardware issue, rather than software?
I'm using the simplest ttl to spdif coax interface found on [this](https://www.epanorama.net/documents/audio/spdif.html) site and some scrap rca audio cable rather than coax 75 Ohm. So I don't reject that possibility.

pschatzmann, 2020-11-01:
If you have some spare time you could try to use another library which provides proper SPIF support to generate music: e.g https://github.com/earlephilhower/ESP8266Audio.
If this library gives good quality it is not a hardware problem.
I am also using this library to stream internet radio...

JohnyMielony, 2020-11-01:
I took code from that library, as I mentioned yesterday. The reason I did that is it didn't compile when I tried to match it with yours library.
I used example from ESP8266Audio library to test if SPDIF works at all. Played mp3 file ok. Maybe that file was too short to notice any issue. File size was limited by a spiffs size.

pschatzmann, 2020-11-01:
I have an example which streams the music from an url:
https://www.pschatzmann.ch/home/2020/03/19/sound-from-a-esp32-microcontroller/

it uses the http://strm112.1.fm/blues_mobile_mp3 as input...

pschatzmann, 2020-11-01:
I tried to go thru the original AudioOutputSPDIF logic but I am afraid that I am giving it up as well:
The following just does not make any sense to me
	- values for uint32_t buf[4] are calculated -> 4 * 32 = 128 bytes
	- i2s_write() only writes 8 * channels -> 8 * 2 = 16 bytes

JohnyMielony, 2020-11-02:
I tried your sketch, and same thing, interruptions. I even added that portMAX_DELAY and changed dma buffer to bigger size in original library.
So the last thing I will try is to add a buffer chip. But this will take me some time because I couldn't get it at local shops, so I need to buy it online.

JohnyMielony, 2020-11-04:
I found interruptions reason. It was hardware issue in fact.
That TTL to coax SPDIF interface was calculated for a 5V level. I recalculated voltage divider for a 3,3V level.
I should have been think about it at the beggining. Mea culpa. Now it works well.
ESP32 freezes or restarts from time to time, but it's probably power issue.
Thanks for your help.
BTW You think there is a way to attach I2S microphone and have a two way communication?

JohnyMielony, 2020-12-08:
Debugging showed that occasional restarts was memory problem. After changing I2S DMA buffer to lower values it's working fine.
So I'm closing this issue while I find it solved.

closed 2020-12-08 by JohnyMielony

pschatzmann, 2023-03-30:
https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-basic-api/basic-a2dp-spdif/basic-a2dp-spdif.ino