pschatzmann/ESP32-A2DP #10: Watch Dog timer resets everytime

craiglindley, 2020-10-28:
The example for bt_music_sender seems to work with my ESP32 TTGO T-Display module. However if I modify the example even slightly, the tones break up and eventually the ESP32 crashes with a watch dog timer issue. In this case, I wanted to generate different frequency tones for each ear.

#define FREQ1  120
#define FREQ2  500

BluetoothA2DPSource a2dp_source;

// The supported audio codec in ESP32 A2DP is SBC. SBC audio stream is encoded
// from PCM data normally formatted as 44.1kHz sampling rate, two-channel 16-bit sample data
int32_t get_data_channels(Channels *channels, int32_t channel_len) {
    static double m_time = 0.0;
    double m_amplitude = 8000.0;  // -32,768 to 32,767
    double m_deltaTime = 1.0 / 44100.0;
    double m_phase = 0.0;
    double double_Pi = PI * 2.0;
    // fill the channel data
    for (int sample = 0; sample < channel_len; ++sample) {
        double angle  = double_Pi * FREQ1 * m_time + m_phase;
        double angle2 = double_Pi * FREQ2 * m_time + m_phase;
        channels[sample].channel1 = m_amplitude * sin(angle);
        channels[sample].channel2 = m_amplitude * sin(angle2);
        m_time += m_deltaTime;
    }

    return channel_len;
}

void setup() {
  a2dp_source.start("BT One", get_data_channels);  
}`

This crashes everytime

ASSERT_PARAM(1 0), in lc_task.c at line 2984
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x400877b2  PS      : 0x00060234  A0      : 0x8002dbb4  A1      : 0x3ffcaa40  
A2      : 0x00000001  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x60008054  
A6      : 0x3ffbdd44  A7      : 0x60008050  A8      : 0x800877ad  A9      : 0x3ffcaa20  
A10     : 0x00000004  A11     : 0x00000000  A12     : 0x6000804c  A13     : 0xffffffff  
A14     : 0x00000000  A15     : 0xfffffffb  SAR     : 0x00000004  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x400876e5  LEND    : 0x400876ec  LCOUNT  : 0x00000000  

Backtrace: 0x400877b2:0x3ffcaa40 0x4002dbb1:0x3ffcaa60 0x40024f79:0x3ffcaac0 0x4001a637:0x3ffcaae0 0x40019d11:0x3ffcab10 0x40055b4d:0x3ffcab30 0x40163917:0x3ffcab50 0x40163e6d:0x3ffcab70 0x4008dbbd:0x3ffcaba0

Any ideas on how to solve this?

craiglindley, 2020-10-28:
This is another crash that I see with the code above

`E (77440) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (77440) task_wdt:  - IDLE0 (CPU 0)
E (77440) task_wdt: Tasks currently running:
E (77440) task_wdt: CPU 0: BtA2dSourceT
E (77440) task_wdt: CPU 1: loopTask
E (77440) task_wdt: Aborting.
abort() was called at PC 0x400d48ef on core 0

B`

pschatzmann, 2020-10-29:
My gut feeling is, that the sin() function might be too slow to generate the necessary audo data. 
I propose that you try to precalculate your sin() values in setup() into an array and just copy
these values in the get_data_channels() method.

pschatzmann, 2021-10-16:
closed due to inactivity...

closed 2021-10-16 by pschatzmann