pschatzmann/ESP32-A2DP #9: A2dp source disconnects bluetooth connection after some seconds

HVaArd, 2020-10-28:
I'm using Ultimate Ears WONDERBOOM with a2dp source. My plan is to prepare the WonderBoom (turn power on and press the button on top of it to pair) then wait for ESP32 to wake-up from deep sleep. The ESP32 wakes up when a timer expires and a2dp source finds the WonderBoom fine, connects and the music starts playing. But after some seconds well closer to 30 the bluetooth connection drops. I suppose it is somehow related to pairing or a2dp source finding the WonderBoom "too early".
If I wait for the ESP32 to wake up and then turn on the WonderBoom and press the pairing button the connection stays on without problems.

pschatzmann, 2020-10-28:
I propose to set the Debug level to ‘Verbose' and check the console. 
That might give you a hint why it is disconnecting...

HVaArd, 2020-10-28:
Hi

Thanks for quick reply.

Here is what was printed before the disconnect. The time stamp print (Connection to BT lost at: 38921
) comes from my code and that is the time when the disconnection happened.

[D][BluetoothA2DPSource.cpp:221] bt_app_work_dispatch(): bt_app_work_dispatch event 0xff00, param len 0
[D][BluetoothA2DPSource.cpp:271] bt_app_task_handler(): bt_app_task_handler, sig 0x1, 0xff00
[I][BluetoothA2DPSource.cpp:550] bt_app_av_sm_hdlr(): bt_app_av_sm_hdlr state 5, evt 0xff00
[D][BluetoothA2DPSource.cpp:221] bt_app_work_dispatch(): bt_app_work_dispatch event 0xff00, param len 0
[D][BluetoothA2DPSource.cpp:271] bt_app_task_handler(): bt_app_task_handler, sig 0x1, 0xff00
[I][BluetoothA2DPSource.cpp:550] bt_app_av_sm_hdlr(): bt_app_av_sm_hdlr state 5, evt 0xff00
[D][BluetoothA2DPSource.cpp:221] bt_app_work_dispatch(): bt_app_work_dispatch event 0x1, param len 16
[D][BluetoothA2DPSource.cpp:271] bt_app_task_handler(): bt_app_task_handler, sig 0x1, 0x1
[I][BluetoothA2DPSource.cpp:550] bt_app_av_sm_hdlr(): bt_app_av_sm_hdlr state 5, evt 0x1
[D][BluetoothA2DPSource.cpp:221] bt_app_work_dispatch(): bt_app_work_dispatch event 0x0, param len 16
[D][BluetoothA2DPSource.cpp:271] bt_app_task_handler(): bt_app_task_handler, sig 0x1, 0x0
[I][BluetoothA2DPSource.cpp:550] bt_app_av_sm_hdlr(): bt_app_av_sm_hdlr state 5, evt 0x0
[I][BluetoothA2DPSource.cpp:703] bt_app_av_state_connected(): a2dp disconnected
Connection to BT lost at: 38921
[D][BluetoothA2DPSource.cpp:221] bt_app_work_dispatch(): bt_app_work_dispatch event 0x0, param len 12
[D][BluetoothA2DPSource.cpp:271] bt_app_task_handler(): bt_app_task_handler, sig 0x1, 0x0
[D][BluetoothA2DPSource.cpp:799] bt_av_hdl_avrc_ct_evt(): bt_av_hdl_avrc_ct_evt evt 0
[I][BluetoothA2DPSource.cpp:805] bt_av_hdl_avrc_ct_evt(): AVRC conn_state evt: state 0, [ec:81:93:5f:74:76]

HVaArd, 2020-10-28:
Hi

Here is what it looks like when I wait for the ESP32 to wake up first and the do the connection. Here the print “Initial sync was found” comes from my code saying that audio is now available for playing.

[D][BluetoothA2DPSource.cpp:221] bt_app_work_dispatch(): bt_app_work_dispatch event 0xff00, param len 0
[D][BluetoothA2DPSource.cpp:271] bt_app_task_handler(): bt_app_task_handler, sig 0x1, 0xff00
[I][BluetoothA2DPSource.cpp:550] bt_app_av_sm_hdlr(): bt_app_av_sm_hdlr state 5, evt 0xff00
[I][BluetoothA2DPSource.cpp:636] bt_app_av_media_proc(): a2dp media ready checking ...
[D][BluetoothA2DPSource.cpp:221] bt_app_work_dispatch(): bt_app_work_dispatch event 0x3, param len 16
[D][BluetoothA2DPSource.cpp:271] bt_app_task_handler(): bt_app_task_handler, sig 0x1, 0x3
[I][BluetoothA2DPSource.cpp:550] bt_app_av_sm_hdlr(): bt_app_av_sm_hdlr state 5, evt 0x3
[I][BluetoothA2DPSource.cpp:642] bt_app_av_media_proc(): a2dp media ready, starting ...
[D][BluetoothA2DPSource.cpp:221] bt_app_work_dispatch(): bt_app_work_dispatch event 0x3, param len 16
[D][BluetoothA2DPSource.cpp:271] bt_app_task_handler(): bt_app_task_handler, sig 0x1, 0x3
[I][BluetoothA2DPSource.cpp:550] bt_app_av_sm_hdlr(): bt_app_av_sm_hdlr state 5, evt 0x3
[I][BluetoothA2DPSource.cpp:654] bt_app_av_media_proc(): a2dp media start successfully.
[D][BluetoothA2DPSource.cpp:221] bt_app_work_dispatch(): bt_app_work_dispatch event 0x1, param len 16
[D][BluetoothA2DPSource.cpp:271] bt_app_task_handler(): bt_app_task_handler, sig 0x1, 0x1
[I][BluetoothA2DPSource.cpp:550] bt_app_av_sm_hdlr(): bt_app_av_sm_hdlr state 5, evt 0x1
[D][BluetoothA2DPSource.cpp:221] bt_app_work_dispatch(): bt_app_work_dispatch event 0x0, param len 12
[D][BluetoothA2DPSource.cpp:271] bt_app_task_handler(): bt_app_task_handler, sig 0x1, 0x0
[D][BluetoothA2DPSource.cpp:799] bt_av_hdl_avrc_ct_evt(): bt_av_hdl_avrc_ct_evt evt 0
[I][BluetoothA2DPSource.cpp:805] bt_av_hdl_avrc_ct_evt(): AVRC conn_state evt: state 1, [ec:81:93:5f:74:76]
[D][BluetoothA2DPSource.cpp:221] bt_app_work_dispatch(): bt_app_work_dispatch event 0x5, param len 12
[D][BluetoothA2DPSource.cpp:271] bt_app_task_handler(): bt_app_task_handler, sig 0x1, 0x5
[D][BluetoothA2DPSource.cpp:799] bt_av_hdl_avrc_ct_evt(): bt_av_hdl_avrc_ct_evt evt 5
[I][BluetoothA2DPSource.cpp:832] bt_av_hdl_avrc_ct_evt(): AVRC remote features
Initial sync was found
[D][BluetoothA2DPSource.cpp:221] bt_app_work_dispatch(): bt_app_work_dispatch event 0xff00, param len 0
[D][BluetoothA2DPSource.cpp:271] bt_app_task_handler(): bt_app_task_handler, sig 0x1, 0xff00
[I][BluetoothA2DPSource.cpp:550] bt_app_av_sm_hdlr(): bt_app_av_sm_hdlr state 5, evt 0xff00
[D][BluetoothA2DPSource.cpp:221] bt_app_work_dispatch(): bt_app_work_dispatch event 0xff00, param len 0
[D][BluetoothA2DPSource.cpp:271] bt_app_task_handler(): bt_app_task_handler, sig 0x1, 0xff00
[I][BluetoothA2DPSource.cpp:550] bt_app_av_sm_hdlr(): bt_app_av_sm_hdlr state 5, evt 0xff00
[D][BluetoothA2DPSource.cpp:221] bt_app_work_dispatch(): bt_app_work_dispatch event 0xff00, param len 0
[D][BluetoothA2DPSource.cpp:271] bt_app_task_handler(): bt_app_task_handler, sig 0x1, 0xff00
[I][BluetoothA2DPSource.cpp:550] bt_app_av_sm_hdlr(): bt_app_av_sm_hdlr state 5, evt 0xff00
[D][BluetoothA2DPSource.cpp:221] bt_app_work_dispatch(): bt_app_work_dispatch event 0xff00, param len 0
[D][BluetoothA2DPSource.cpp:271] bt_app_task_handler(): bt_app_task_handler, sig 0x1, 0xff00
[I][BluetoothA2DPSource.cpp:550] bt_app_av_sm_hdlr(): bt_app_av_sm_hdlr state 5, evt 0xff00
[D][BluetoothA2DPSource.cpp:221] bt_app_work_dispatch(): bt_app_work_dispatch event 0xff00, param len 0
[D][BluetoothA2DPSource.cpp:271] bt_app_task_handler(): bt_app_task_handler, sig 0x1, 0xff00
[I][BluetoothA2DPSource.cpp:550] bt_app_av_sm_hdlr(): bt_app_av_sm_hdlr state 5, evt 0xff00

pschatzmann, 2020-10-28:
As far as I can see the following is happening:

The method 

void BluetoothA2DPSource::bt_app_av_state_connected(uint16_t event, void *param)
{
    esp_a2d_cb_param_t *a2d = NULL;
    switch (event) {
    case ESP_A2D_CONNECTION_STATE_EVT: {
        a2d = (esp_a2d_cb_param_t *)(param);
        if (a2d->conn_stat.state == ESP_A2D_CONNECTION_STATE_DISCONNECTED) {
            ESP_LOGI(BT_AV_TAG, "a2dp disconnected");
            s_a2d_state = APP_AV_STATE_UNCONNECTED;
            //esp_bt_gap_set_scan_mode(ESP_BT_CONNECTABLE, ESP_BT_GENERAL_DISCOVERABLE);
            esp_bt_gap_set_scan_mode(ESP_BT_SCAN_MODE_CONNECTABLE_DISCOVERABLE);

        }
        break;
    }

Is receiving a connected event with the state Disconnected. 

You could try to overwrite this method and comment out this logic. 
But it is likely that this will not help because I think the WONDERBOOM already stopped the connection.

Kind regards
Phil

HVaArd, 2020-10-28:
Hi

Infact I did try that with your older version of a2dp. And yes that did not help.

My project is only for fun, so not that serious. But I sure would have wanted the music to start playing by only turning on the speaker at any given time. And everything else would be automatic. And the ESP32 would be sleeping most of the time and power would be saved.

Br. Hannu

HVaArd, 2020-10-29:
Hi Phil

Sorry to bother you but I did notice that if the red action below happens before media is started the connection drops. If the red action happens after the media the connection works.

Maybe you can give a quick advice.

I’m using the library as is, I have not modified it at all.

Br. HVa

[D][BluetoothA2DPSource.cpp:152] startRaw(): x3f402348,
[D][BluetoothA2DPSource.cpp:221] bt_app_work_dispatch(): bt_app_work_dispatch event 0x0, param len 0
[D][BluetoothA2DPSource.cpp:271] bt_app_task_handler(): bt_app_task_handler, sig 0x1, 0x0
[D][BluetoothA2DPSource.cpp:490] bt_av_hdl_stack_evt(): bt_av_hdl_stack_evt evt 0
[I][BluetoothA2DPSource.cpp:518] bt_av_hdl_stack_evt(): Starting device discovery...
ed

[I][BluetoothA2DPSource.cpp:433] bt_app_gap_cb(): Discovery started.
[I][BluetoothA2DPSource.cpp:361] filter_inquiry_scan_result(): Scanned device: ec:81:93:5f:74:76
[I][BluetoothA2DPSource.cpp:367] filter_inquiry_scan_result(): --Class of Device: 0x240404
[I][BluetoothA2DPSource.cpp:371] filter_inquiry_scan_result(): --RSSI: -54
[I][BluetoothA2DPSource.cpp:391] filter_inquiry_scan_result(): Device discovery found: WONDERBOOM
[I][BluetoothA2DPSource.cpp:395] filter_inquiry_scan_result(): Checking name: WONDERBOOM
[I][BluetoothA2DPSource.cpp:403] filter_inquiry_scan_result(): Found a target device, address ec:81:93:5f:74:76, name WONDERBOOM
[I][BluetoothA2DPSource.cpp:406] filter_inquiry_scan_result(): Cancel device discovery ...
[I][BluetoothA2DPSource.cpp:424] bt_app_gap_cb(): Device discovery stopped.
[I][BluetoothA2DPSource.cpp:425] bt_app_gap_cb(): a2dp connecting to peer: WONDERBOOM
[D][BluetoothA2DPSource.cpp:221] bt_app_work_dispatch(): bt_app_work_dispatch event 0x0, param len 16
[D][BluetoothA2DPSource.cpp:271] bt_app_task_handler(): bt_app_task_handler, sig 0x1, 0x0
[I][BluetoothA2DPSource.cpp:550] bt_app_av_sm_hdlr(): bt_app_av_sm_hdlr state 4, evt 0x0
[D][BluetoothA2DPSource.cpp:221] bt_app_work_dispatch(): bt_app_work_dispatch event 0x0, param len 16
[D][BluetoothA2DPSource.cpp:271] bt_app_task_handler(): bt_app_task_handler, sig 0x1, 0x0
[I][BluetoothA2DPSource.cpp:550] bt_app_av_sm_hdlr(): bt_app_av_sm_hdlr state 4, evt 0x0
[I][BluetoothA2DPSource.cpp:603] bt_app_av_state_connecting(): a2dp connected

[D][BluetoothA2DPSource.cpp:221] bt_app_work_dispatch(): bt_app_work_dispatch event 0x0, param len 12
[D][BluetoothA2DPSource.cpp:271] bt_app_task_handler(): bt_app_task_handler, sig 0x1, 0x0
[D][BluetoothA2DPSource.cpp:799] bt_av_hdl_avrc_ct_evt(): bt_av_hdl_avrc_ct_evt evt 0
[I][BluetoothA2DPSource.cpp:805] bt_av_hdl_avrc_ct_evt(): AVRC conn_state evt: state 1, [ec:81:93:5f:74:76]
[D][BluetoothA2DPSource.cpp:221] bt_app_work_dispatch(): bt_app_work_dispatch event 0x5, param len 12
[D][BluetoothA2DPSource.cpp:271] bt_app_task_handler(): bt_app_task_handler, sig 0x1, 0x5
[D][BluetoothA2DPSource.cpp:799] bt_av_hdl_avrc_ct_evt(): bt_av_hdl_avrc_ct_evt evt 5
[I][BluetoothA2DPSource.cpp:832] bt_av_hdl_avrc_ct_evt(): AVRC remote features

[D][BluetoothA2DPSource.cpp:221] bt_app_work_dispatch(): bt_app_work_dispatch event 0xff00, param len 0
[D][BluetoothA2DPSource.cpp:271] bt_app_task_handler(): bt_app_task_handler, sig 0x1, 0xff00
[I][BluetoothA2DPSource.cpp:550] bt_app_av_sm_hdlr(): bt_app_av_sm_hdlr state 5, evt 0xff00
[I][BluetoothA2DPSource.cpp:636] bt_app_av_media_proc(): a2dp media ready checking ...
[D][BluetoothA2DPSource.cpp:221] bt_app_work_dispatch(): bt_app_work_dispatch event 0x3, param len 16
[D][BluetoothA2DPSource.cpp:271] bt_app_task_handler(): bt_app_task_handler, sig 0x1, 0x3
[I][BluetoothA2DPSource.cpp:550] bt_app_av_sm_hdlr(): bt_app_av_sm_hdlr state 5, evt 0x3
[I][BluetoothA2DPSource.cpp:642] bt_app_av_media_proc(): a2dp media ready, starting ...
[D][BluetoothA2DPSource.cpp:221] bt_app_work_dispatch(): bt_app_work_dispatch event 0x3, param len 16
[D][BluetoothA2DPSource.cpp:271] bt_app_task_handler(): bt_app_task_handler, sig 0x1, 0x3
[I][BluetoothA2DPSource.cpp:550] bt_app_av_sm_hdlr(): bt_app_av_sm_hdlr state 5, evt 0x3
[I][BluetoothA2DPSource.cpp:654] bt_app_av_media_proc(): a2dp media start successfully.
[D][BluetoothA2DPSource.cpp:221] bt_app_work_dispatch(): bt_app_work_dispatch event 0x1, param len 16
[D][BluetoothA2DPSource.cpp:271] bt_app_task_handler(): bt_app_task_handler, sig 0x1, 0x1
[I][BluetoothA2DPSource.cpp:550] bt_app_av_sm_hdlr(): bt_app_av_sm_hdlr state 5, evt 0x1

pschatzmann, 2020-10-29:
Maybe it might help if you add some delays() in your overall logic ?
You could also check if the start() was successful by checking with isConnected() and if not retry again ?

HVaArd, 2020-10-29:
Hi

Thanks. I will try for sure.

Br. HVa

HVaArd, 2020-10-29:
Hi Phil

I got it working, but have not been testing it yet thoroughly.

I did not add delay, I made things faster.

I changed the heart beat event timer from 10 seconds to 2 and now procedures happen in right order. I suppose I need to check what would be the best value for my application. Maybe something between 2 and ...

Thanks for your help.

Br. HVa

closed 2020-10-30 by HVaArd