pschatzmann/ESP32-A2DP #11: Adapting your code to ESP8266Audio sound library

Martin-Laclaustra, 2020-11-01:
In spite of its name, @earlephilhower library supports ESP32, and there have been request for A2DP sources and sinks.
https://github.com/earlephilhower/ESP8266Audio
The structure of the library is very similar to your work, so it would be easily integrated.
The advantage would be to use all the sources and sinks already implemented there (for example SPDIF output is available).
Regards.

pschatzmann, 2020-11-01:
I also love the ESP8266Audio library and use it for my ESP32 music projects as well, but so far I did not have any reason to use one as input and the other as output.
I am currently actifily using the following scenarios
 - a) ESP32 Internet Radio using the ESP8266Audio with output to I2S DAC
 - b) ESP32-A2DP Library to stream music (Spotify) from my IPAD to the I2S DAC

I am very happy with the results from b) where I always get consistently good quality music. 
a) however is quite tricky because it depends on many variables like network speed and sampling rates of the music stream. 

I would see potentially 2 scenarios for your request:

a) Generation of music with ESP8266Audio and send it to Bluetooth
b) Using Bluetooth as Data Source for ESP8266Audio

It is my gut feeling that b) will technically not work because there is just not enough memory and processing power in the ESP32 to convert the callback function which automaticlly provides data into a „File Based Interface“ and for
a) I am also sceptical because of my bad exerience of the inconsistent quaility. 

But I challange you to proof me wrong.

What kind of scenarios did you have in your mind ?

Martin-Laclaustra, 2020-11-01:
Thank you for your quick response.
Let me start answering your question:
- Main scenario: to broaden the flexibility for developing and testing any hardware that uses ESP with audio. E.g. when lacking a 4-8 ohms speaker or amplifier being able to drop in a BT speaker instead... (simplifying all electronics to an all-in-software project).
- The library allows decoding a wide variety (for microcontrollers) of files from SD (mp3, flac, opus, MIDI), so a player with BT output is also a possibility. Furthermore, SAM TTS system could also be connected to bluetooth. Also, may be creating a MIDI instrument that outputs to BT...
- Your application of a BT speaker would benefit from the variety of output options available (including SPDIF) without writing new code (i.e. external I2S amplifier, or DAC,...). A BT adapter for older audio stereo systems is another thing that comes to mind.
- The library will possibly support I2S microphones... mic to BT Speaker broadcaster?

Now, about your reflections:
- I do not think that this microcontroller is an optimal solution for internet radio, but local audio (from SD) reproduces OK.
- In your second application (ESP32-based BT speaker for Spotify) you did the effort to write the bridging code to I2S-DAC... This could be done directly in ESP8266Audio (with your integrated source) for other projects. The library has a mixer, filters...

You described perfectly both scenarios of the "request" (I would prefer to call it suggestion).
I do not think that adapting the code would be so difficult as you picture it. Audio source just needs a buffer, but you do not need to hold many data in memory, just enough to pass it in "real time" to a sink. With respect to audio output, it has a "consume sample" function that is called to pass samples (or blocks) to the rendering device.
However, may be we should hear the opinion of @earlephilhower who can gauge the difficulty and also provide good guidance on how to proceed, should you agree to it. Also, may be he/his collaborators could adapt the code with your guidance, should you gave permission.

Thank you again for your attention.

earlephilhower, 2020-11-02:
This is a neat library and seems to hide a lot of the complexity of the BT stack from an app just trying to receive or play data.

After looking at your sender example, I think this could actually easily be used with a small shim to work with the ESP8266Audio library as an AudioOutput (basically an audio sink).  Conceptually it looks like you have a callback which supplies new PCM data for the BT stack to send to the remote speaker.  That data could easily come from the MP3 or AAC decoder (which receives a callback saying "here's XXX samples from the codec, go and play it somehow) with some housekeeping (i.e. store it away when received from the codec and send whatever is available on the BT callback).   The devil is in the details, so things like the sample rate, bit depth, and sample format might need adjustment, of course.

My first impression is that this library doesn't really need any specific changes and a simple ESP8266Audio::AudioOutput wrapper class could be done by an interested party.

Going the other way (being a BT sink) might be more involved and I'm not really sure the utility there.  The samples have already come in as PCM, and you have code that pumps it out over I2S so that's pretty much a solved problem for you and the ESP8266Audio library doesn't have much to add.

HX2003, 2020-11-08:
I've done some work on the A2DPSink [here](https://github.com/HX2003/ESP8266Audio) out of interest. Instead of a file source, I made it to be a generator. My use case is where I use a mixer, to play wav files after boot and on connection change. However, it does suffer from occasions underflows especially at the start.

panfeng-zhang, 2020-12-10:
> I've done some work on the A2DPSink [here](https://github.com/HX2003/ESP8266Audio) out of interest. Instead of a file source, I made it to be a generator for simplicity. My use case is where I use a mixer, to play wav files after boot and on connection change. However, it does suffer from occasions underflows especially at the start.

here have a demo:
https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/classic_bt/a2dp_source

panfeng-zhang, 2020-12-10:
> I've done some work on the A2DPSink [here](https://github.com/HX2003/ESP8266Audio) out of interest. Instead of a file source, I made it to be a generator for simplicity. My use case is where I use a [...]

Hi HX2003,
I have use your code,the code can't compile.

Arduino:1.8.14 Hourly Build 2020/09/23 10:33 (Windows 10), ："ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 16MB (128Mb), 921600, None"

In file included from C:\Users\Desktop\ESP8266Audio-master\examples\PlayA2DPANDMP3FromSPIFFS\PlayA2DPANDMP3FromSPIFFS.ino:17:0:

D:\UserBakcup\Arduino\arduino-nightly-windows1.8.14\libraries\ESP8266Audio\src/AudioGeneratorA2DP.h:107:50: error: 'esp_avrc_rn_param_t' has not been declared

     void av_notify_evt_handler(uint8_t event_id, esp_avrc_rn_param_t event_parameter);

HX2003, 2020-12-10:
@panfeng-zhang I think that's because I am using Arduino with IDF Release >= 4.0.

panfeng-zhang, 2020-12-14:
@HX2003 I have downloaded the arduino-esp32-idf-release-v4.0 , But there are other compilation errors.
I want to know, what IDE do you use to compile?ESP-IDF32 or arduino、VScode + platformio+ESP-IDF32.

Thank you very much!

pschatzmann, 2021-04-29:
Here is another example that shows how to make both libraries work together:

https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/file_mp3-a2dp

Martin-Laclaustra, 2021-04-30:
This looks very interesting!
Actually integration as an audio sink is a small step away.
I am unfortunately quite busy these days, but I keep this in mind.
@earlephilhower

closed 2021-11-09 by pschatzmann