pschatzmann/ESP32-A2DP #17: Bluetooth A2DP Source - Can I use I2S as input?

MMSchneider, 2020-12-13:
Hello,

thank you so much for this excellent library. I do have a question. Can I use I2S incoming data as source and then send it via Bluetooth to a speaker? I was looking at the "bt_music_sender.ino" example and I guess I have to modify the get_data_channels() function. Any hint or code would be very appreciated. 

Best regards
Mike

pschatzmann, 2020-12-13:
I have never tried this myself but I would expect that you can just call esp_err_ti2s_read() in the get_data_channels() function.
Further info you can find in https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2s.html

Greetings from Switzerland
Phil

MMSchneider, 2020-12-13:
Thanks Phil, I will try that. If I am successful I will report back.

afmgomes, 2020-12-23:
Hello,

Thank you for the creation of this library. Great job.
I'm also trying to make the same as @MMSchneider. 

My question is, how can I verify if the queue is full? Also how can I start/stop the write to bluetooth?

Best regards,
André

Bollebie, 2021-02-07:
Hello,

I´m trying to do the same as MMSchneider.
Has anybody booked succes in this mather and if so, can you post the code.

greetings,
Bollebie

Electricious, 2021-03-16:
Hello all,

I've been trying to find a way to push data from the I2S to Bluetooth and the BT speaker for some time now, unfortunately without success. I seem to be missing something.

If anyone knows more, please share.

Best regards.
Electricious

rmond18, 2021-03-30:
Hi,

I'm also currently working on the same function for my I2S mic connected to esp32, but I'm also having a hard time figuring out how to stream the data continuously over Bluetooth to a BT sink.  Do you have any update on this? Thank you very much!

Regards,
rmond18

pschatzmann, 2021-04-29:
I finally had some time to look into this. It works like a charm!

https://www.pschatzmann.ch/home/2021/04/29/bluetooth-a2dp-streaming-from-an-digital-i2s-microphone/

closed 2021-04-29 by pschatzmann

Eternyt, 2021-05-05:
Just tried and works really well, thanks for the amazing work!

rmond18, 2021-05-06:
I also tried it, it works amazingly!! Thank you very much for you hard work!