pschatzmann/ESP32-A2DP #19: WiFi coexistence

judge2005, 2020-12-19:
Have you been able to get A2DP to coexist with WiFi? The ESP32 docs say it is possible, but I haven't succeeded so far. I haven't tried a simple test case yet - I have been trying to shoe-horn A2DP into an existing, fairly complex application, but just knowing if someone has succeeded would be useful.

pschatzmann, 2020-12-19:
I tried and failed and finally came to the conclusion that this is not working for the bandwidth that A2DP needs.
see the discussion at https://esp32.com/viewtopic.php?t=6707

An alternative migth be to connect 2 devices - one doing bluetooth and a separate one for Wifi (eg. ESP8277)…

judge2005, 2020-12-20:
It appears to be a heap size issue. I gradually built up the sophistication of my test application and monitored the free heap. Each time, if the free heap falls below around 48k, things start to go wrong.

You can free up around 8k by writing your own btStart() that only initializes ESP_BT_MODE_CLASSIC_BT mode, and then calling calling esp_bt_mem_release(ESP_BT_MODE_BLE) beforehand.

I'm not sure why everything starts to go wrong when the ESP32 is reporting 48k free heap, you would think that would be enough, and it is hard to believe it is so fragmented as to be useless. 

As it is, it is really only possible to write a basic application that uses both A2DP and WiFi. Maybe I need to investigate a chip with PSRAM.

judge2005, 2020-12-20:
I don’t want to spam, but i think it is worth leaving this link here for anyone who comes after https://docs.espressif.com/projects/esp-adf/en/latest/design-guide/design-considerations.html

closed 2022-12-09 by pschatzmann