pschatzmann/ESP32-A2DP #21: How to use 'esp_a2d_media_ctrl' command?

KIdon-Park, 2021-01-06:
Hi, Thank you very much for the work you made.

I use this library to use the BT speaker function, and I want to add some control functions.
I want to stop or restart music playing directly from the firmware.

I confirmed that esp_a2d_media_ctrl exists in esp_a2dp_api.h.
So, I was trying to use the function. When playing, enter the following command to stop
esp_a2d_media_ctrl(ESP_A2D_MEDIA_CTRL_SUSPEND)

In case of STOP or SUSPEND, execute the following to start playback.
esp_a2d_media_ctrl(ESP_A2D_MEDIA_CTRL_START)

Unfortunately, neither of these works.
Commands such as esp_bt_gap_set_scan_mode(ESP_BT_SCAN_MODE_CONNECTABLE) work.

Why can't command like play/pause music work? Is there any solution that you help me?

pschatzmann, 2021-01-07:
Unfortunately I never tried to do this myself. 

Did you have a look at AVRCP API (https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/esp_avrc.html). My gut feeling is that you might be able to pause the sender with the esp_avrc_ct_send_passthrough_cmd() by using ESP_AVRC_PT_CMD_PAUSE .

If you just want to switch the sound on and off you could overwrite the audio_data_callback() method.

KIdon-Park, 2021-01-12:
Thank you pschatzmann|
I will read the docu on that link. and try again. if I have any progress, will leave a message here.

KIdon-Park, 2021-01-20:
Thank you. it's helpful advice.
I try to these commands today. And it works great.

if anyone wants to use this function as below.

1. PLAY
      esp_avrc_ct_send_passthrough_cmd(0, ESP_AVRC_PT_CMD_PLAY, ESP_AVRC_PT_CMD_STATE_PRESSED);
      esp_avrc_ct_send_passthrough_cmd(0, ESP_AVRC_PT_CMD_PLAY, ESP_AVRC_PT_CMD_STATE_RELEASED);
      
2. PAUSE
      esp_avrc_ct_send_passthrough_cmd(0, ESP_AVRC_PT_CMD_PAUSE, ESP_AVRC_PT_CMD_STATE_PRESSED);
      esp_avrc_ct_send_passthrough_cmd(0, ESP_AVRC_PT_CMD_PAUSE, ESP_AVRC_PT_CMD_STATE_RELEASED);
      
3. STOP
      esp_avrc_ct_send_passthrough_cmd(0, ESP_AVRC_PT_CMD_STOP, ESP_AVRC_PT_CMD_STATE_PRESSED);
      esp_avrc_ct_send_passthrough_cmd(0, ESP_AVRC_PT_CMD_STOP, ESP_AVRC_PT_CMD_STATE_RELEASED);
      
4. NEXT
      esp_avrc_ct_send_passthrough_cmd(0, ESP_AVRC_PT_CMD_FORWARD, ESP_AVRC_PT_CMD_STATE_PRESSED);
      esp_avrc_ct_send_passthrough_cmd(0, ESP_AVRC_PT_CMD_FORWARD, ESP_AVRC_PT_CMD_STATE_RELEASED);
      
5. PREV
      esp_avrc_ct_send_passthrough_cmd(0, ESP_AVRC_PT_CMD_BACKWARD, ESP_AVRC_PT_CMD_STATE_PRESSED);
      esp_avrc_ct_send_passthrough_cmd(0, ESP_AVRC_PT_CMD_BACKWARD, ESP_AVRC_PT_CMD_STATE_RELEASED);

closed 2021-01-20 by KIdon-Park

biglee991228, 2021-03-29:
@KIdon-Park @pschatzmann 
thanks a lot for your advice,I got it!

void loop() {
 a2dp_sink.set_AVRC_metadata(avrc_metadata_callback); 
 //a2dp_sink.set_stream_reader(stream_reader_callback);
 if(Serial.available()){
  //read until a terminator. after this point there should only be numbers
  command = Serial.readStringUntil('#');
  if(command.equals("play")){
   esp_avrc_ct_send_passthrough_cmd(0, ESP_AVRC_PT_CMD_PLAY, ESP_AVRC_PT_CMD_STATE_PRESSED);
   esp_avrc_ct_send_passthrough_cmd(0, ESP_AVRC_PT_CMD_PLAY, ESP_AVRC_PT_CMD_STATE_RELEASED);
   Serial.println("play");
    }
  if(command.equals("pause")){
   esp_avrc_ct_send_passthrough_cmd(0, ESP_AVRC_PT_CMD_PAUSE, ESP_AVRC_PT_CMD_STATE_PRESSED);
   esp_avrc_ct_send_passthrough_cmd(0, ESP_AVRC_PT_CMD_PAUSE, ESP_AVRC_PT_CMD_STATE_RELEASED);
   Serial.println("pause");
    }  
  if(command.equals("stop")){
   esp_avrc_ct_send_passthrough_cmd(0, ESP_AVRC_PT_CMD_STOP, ESP_AVRC_PT_CMD_STATE_PRESSED);
   esp_avrc_ct_send_passthrough_cmd(0, ESP_AVRC_PT_CMD_STOP, ESP_AVRC_PT_CMD_STATE_RELEASED);
   Serial.println("stop");
    }  
  if(command.equals("forward")){
   esp_avrc_ct_send_passthrough_cmd(0, ESP_AVRC_PT_CMD_FORWARD, ESP_AVRC_PT_CMD_STATE_PRESSED);
   esp_avrc_ct_send_passthrough_cmd(0, ESP_AVRC_PT_CMD_FORWARD, ESP_AVRC_PT_CMD_STATE_RELEASED);
   Serial.println("forward");
    }
   if(command.equals("backward")){
   esp_avrc_ct_send_passthrough_cmd(0, ESP_AVRC_PT_CMD_BACKWARD, ESP_AVRC_PT_CMD_STATE_PRESSED);
   esp_avrc_ct_send_passthrough_cmd(0, ESP_AVRC_PT_CMD_BACKWARD, ESP_AVRC_PT_CMD_STATE_RELEASED);
   Serial.println("backward");
    }
 }
}
my phone was controlled correctly!

pschatzmann, 2021-03-29:
Cool - thanks for your feed back. 

Since this is so simple, I have finally decided to add this to my API as well...

BR-Kacha, 2024-05-08:
@biglee991228

Can you help me to solve this?
Compilation error: 'class BluetoothA2DPSink' has no member named 'set_AVRC_metadata'

I am including "BluetoothA2DPSink.h" library.

pschatzmann, 2024-05-08:
What do you expect to get if you add a non existing method ? 
I think the error message makes a lot of sense !

Please check the documentation for the available methods and their syntax. I guess you wanted to call set_avrc_metadata_callback