pschatzmann/ESP32-A2DP #22: Platformio

keeshanam, 2021-01-07:
Anyone have success when running under platformio?
I have tried and get this error:

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac

Works great under Arduino but personal prefrence would be to have in platformio.

Regards,

Aidan

yeckel, 2021-05-05:
I've got the same error when using Arduino with the wrong board. Not working was: _ESP32 Wrower module_ working is _TTGO LoRa32 OLED v1_

In platformio the ```board = ttgo-lora32-v21``` in platformio.ini works for me.

standarddeviant, 2021-05-28:
I'm able to build a PlatformIO project with ESP32-A2DP by making it a git submodule inside the `lib` folder of a new PlatformIO project. I haven't verified correct functionality, but it builds, loads, and runs on a TinyPico ESP32 board. I'm trying to play from ESP32 to earbuds over A2DP. I know the code is running because I print an updated timestamp every 5 seconds and I see that counting up.

When I get the bluetooth functionality working I'll post an example. I'm still trying to figure out how to get the ESP32 to initially connect to earbuds.

riraosan, 2021-05-31:
ESP32-A2DP  library worked in my PlatformIO environment. Device is ESP32(Rev.0).
I've made a few changes to the library for the My external DAC(ES9038Q2M VR1.07 DAC Board(eBay item number:263908779821))
Here is my Git repository.
Please for your reference.
https://github.com/riraosan/ESP32_BT-A2DP_Receiver

pschatzmann, 2021-07-31:
I finally tried to reproduce your issue, but everthing is working perfectly. I have added some instructions https://github.com/pschatzmann/ESP32-A2DP/wiki/PlatformIO

closed 2021-07-31 by pschatzmann