pschatzmann/ESP32-A2DP #24: Noise when playing using an Android phone.

KIdon-Park, 2021-01-12:
I am testing in several ways.
There was a (very strong) noise problem during testing.

I connected to ESP32-BT (with PCM5102 module) through my phone (Galaxy S10). When I play music, it sounds great.
However, during music playback, if I make a phone call or receive a call, the music that was playing is paused, and (very strong) noise occurs on the speaker connected to PCM5102 module. When you end the call, the music plays again and the noise disappears.

Interestingly, it only happens on Android phones. Other phones (iPhone SE) do not cause such a problem.

Can I solve this problem?

pschatzmann, 2021-01-12:
I am sorry, I don't have any experience with this scenario. 

I suggest to activate logging by setting the "Core Debug Level" in Arduino to Debug or Verbose and check if you get any event that you can use to react...

JohnyMielony, 2021-01-17:
I know how to solve this. Done this in my code. This same thing happens when you go out of range while playing music, right?
When I'll have some time I will make a pull request. Or you can do this easly by your own. Just stop I2S when sound is suspended, and start it when sound is started.

KIdon-Park, 2021-01-20:
JohnyMielony. I think you are right that the cause of appears noise after stop play.
Fortunately, I use the external DAC(PCM5102 module). So I decide to use SOFT-MUTE pin.
The SOFT-MUTE pin of PCM5102 is connected to the IO pin of ESP32, and when music playback is stopped or paused, a signal is given to SOFT-MUTE PIN of PCM5102 to perform HW MUTE.

I saw your library. Many new functions have been added, and I2S SUSPEND will help users who want to use the built-in DAC of ESP32. Over time, I'll try to apply your library.

Thank you.

JohnyMielony, 2021-01-20:
Don't use my fork directly. I modified it for my specific use case.
Go there:
https://github.com/pschatzmann/ESP32-A2DP/blob/1da399513b3581dc0b2d23376ada262526cc72b4/src/BluetoothA2DPSink.cpp#L377-L379
And replace it with that:
`if (ESP_A2D_AUDIO_STATE_STARTED == a2d->audio_stat.state) {
            m_pkt_cnt = 0;
			i2s_start(i2s_port);
        }
		else if ( ESP_A2D_AUDIO_STATE_REMOTE_SUSPEND == a2d->audio_stat.state ){
			i2s_stop(i2s_port);
		}`

That's just it, and the noise is gone.

KIdon-Park, 2021-01-21:
> Don't use my fork directly. I modified it for my specific use case.
> Go there: [...]

Many thank you
So I take your advice to use my project and also I add a condition on STOP in case of "ESP_A2D_AUDIO_STATE_STOPPED" also as below.

       if (ESP_A2D_AUDIO_STATE_STARTED == a2d->audio_stat.state) 
       { 
         m_pkt_cnt = 0; 
         i2s_start(i2s_port); 
       } else if ( ESP_A2D_AUDIO_STATE_REMOTE_SUSPEND == a2d->audio_stat.state ||
                   ESP_A2D_AUDIO_STATE_STOPPED == a2d->audio_stat.state                              // <- I add this
                 )
       { 
         i2s_stop(i2s_port); 
       }

KIdon-Park, 2021-01-21:
I am doing some tests related to the above i2s_start & stop, but there seems to be a little problem.
Regarding start & stop, the function seems to work. By the way, there is a problem that BT cannot connect again if I repeat disconnecting and connecting BT.
It is presumed that it is a problem caused by connecting and disconnecting only the i2s port by omitting the before and after steps of A2DPSink.
Maybe more tests are needed.

grisuthedragon, 2021-02-15:
I have a simliar problem when using the PCM5102 as DAC but independent of the source. I used my Galaxy S9 and my Ubuntu Laptop as source. If a let a playlist go through I have the noise only at the begin of the first track. If I jump between songs, I have the noise the begin of nearly each track.  I already integrated the patch from above but this does not change the situation. So any progress on this?

sidney-ikeda, 2021-03-13:
> > Don't use my fork directly. I modified it for my specific use case.
> > Go there:
> > https://github.com/pschatzmann/ESP32-A2DP/blob/1da399513b3581dc0b2d23376ada262526cc72b4/src/BluetoothA2DPSink.cpp#L377-L379
> > 
> > And replace it with that:
> > `if (ESP_A2D_AUDIO_STATE_STARTED == a2d->audio_stat.state) { m_pkt_cnt = 0; i2s_start(i2s_port); } else if ( ESP_A2D_AUDIO_STATE_REMOTE_SUSPEND == a2d->audio_stat.state ){ i2s_stop(i2s_port); }`
> > That's just it, and the noise is gone.
> 
> Many thank you
> So I take your advice to use my project and also I add a condition on STOP in case of "ESP_A2D_AUDIO_STATE_STOPPED" as below.
> 
> ```
>    if (ESP_A2D_AUDIO_STATE_STARTED == a2d->audio_stat.state) 
>    { 
>      m_pkt_cnt = 0; 
>      i2s_start(i2s_port); 
>    } else if ( ESP_A2D_AUDIO_STATE_REMOTE_SUSPEND == a2d->audio_stat.state ||
>                ESP_A2D_AUDIO_STATE_STOPPED == a2d->audio_stat.state                              // <- I add this
>              )
>    { 
>      i2s_stop(i2s_port); 
>    }
> ```

> > Don't use my fork directly. I modified it for my specific use case.
> > Go there:
> > https://github.com/pschatzmann/ESP32-A2DP/blob/1da399513b3581dc0b2d23376ada262526cc72b4/src/BluetoothA2DPSink.cpp#L377-L379
> > 
> > And replace it with that:
> > `if (ESP_A2D_AUDIO_STATE_STARTED == a2d->audio_stat.state) { m_pkt_cnt = 0; i2s_start(i2s_port); } else if ( ESP_A2D_AUDIO_STATE_REMOTE_SUSPEND == a2d->audio_stat.state ){ i2s_stop(i2s_port); }`
> > That's just it, and the noise is gone.
> 
> Many thank you
> So I take your advice to use my project and also I add a condition on STOP in case of "ESP_A2D_AUDIO_STATE_STOPPED" as below.
> 
> ```
>    if (ESP_A2D_AUDIO_STATE_STARTED == a2d->audio_stat.state) 
>    { 
>      m_pkt_cnt = 0; 
>      i2s_start(i2s_port); 
>    } else if ( ESP_A2D_AUDIO_STATE_REMOTE_SUSPEND == a2d->audio_stat.state ||
>                ESP_A2D_AUDIO_STATE_STOPPED == a2d->audio_stat.state                              // <- I add this
>              )
>    { 
>      i2s_stop(i2s_port); 
>    }
> ```

Thank you for this suggestion. Also I included this i2s STOP in my project but still some noise when skip or stop the sound suddenly. I guess the noise occur when the  app_a2d_callback_2() function is called. Including i2s_stop in line 641 resolved the problem:

extern "C" void app_a2d_callback_2(esp_a2d_cb_event_t event, esp_a2d_cb_param_t *param){
i2s_stop(i2s_port); // <- added
ESP_LOGD(BT_AV_TAG, "%s", func);
if (actualBluetoothA2DPSink)
actualBluetoothA2DPSink->app_a2d_callback(event, param);
}

pschatzmann, 2021-04-16:
Merged the proposed improvements to my code

closed 2021-04-16 by pschatzmann