pschatzmann/ESP32-A2DP #25: Using bt_music_sender_write example but with input from SD card?

cuongnguyenvirtual, 2021-01-13:
Hi,

I'm a beginner to ESP32 and such, and your lib is a great help.
I'm trying to make a MP3 player that read from SD card and output audio to bluetooth speaker.
How can I modify your bt_music_sender_write example to do this? Specifically, how should I read raw data from storage instead of predefined const variable (which makes sketch size big as you mentioned in README).
I'm using a M5Stack device, and trying to achieve this with this example: https://github.com/m5stack/M5Stack/tree/master/examples/Advanced/Audio/PlayMP3FromSDToDAC
In the example above, they uses lib https://github.com/earlephilhower/ESP8266Audio, is it possible to integrate this lib with yours?

Best regards,
Cuong.

cuongnguyenvirtual, 2021-01-13:
oh, I should mention that the m5stack example above plays mp3 files from SD card but to built-in speaker of the device

KingJimmy27, 2021-02-21:
I would also be interested in this. I am trying to send numerical audio information to bluetooth headphones.

pschatzmann, 2021-04-29:
I am currently working on some more scenarios with examples. Here is a first basic SD example

https://github.com/pschatzmann/arduino-sound-tools/tree/main/examples/file_raw-a2dp

closed 2021-04-29 by pschatzmann