pschatzmann/ESP32-A2DP #26: Cannot reconnect to BT sink

leventeBajczi, 2021-01-15:
I use the BluetoothA2DPSink class to create a BT sink, which works perfectly when first connected. However, after disconnecting I cannot seem to reconnect from either a laptop (Arch Linux, BlueZ 5.55, Thinkpad E490) or a phone (Samsung Galaxy A50). I have not tested with other devices, but I reckon that would be the same. The BT connection is in the `ESP_A2D_CONNECTION_STATE_DISCONNECTED` state at this point, and a manual reconnection attempt yields no debug info on the console.

I have solved the issue by performing a full restart (using `esp_restart()`) every time a device disconnects, and also inhibiting connection to the last known device on startup (as otherwise disconnecting would be impossible), but this is very hacky and I would much rather use a solution to the actual problem itself.

The code I am using: 

```
#include "BluetoothA2DPSink.h"
BluetoothA2DPSink a2dp_sink;
void setup() {
   static const i2s_config_t i2s_config = {
      .mode = (i2s_mode_t) (I2S_MODE_MASTER | I2S_MODE_TX),
      .sample_rate = 44100,
      .bits_per_sample = (i2s_bits_per_sample_t) 16,
      .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,
      .communication_format = I2S_COMM_FORMAT_I2S_MSB,
      .intr_alloc_flags = 0,
      .dma_buf_count = 8,
      .dma_buf_len = 64,
      .use_apll = false
  };
  static const i2s_pin_config_t pin_config = {
      .bck_io_num = 26,
      .ws_io_num = 25,
      .data_out_num = 27,
      .data_in_num = I2S_PIN_NO_CHANGE
  };
  a2dp_sink.set_i2s_config(i2s_config);  
  a2dp_sink.set_pin_config(pin_config);
  a2dp_sink.set_on_data_received(data_received_callback);
  a2dp_sink.start("Proba");
}
void loop(){}
```

pschatzmann, 2021-01-16:
I guess that this is caused by the change which automatically reconnects to the last device which is available in the snapshot release. Can you try out the release v1.1.0 which does not contain this change and give me some feed back ?

If you confirm that this is the issue I can try to implement a switch which deactivates this functionality....

leventeBajczi, 2021-01-16:
With v1.1 it works exactly as I would expect it to - I can connect, disconnect, reconnect and restart the device in any order without having to worry about manually resetting it. 

If I understand correctly, the only lost functionality is the auto-reconnect feature that would initiate a connection to the last device on startup. On any BT speakers I have used, this is only done when a device was not manually disconnected, but rather the speaker was switched off (or unplugged, etc). Maybe upon a clean disconnect, the device should clear the last known source's address, and it should only attempt to reconnect if no there was no manual disconnect the last time?

JohnyMielony, 2021-01-17:
I've coded that auto reconnect. There is setting that determines how many times esp tries to connect to last device (defaults to 5), while it tries you can't connect to it with other device, if it couldn't connect for the fifth time it then goes to discoverable and connectable mode. So then you should be able to connect it.

leventeBajczi, 2021-01-17:
Initially I found this bug trying to connect to the device _hours_ after the first disconnect, to no avail. If the 5 or so attempts time out after a while, maybe there is another problem preventing me from reconnecting - any idea what that could be?

JohnyMielony, 2021-01-18:
Before pulling this change I tested it with two phones, and it worked well. I would try to turn on debug messages and see what is happening.
Maybe add this #define AUTOCONNECT_TRY_NUM (5) to your sketch with a lower value.

leventeBajczi, 2021-01-18:
I have tried this out again with a different device, and I can successfully reconnect - so I think the problem might be with the devices I tested with before. However, I still think this change is useful in that it disallows reconnection to devices which had been previously disconnected manually, so I am leaving PR #27 open - but I am closing this issue as the mentioned bug is not actually the fault of this project.

closed 2021-01-18 by leventeBajczi