pschatzmann/ESP32-A2DP #29: Implementing metadata on Arduino IDE

GregOneX, 2021-01-21:
Hello, recently I've discovered this awesome library which helped me make my project a lot easier. As adding an OLED display and an ADC took only a few lines of code. The thing is that I don't know how to implement the "send_metadata_cmd" as well as CASE "ESP_AVRC_CT_METADATA_RSP_EVT" so that I could store these attributes to some variables on Arduino IDE. 

Adding this code: esp_avrc_ct_send_metadata_cmd(0, ESP_AVRC_MD_ATTR_TITLE); to void loop() compiled successfully, but I just can't confirm it works as I mentioned above.

If someone already tried this, I would really appreciate the help, as I just can't seem to find answers.

Best regards,
Greg

lukasilestam, 2021-02-05:
try add:

BluetoothA2DPSink.h:
```
public:
  virtual void set_AVRC_metadata(void (*callBack)(uint8_t, const uint8_t*));

protected:
  void (*AVRC_metadata)(uint8_t, const uint8_t*) = NULL;
```

BluetoothA2DPSink.cpp
```
void BluetoothA2DPSink::set_AVRC_metadata(void (*callBack)(uint8_t, const uint8_t*)){
   this->AVRC_metadata = callBack;
}
```

and replace:

https://github.com/pschatzmann/ESP32-A2DP/blob/1da399513b3581dc0b2d23376ada262526cc72b4/src/BluetoothA2DPSink.cpp#L458-L462

```
 case ESP_AVRC_CT_METADATA_RSP_EVT: {
        if (AVRC_metadata!=NULL){
            AVRC_metadata(rc->meta_rsp.attr_id, rc->meta_rsp.attr_text);
        }
        ESP_LOGI(BT_AV_TAG, "AVRC metadata rsp: attribute id 0x%x, %s", rc->meta_rsp.attr_id, rc->meta_rsp.attr_text);
        free(rc->meta_rsp.attr_text);
        break;
}
```

and in project:

```
void avrc_metadata_callback(uint8_t data1, const uint8_t *data2)
{
  Serial.printf("AVRC metadata rsp: attribute id 0x%x, %s\n", data1, data2);
}
a2dp_sink.set_AVRC_metadata(avrc_metadata_callback);
a2dp_sink.start("BT");
```

biglee991228, 2021-03-22:
@lukasilestam 
thanks a lot !
I got it and successed
ESP_LOGD(BT_AV_TAG, "%s", __func__);
    //Register notifications and request metadata
    esp_avrc_ct_send_metadata_cmd(0, ESP_AVRC_MD_ATTR_TITLE | ESP_AVRC_MD_ATTR_ARTIST | ESP_AVRC_MD_ATTR_ALBUM | ESP_AVRC_MD_ATTR_GENRE | ESP_AVRC_MD_ATTR_PLAYING_TIME);//
    esp_avrc_ct_send_register_notification_cmd(1, ESP_AVRC_RN_TRACK_CHANGE, 0);
and I added "ESP_AVRC_MD_ATTR_PLAYING_TIME"

pschatzmann, 2021-03-25:
I have add this logic to the basic code. 
Please note that the method to register the callback is called:

a2dp_sink.set_avrc_metadata_callback

closed 2021-03-25 by pschatzmann