pschatzmann/ESP32-A2DP #31: Output to VS1053.

Aleks-Ale, 2021-02-27:
Is it possible to output the stream using the VS1053?
I have a web radio with ESP32/VS1053 and it would be very interesting to add BT mode for playing from a smartphone.

pschatzmann, 2021-02-27:
I was running over the spec of the VS1053 and my gut feeling is it's worth a try:

- You could try to feed the PCM output of the Bluetooth to the built in DAC of the ESP32 and connect it to the Audio In of the  VS1053 but the result will not be great.
- Another better way might be to use the method described in chapter 9.6 Feeding PCM data.
 https://www.sparkfun.com/datasheets/Components/SMD/vs1053.pdf. I have seen that the API provides the method void playData(uint8_t *buffer, uint8_t buffsiz) - So you could prefix the first data with a WAV header and then just keep on sending the PCM data. You would need to create a subclass of BluetoothA2DPSink and overwrite the method audio_data_callback(const uint8_t *data, uint32_t len) which currently writes the data to I2S.

Semenov-Alexander, 2021-02-27:
Thank you for your answer. 
Unfortunately, my level of knowledge will not allow me to implement this. 
Is this implementation not interesting to you? 
This would be a really very useful project for a lot of people.

Aleks-Ale, 2021-02-27:
Thank you for your answer.
Unfortunately, my level of knowledge will not allow me to implement this.
Is this implementation not interesting to you?
This would be a really very useful project for a lot of people.

pschatzmann, 2021-02-28:
I leave this open - maybe somebody who has a VS1053 might take this up...

Derek-K, 2021-06-07:
I was just thinking about the same thing https://github.com/Edzelf/ESP32-Radio/issues/466 the other day.. and upon some research, if I am not mistaken, please correct me if I am wrong. One could use esp_a2d_sink_register_data_callback function and get a block of audio data received by ESP32 in PCM format. Then somehow repackage it with WAV header (?) and pass it onto VS1053.

Just too bad VS1053 cannot take the I2S signal directly to it's DAC...

I am still learning A2DP, but isn't it suppose to support MP3 codec as well (assume the source device can stream)? If so that would make it a lot easier to include in ESP32-Radio code... Just a thought.

fodigyorgy, 2021-06-12:
I was thinking the same, but I am not educated enough to get it working, the vs1053 library has a function to send datapackeges via spi, maybe it could work. 
player.playChunk(mp3buff, 64);

Derek-K, 2021-06-18:
Hi @pschatzmann, I have been going back and forth on documentation and forums, is there an API call to access the A2DP data or redirect the I2S stream to a buffer, so that another routine can do the 'conversion' to PCM/WAV format then feed to VS1053?

pschatzmann, 2021-06-18:
Definitly yes - copy paste from the documentation...

// In the setup function:
a2dp_sink.set_stream_reader(read_data_stream);

// Then somewhere in your sketch:
void read_data_stream(const uint8_t *data, uint32_t length)
{
  // Do something with the data packet
}
In the a2dp_sink.set_stream_reader() method you can provide an optional parameter that defines if you want the output to I2S to be active or deactive - So you can use this method to e.g. to switch off I2S just by calling a2dp_sink.set_stream_reader(read_data_stream, false)

Derek-K, 2021-06-18:
Thanks. I will take a closer look when I get a chance.

BTW, for those interested in using it with ESP32 Radio, there could be an issue... If I am not mistaken, A2DP cannot co-exist with WiFi Enable. 

So I am thinking about how to switch back and forth between web and A2DP streams. (Sorry if this is off topic maybe I will move the discussion over to ESP32 radio) Just before I go too deep into the rabbit hole to trying to make A2DP->VS1053 to work, please feel free to comment...

In normal Web stream mode, WiFi is enabled. If the user wants to stream BT-Audio, a command will be sent (either via Web interface/MQTT/Serial). After displaying the BT name, disable WiFi and start waiting for BT pairing. (Optionally a time out could be set, if waited for too long w/o connection, will stop BT and re-enable WiFi)

If BT pairing is successful, stream BT audio as designed.

In order to switch back to WiFi mode, disconnect BT pairing, and maybe wait for a short period (e.g. 30 seconds), disable BT and re-enable WiFi.

Does the logic make sense?

As for dealing with raw A2DP data, does anyone have experience here?

pschatzmann, 2021-06-18:
Its int16_t PCM data with 2 channels. You can directly use it w/o any manipulation. All you need is to prefix a valid WAV header with 
- format: 0x01 PCM
- sample_rate from i2s_config.sample_rate
- channels: 2
- BitsPerSample: 16

jeroenlukas, 2021-06-18:
I've been using this library for a week now, combined with the VS1053. It is possible to intercept the PCM-data packets with the stream reader and send them to the VS1053. However keep in mind you will have to send the correct WAV/RIFF header first (44 bytes). 

For an example, you can check out my project: https://github.com/jeroenlukas/KitchenRadio/blob/master/src/kr_bluetoothsink.cpp (please bear in mind this is very much a work in progress :D)

I am using the bluetooth sink combined with a circular buffer. But I think you'll get an idea of how it works. Basically I add the PCM data from the bluetooth stream (most of the time 4095 bytes) to the circular buffer. Then you just send 32 bytes from the circular buffer to the VS1053 on a regular basis, like you normally would. Keep in mind that PCM has no data compression so it's more likely you'll get buffer underruns etc. I'm still working on that. 

Just remember to always send the WAVE-header before you're sending PCM data to the 1053. The contents of this header - I think - could be left unchanged. The header describes no. of channels, bitrate, sample rate etc. but they don't seem to change between apps. I've tested this with Spotify, Youtube, Youtube music, PocketCasts and they all result in a 48kHz 16 bit stereo stream.

Last but not least a pro tip: don't forget to turn up the media volume on your phone when you're testing this, it'll save you a couple of needless hours of debugging ;)

psolyca, 2021-06-27:
@jeroenlukas thanks for your project.
I have used your method and it is working perfectly.
I just have changed the header of the WAV file to 44k1 (will upload it later).

And last but no least, a second pro tip: do not forget to remove DND mode ^^;

paulgreg, 2022-01-02:
For the record, I recently made a bluetooth receiver using your excellent library with an Adafruit VS1053. It works great (except for bluetooth range but that’s another issue).
If someone needs an example, code is here : https://github.com/paulgreg/esp32-bluetooth-audio/ and project is on hackaday : https://hackaday.io/project/183183-esp32-bluetooth-audio-receiver