pschatzmann/ESP32-A2DP #34: Examples don't compile

d-999, 2021-03-24:
Hey there...
It's pretty cool you put in the work to "port" the A2DP API from Espressif to Arduino. I wanted to quickly try and flash it on an ESP32 Wemos D1 mini using the Arduino IDE of course (Version 1.8.13 on Windows). Unfortunately both receiver examples (the only ones I tried) fail to compile. Perhaps I'm missing something. Are there any dependencies that I'm not aware of? I looked through your description, but didn't see any obvious dependencies mentioned.

pschatzmann, 2021-03-24:
I am not aware of any dependencies. 
What errors are you getting ?

d-999, 2021-03-24:
Hi...
Well, the compilation goes nomally this happens:
`C:\Users\Delphino\Documents\Arduino\libraries\ESP32-A2DP-master\src\BluetoothA2DPSink.cpp:` In member function 'virtual void BluetoothA2DPSink::init_nvs()':

C:\Users\Delphino\Documents\Arduino\libraries\ESP32-A2DP-master\src\BluetoothA2DPSink.cpp:583:52: error: 'ESP_ERR_NVS_NEW_VERSION_FOUND' was not declared in this scope
     if (err == ESP_ERR_NVS_NO_FREE_PAGES || err == ESP_ERR_NVS_NEW_VERSION_FOUND){
                                                    ^
In file included from C:\Users\Delphino\Documents\Arduino\libraries\ESP32-A2DP-master\src\BluetoothA2DPSource.cpp:16:0:
C:\Users\Delphino\Documents\Arduino\libraries\ESP32-A2DP-master\src\BluetoothA2DPSource.h:93:46: error: 'esp_bt_pin_type_t' has not been declared
     virtual  void setPinCode(char* pin_code, esp_bt_pin_type_t pin_type=ESP_BT_PIN_TYPE_VARIABLE);
                                              ^
C:\Users\Delphino\Documents\Arduino\libraries\ESP32-A2DP-master\src\BluetoothA2DPSource.h:151:5: error: 'esp_bt_pin_type_t' does not name a type
     esp_bt_pin_type_t pin_type;
     ^
C:\Users\Delphino\Documents\Arduino\libraries\ESP32-A2DP-master\src\BluetoothA2DPSource.h:152:5: error: 'esp_bt_pin_code_t' does not name a type
     esp_bt_pin_code_t pin_code;
     ^
C:\Users\Delphino\Documents\Arduino\libraries\ESP32-A2DP-master\src\BluetoothA2DPSource.h:93:73: error: 'ESP_BT_PIN_TYPE_VARIABLE' was not declared in this scope
     virtual  void setPinCode(char* pin_code, esp_bt_pin_type_t pin_type=ESP_BT_PIN_TYPE_VARIABLE);
                                                                         ^
C:\Users\Delphino\Documents\Arduino\libraries\ESP32-A2DP-master\src\BluetoothA2DPSource.cpp: In constructor 'BluetoothA2DPSource::BluetoothA2DPSource()':
C:\Users\Delphino\Documents\Arduino\libraries\ESP32-A2DP-master\src\BluetoothA2DPSource.cpp:114:11: error: 'class BluetoothA2DPSource' has no member named 'pin_type'
     this->pin_type = ESP_BT_PIN_TYPE_VARIABLE;
           ^
C:\Users\Delphino\Documents\Arduino\libraries\ESP32-A2DP-master\src\BluetoothA2DPSource.cpp:114:22: error: 'ESP_BT_PIN_TYPE_VARIABLE' was not declared in this scope
     this->pin_type = ESP_BT_PIN_TYPE_VARIABLE;
                      ^
C:\Users\Delphino\Documents\Arduino\libraries\ESP32-A2DP-master\src\BluetoothA2DPSource.cpp:117:19: error: 'pin_code' was not declared in this scope
     strcpy((char*)pin_code, "1234");
                   ^
C:\Users\Delphino\Documents\Arduino\libraries\ESP32-A2DP-master\src\BluetoothA2DPSource.cpp: At global scope:
C:\Users\Delphino\Documents\Arduino\libraries\ESP32-A2DP-master\src\BluetoothA2DPSource.cpp:137:54: error: 'esp_bt_pin_type_t' has not been declared
 void BluetoothA2DPSource::setPinCode(char *pin_code, esp_bt_pin_type_t pin_type){
                                                      ^
C:\Users\Delphino\Documents\Arduino\libraries\ESP32-A2DP-master\src\BluetoothA2DPSource.cpp: In member function 'virtual void BluetoothA2DPSource::setPinCode(char*, int)':
C:\Users\Delphino\Documents\Arduino\libraries\ESP32-A2DP-master\src\BluetoothA2DPSource.cpp:139:11: error: 'class BluetoothA2DPSource' has no member named 'pin_type'
     this->pin_type = pin_type;
           ^
C:\Users\Delphino\Documents\Arduino\libraries\ESP32-A2DP-master\src\BluetoothA2DPSource.cpp:141:25: error: 'class BluetoothA2DPSource' has no member named 'pin_code'
     strcpy((char*)this->pin_code, pin_code);
                         ^
C:\Users\Delphino\Documents\Arduino\libraries\ESP32-A2DP-master\src\BluetoothA2DPSource.cpp: In member function 'virtual void BluetoothA2DPSource::startRaw(std::vector<char*>, music_data_cb_t, bool)':
C:\Users\Delphino\Documents\Arduino\libraries\ESP32-A2DP-master\src\BluetoothA2DPSource.cpp:175:56: error: 'ESP_ERR_NVS_NEW_VERSION_FOUND' was not declared in this scope
         if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
                                                        ^
C:\Users\Delphino\Documents\Arduino\libraries\ESP32-A2DP-master\src\BluetoothA2DPSource.cpp:222:9: error: 'esp_bt_sp_param_t' was not declared in this scope
         esp_bt_sp_param_t param_type = ESP_BT_SP_IOCAP_MODE;
         ^
C:\Users\Delphino\Documents\Arduino\libraries\ESP32-A2DP-master\src\BluetoothA2DPSource.cpp:223:9: error: 'esp_bt_io_cap_t' was not declared in this scope
         esp_bt_io_cap_t iocap = ESP_BT_IO_CAP_IO;
         ^
C:\Users\Delphino\Documents\Arduino\libraries\ESP32-A2DP-master\src\BluetoothA2DPSource.cpp:224:39: error: 'param_type' was not declared in this scope
         esp_bt_gap_set_security_param(param_type, &iocap, sizeof(uint8_t));
                                       ^
C:\Users\Delphino\Documents\Arduino\libraries\ESP32-A2DP-master\src\BluetoothA2DPSource.cpp:224:52: error: 'iocap' was not declared in this scope
         esp_bt_gap_set_security_param(param_type, &iocap, sizeof(uint8_t));
                                                    ^
C:\Users\Delphino\Documents\Arduino\libraries\ESP32-A2DP-master\src\BluetoothA2DPSource.cpp:224:74: error: 'esp_bt_gap_set_security_param' was not declared in this scope
         esp_bt_gap_set_security_param(param_type, &iocap, sizeof(uint8_t));
                                                                          ^
C:\Users\Delphino\Documents\Arduino\libraries\ESP32-A2DP-master\src\BluetoothA2DPSource.cpp:231:24: error: 'pin_type' was not declared in this scope
[... cut ...]
C:\Users\Delphino\Documents\Arduino\libraries\ESP32-A2DP-master\src\BluetoothA2DPSource.cpp:482:10: error: 'ESP_BT_GAP_KEY_NOTIF_EVT' was not declared in this scope
     case ESP_BT_GAP_KEY_NOTIF_EVT:
          ^
C:\Users\Delphino\Documents\Arduino\libraries\ESP32-A2DP-master\src\BluetoothA2DPSource.cpp:486:10: error: 'ESP_BT_GAP_KEY_REQ_EVT' was not declared in this scope
     case ESP_BT_GAP_KEY_REQ_EVT:
          ^
Using library ESP32-A2DP-master at version 1.0 in folder: C:\Users\Delphino\Documents\Arduino\libraries\ESP32-A2DP-master 
exit status 1
Error compiling for board WEMOS LOLIN32`

d-999, 2021-03-24:
Ok, interestingly I tried the previous version of your ported library (version 1.0.1) and it compiled. So whatever the problem is, it seems to only occur at the current version.

d-999, 2021-03-24:
Ok, I tried flashing version 1.0.1 to an ESP32 (same setup). And it flashes successfully. But then it goes into a boot loop and just continuously starts up and crashes again.

pschatzmann, 2021-03-25:
Strange....just to confirm I deleted my local library and installed the version from github.
I can compile both receiver examples without any issues. 
I was using the setting
-> Boards -> ESP32 Arduino -> ESP32 Dev Module

What Board settings are you using ?

d-999, 2021-03-25:
Huh... Well, I'm using "Boards -> ESP32 Arduino -> WEMOS LOLIN32" ESP32 I also tried "Boards -> ESP32 Arduino -> Wrover Module". But both didn't work. As I already mentioned version 1.0.1 of the library compiles and uploads successfully, but it still doesn't run.
I tried uploading the original "a2dp_sink" example from esp-idf. And it works fine, so at least it doesn't seem to be a hardware problem.

pschatzmann, 2021-03-25:
Strange...both of these are compiling.
I dont think this is the issue but, if you check in the Boards Manager: which ESP32 version is installed ?

d-999, 2021-03-25:
Oh my god... That totally was the problem. For some reason I was still running ESP32 version 1.0.0, I think I had installed an older version once to test something and forgot to install the new version again afterwards.
I'm sorry, I should have thought of that myself. Anyway, with ESP32 version 1.0.5 everything works perfectly.
Thanks for the good work and sorry for kind of wasting you time.

pschatzmann, 2021-03-25:
cool

closed 2021-03-25 by pschatzmann