pschatzmann/ESP32-A2DP #35: Support for Metadata only work in Arduino Framework 1.0.4

sidney-ikeda, 2021-04-12:
Hi,

Let me know if someone is facing in this problem.
I'm using the sketch bt_music_receiver_metadata.ino as example.
The support for Metadata (and also AVRC Commands) only works in Arduino Framework 1.0.4. Above this version as 1.0.5 and 1.0.6 has this issue.
Debbuging look like callback 'app_rc_ct_callback()' is not working for newest version of Arduino Framework.
Also I tried in Visual Studio Platformio with Arduino Framework and only work in the version 3.0.0(framework 1.0.4), above this version same issue.

**Follow the debbuging for 1.0.4:**
[D][BluetoothA2DPSink.cpp:122] start(): start
[I][BluetoothA2DPSink.cpp:127] start(): Device name will be set to 'ESP32_SPEAKER'
[D][BluetoothA2DPSink.cpp:174] init_bluetooth(): init_bluetooth
[I][BluetoothA2DPSink.cpp:179] init_bluetooth(): controller initialized
[I][BluetoothA2DPSink.cpp:188] init_bluetooth(): bluedroid initialized
[I][BluetoothA2DPSink.cpp:196] init_bluetooth(): bluedroid enabled
[D][BluetoothA2DPSink.cpp:274] app_task_start_up(): app_task_start_up
[D][BluetoothA2DPSink.cpp:681] app_task_handler_2(): app_task_handler_2
[D][BluetoothA2DPSink.cpp:250] app_task_handler(): app_task_handler
[D][BluetoothA2DPSink.cpp:203] app_work_dispatch(): app_work_dispatch event 0x0, param len 0
[D][BluetoothA2DPSink.cpp:235] app_send_msg(): app_send_msg
[D][BluetoothA2DPSink.cpp:254] app_task_handler(): app_task_handler, sig 0x1, 0x0
[W][BluetoothA2DPSink.cpp:257] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[D][BluetoothA2DPSink.cpp:226] app_work_dispatched(): app_work_dispatched
[D][BluetoothA2DPSink.cpp:141] operator()(): operator()
[D][BluetoothA2DPSink.cpp:508] av_hdl_stack_evt(): av_hdl_stack_evt av_hdl_stack_evt BT_APP_EVT_STACK_UP
[D][BluetoothA2DPSink.cpp:693] app_a2d_callback_2(): app_a2d_callback_2
[D][BluetoothA2DPSink.cpp:545] app_a2d_callback(): app_a2d_callback
[D][BluetoothA2DPSink.cpp:548] app_a2d_callback(): app_a2d_callback ESP_A2D_CONNECTION_STATE_EVT
[D][BluetoothA2DPSink.cpp:203] app_work_dispatch(): app_work_dispatch event 0x0, param len 16
[D][BluetoothA2DPSink.cpp:235] app_send_msg(): app_send_msg
[D][BluetoothA2DPSink.cpp:254] app_task_handler(): app_task_handler, sig 0x1, 0x0
[W][BluetoothA2DPSink.cpp:257] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[D][BluetoothA2DPSink.cpp:226] app_work_dispatched(): app_work_dispatched
[D][BluetoothA2DPSink.cpp:693] app_a2d_callback_2(): app_a2d_callback_2
[D][BluetoothA2DPSink.cpp:539] operator()(): operator()
[D][BluetoothA2DPSink.cpp:348] av_hdl_a2d_evt(): av_hdl_a2d_evt evt 0
[D][BluetoothA2DPSink.cpp:352] av_hdl_a2d_evt(): av_hdl_a2d_evt ESP_A2D_CONNECTION_STATE_EVT
[I][BluetoothA2DPSink.cpp:356] av_hdl_a2d_evt(): A2DP connection state: Connecting, [00:57:c1:43:a8:5b]
[I][BluetoothA2DPSink.cpp:380] av_hdl_a2d_evt(): ESP_A2D_CONNECTION_STATE_CONNECTING
[D][BluetoothA2DPSink.cpp:545] app_a2d_callback(): app_a2d_callback
[D][BluetoothA2DPSink.cpp:557] app_a2d_callback(): app_a2d_callback ESP_A2D_AUDIO_CFG_EVT
[D][BluetoothA2DPSink.cpp:203] app_work_dispatch(): app_work_dispatch event 0x2, param len 16
[D][BluetoothA2DPSink.cpp:235] app_send_msg(): app_send_msg
[D][BluetoothA2DPSink.cpp:254] app_task_handler(): app_task_handler, sig 0x1, 0x2
[W][BluetoothA2DPSink.cpp:257] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[D][BluetoothA2DPSink.cpp:226] app_work_dispatched(): app_work_dispatched
[D][BluetoothA2DPSink.cpp:539] operator()(): operator()
[D][BluetoothA2DPSink.cpp:348] av_hdl_a2d_evt(): av_hdl_a2d_evt evt 2
[D][BluetoothA2DPSink.cpp:403] av_hdl_a2d_evt(): av_hdl_a2d_evt ESP_A2D_AUDIO_CFG_EVT
[I][BluetoothA2DPSink.cpp:407] av_hdl_a2d_evt(): a2dp audio_cfg_cb , codec type 0
[I][BluetoothA2DPSink.cpp:426] av_hdl_a2d_evt(): configure audio player 21-15-2-35

[I][BluetoothA2DPSink.cpp:427] av_hdl_a2d_evt(): audio player configured, samplerate=44100
[D][BluetoothA2DPSink.cpp:693] app_a2d_callback_2(): app_a2d_callback_2
[D][BluetoothA2DPSink.cpp:545] app_a2d_callback(): app_a2d_callback
[D][BluetoothA2DPSink.cpp:548] app_a2d_callback(): app_a2d_callback ESP_A2D_CONNECTION_STATE_EVT
[D][BluetoothA2DPSink.cpp:203] app_work_dispatch(): app_work_dispatch event 0x0, param len 16
[D][BluetoothA2DPSink.cpp:235] app_send_msg(): app_send_msg
[D][BluetoothA2DPSink.cpp:254] app_task_handler(): app_task_handler, sig 0x1, 0x0
[W][BluetoothA2DPSink.cpp:257] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[D][BluetoothA2DPSink.cpp:226] app_work_dispatched(): app_work_dispatched
[D][BluetoothA2DPSink.cpp:539] operator()(): operator()
[D][BluetoothA2DPSink.cpp:348] av_hdl_a2d_evt(): av_hdl_a2d_evt evt 0
[D][BluetoothA2DPSink.cpp:352] av_hdl_a2d_evt(): av_hdl_a2d_evt ESP_A2D_CONNECTION_STATE_EVT
[I][BluetoothA2DPSink.cpp:356] av_hdl_a2d_evt(): A2DP connection state: Connected, [00:57:c1:43:a8:5b]
[I][BluetoothA2DPSink.cpp:374] av_hdl_a2d_evt(): ESP_A2D_CONNECTION_STATE_CONNECTED
[D][BluetoothA2DPSink.cpp:699] app_rc_ct_callback_2(): app_rc_ct_callback_2 
**_[D][BluetoothA2DPSink.cpp:307] app_rc_ct_callback(): app_rc_ct_callback  (<-here callback is working)_**
[D][BluetoothA2DPSink.cpp:324] app_rc_ct_callback(): app_rc_ct_callback ESP_AVRC_CT_CONNECTION_STATE_EVT
[D][BluetoothA2DPSink.cpp:203] app_work_dispatch(): app_work_dispatch event 0x0, param len 12
[... 146 lines cut ...]
[D][BluetoothA2DPSink.cpp:254] app_task_handler(): app_task_handler, sig 0x1, 0x0
[W][BluetoothA2DPSink.cpp:257] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[D][BluetoothA2DPSink.cpp:226] app_work_dispatched(): app_work_dispatched
[D][BluetoothA2DPSink.cpp:539] operator()(): operator()
[D][BluetoothA2DPSink.cpp:348] av_hdl_a2d_evt(): av_hdl_a2d_evt evt 0
[D][BluetoothA2DPSink.cpp:352] av_hdl_a2d_evt(): av_hdl_a2d_evt ESP_A2D_CONNECTION_STATE_EVT
[I][BluetoothA2DPSink.cpp:356] av_hdl_a2d_evt(): A2DP connection state: Connected, [00:57:c1:43:a8:5b]
[I][BluetoothA2DPSink.cpp:374] av_hdl_a2d_evt(): ESP_A2D_CONNECTION_STATE_CONNECTED
**_[D][BluetoothA2DPSink.cpp:693] app_a2d_callback_2(): app_a2d_callback_2  (<-here app_rc_ct_callback() never is call)_**
[D][BluetoothA2DPSink.cpp:545] app_a2d_callback(): app_a2d_callback
[D][BluetoothA2DPSink.cpp:552] app_a2d_callback(): app_a2d_callback ESP_A2D_AUDIO_STATE_EVT
[D][BluetoothA2DPSink.cpp:203] app_work_dispatch(): app_work_dispatch event 0x1, param len 16
[D][BluetoothA2DPSink.cpp:235] app_send_msg(): app_send_msg
[D][BluetoothA2DPSink.cpp:254] app_task_handler(): app_task_handler, sig 0x1, 0x1
[W][BluetoothA2DPSink.cpp:257] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[D][BluetoothA2DPSink.cpp:226] app_work_dispatched(): app_work_dispatched
[D][BluetoothA2DPSink.cpp:539] operator()(): operator()
[D][BluetoothA2DPSink.cpp:348] av_hdl_a2d_evt(): av_hdl_a2d_evt evt 1
[D][BluetoothA2DPSink.cpp:386] av_hdl_a2d_evt(): av_hdl_a2d_evt ESP_A2D_AUDIO_STATE_EVT
[I][BluetoothA2DPSink.cpp:388] av_hdl_a2d_evt(): A2DP audio state: Started
[D][BluetoothA2DPSink.cpp:571] audio_data_callback(): audio_data_callback
[D][BluetoothA2DPSink.cpp:571] audio_data_callback(): audio_data_callback
**END**

Thank you pschatzmann for great jog!

pschatzmann, 2021-04-14:
Hmm - why are you using such an old version ? Arduino is currently at release 1.8.13.
But I think more important: which ESP32 Board Release are you using. The current release is 1.0.6

sidney-ikeda, 2021-04-15:
I'm using Arduino release 1.8.13. 
But when I update the Arduino Framework (from Boards Manager) from 1.0.4 to 1.0.5 or above occur this issue.

[image]

For you the support for Metadata and AVRC commands is working in 1.0.6 version?
Very strange. Look like AVRC features was disabled in 1.0.5, 1.0.6 version...

pschatzmann, 2021-04-15:
I confirm that it not working in any release after 1.0.4.
I tried to extend the error handling to make sure that I don't oversee any failed calls, but so far I havn't found any clue what happened...

pschatzmann, 2021-04-15:
Can you double check that the latest correction is working for you as well now ?

sidney-ikeda, 2021-04-15:
Great! Now Metadata and AVRC commands are also working in ESP32 1.0.6 release!
Thank you very much!

closed 2021-04-16 by pschatzmann

Uspizig, 2021-08-28:
Same occurs again on ESP32 1.0.6 release and Arduino 1.8.15
Tried first to update from ESP32 1.0.4 (where it was not working) to ESP32 1.0.6 with Arduino 1.8.13
Now updated to ESP32 1.0.6 with Arduino 1.8.15 where it is as well not supported.
I am somehow stuck.
[avrcpTest.txt](https://github.com/pschatzmann/ESP32-A2DP/files/7071131/avrcpTest.txt)

pschatzmann, 2021-08-28:
Please open a new issue and describe what you try to do.