pschatzmann/ESP32-A2DP #36: Metadata and controls examples

Tarkhrodd, 2021-04-15:
Hello everyone,
I'm  new on the esp32 and his code so i can't understand much...
So i see that functions to control the device connected have been added but I can't see how to use them.
I also see that we can obtain metadata but i don't know how to get them.
Sorry to understand nothing :/ and thank you for your help.

pschatzmann, 2021-04-15:
I extended the example and added some comments.

Let's assume you use your mobile phone as bluetooth source. If it is sending metadata the avrc_metadata_callback is called with an id and a text. In the example we just print the received information and finally it is then up to you to decide what you really want to do with this information.  To be honest I don't know which applications are providing any metadata since I have never used this functionality myself.

If your phone supports AVRC you can control it from the ESP32: eg by sending a pause which stops the phone from playing or
play which resumes the playing....

closed 2021-04-15 by pschatzmann

Tarkhrodd, 2021-04-16:
Thanks for the exemple but when I compile, it exit with this error : 
'class BluetoothA2DPSink' has no member named 'play'
Any suggestions ?

pschatzmann, 2021-04-16:
I guess you still have an old version of the library installed. If you installed it with git clone you can do a git pull…