pschatzmann/ESP32-A2DP #37: Trying to stop the Bluetooth Broadcast

sabbao, 2021-04-15:
Hello there! 

I am trying to call "a2dp_sink.app_task_shut_down()" on the vanilla example and it seems like its an protected call.

The goal here is to make it stop at some point and NOT broadcast any BT sink so my I2S receiver would be available to do another things. Should I try to unprotect it? 

I always feel like its a bummer and an inconvenience to bother people like you with this issues, but I will be very glad to whom (or the author) could help me on this (I will name my cat after you).

pschatzmann, 2021-04-16:
No worries...

If the intention is to stop A2DP completely I would suggest the following:
create the object with new:
```
BluetoothA2DPSink *a2dp_sink_ptr = new BluetoothA2DPSink();
a2dp_sink_ptr->start("MyMusic");  

```
and when you're done remove it with delete:
```
delete a2dp_sink_ptr;
```
This should stop the A2DP completely __and releases all resources__.

If you just want to stop - with the intention to resume and you have enough memory for your other activities you can just call 
stop() and later coninue with play().

sabbao, 2021-04-26:
Hello There my friend.

I have tried your method and on a very simple sketch I am getting some crashes:

```

/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:719 (xQueueGenericSend)- assert failed!
abort() was called at PC 0x4008fa06 on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x4008eddc:0x3ffcd290 0x4008f059:0x3ffcd2b0 0x4008fa06:0x3ffcd2d0 0x400d15cf:0x3ffcd310 0x400d158e:0x3ffcd330 0x400d1963:0x3ffcd360 0x400d148a:0x3ffcd380 0x400eb841:0x3ffcd3a0 0x400ec204:0x3ffcd3d0 0x40178772:0x3ffcd3f0 0x400ec4e9:0x3ffcd410 0x400e3976:0x3ffcd430 0x4009010e:0x3ffcd460

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4

```

I have tried 3 different ways: 
1) delete a2dp_sink_ptr; and got the crash above

2) with a2dp_sink_ptr->stop(); (my boy esp32 keeps broadcasting your marvelous BT sink).

3) btStop(); (This works pretty well, but I can't seem to manage to restart the BT).

My sketch used for this test:

```

#include "BluetoothA2DPSink.h"

BluetoothA2DPSink *a2dp_sink_ptr = new BluetoothA2DPSink();

void TurnMeOn()
{
  Serial.println("Im turning my self on now baby, hang on..."); 
  a2dp_sink_ptr->start("ImOnlineForNow"); 
}

void StopBT()
{
  Serial.println("Meh, not feeling like turning off now..."); 
  a2dp_sink_ptr->stop(); 
}

void HardcoreOFF()
{
  delete a2dp_sink_ptr;
}

void VanillaBTSTOP()
{
  Serial.println("Ohhh, you want me to stop??? Have it then..."); 
  btStop();
}

void setup() {
  Serial.begin(115200);  
  TurnMeOn(); // This works fine like an swiss clock
  delay(8000); // Lets check my phone for connectivity
  StopBT(); // This wont do anything =( bad boy
  delay(8000); // Lets check my phone again and get frustrated.
  HardcoreOFF(); // This boy works 10% of the time without crashing
  delay(8000); 
  VanillaBTSTOP();
  Serial.println("BT ReStart"); 
}

void loop() {
}

```

Im creating an bedside ESP32 internet radio / bluetooth to an old folk retirement place which its location doesn't get any Radio broadcast waves, so their favorite Radios are out of range. Switching the Bluetooth ON and OFF is pretty important so their personal devices would work automatically.

Any help would bring me joy!

Thank you for the library and the past help!

pschatzmann, 2021-04-26:
I think it might help if you set the log level to Debug or Verbose - and then you might see in the log which step in the destructor might be causing the issue.

I also recommend that you use the Exception decoder which will translate the backtrace into something meaningful...

pschatzmann, 2021-06-20:
merged correction provided by alexus2033

closed 2021-06-20 by pschatzmann