pschatzmann/ESP32-A2DP #38: Preparing simple c array for bt_music_sender_write

rmond18, 2021-04-16:
Hi, I'm trying to run the bt_music_sender_write, however I got an error:

> "Sketch` uses 2230311 bytes (170%) of program storage space. Maximum is 1310720 bytes."

I tried to downsize the sample StarWars.h file, but I don't know how to do it right. Instead, I tried following your instructions on how to prepare my own simple c array, but I don't know how/where this step should be done:

> - Convert to c file e.g. with "xxd -i file_example_WAV_1MG.raw file_example_WAV_1MG.c"

How should I convert the raw file to c file? (sorry for the dumb question)
Also, how do you get the length just like what is written in the end of StarWars.h file?

Thank you very much!

Regards,
rmond18

pschatzmann, 2021-04-16:
If I remember right you need to choose Partition Scheme - Huge App

rmond18, 2021-04-16:
It worked, thank you very much!

May I also asked what is the difference of bt_music_sender_write to the bt_music_sender_mic?
Does the bt_music_sender_mic only plots/prints the data from the mic into the serial monitor? 

Thank you very much for replying!

pschatzmann, 2021-04-16:
Hmm - I am confused.
Where did you find bt_music_sender_mic ?

rmond18, 2021-04-16:
Ohhh i'm sorry I got it mixed with the code that I'm developing where the input of the bt_sender is an i2s mic, thank you again! Sorry for the confusion.

closed 2021-04-16 by pschatzmann