pschatzmann/ESP32-A2DP #40: Add function to receive connection status

rbuehlma, 2021-04-18:
Is it possible to receive a connection status of the sink may be even with the name of the connected device?

Reutertu3, 2021-04-20:
Sure you can receive the connection status by adding lines very analogous to how the audio_state is determined.

**BluetoothA2DPSink.h**
Addition to public members:
```cpp
 /// Determine the actual connection state
 virtual esp_a2d_connection_state_t get_connection_state();
```

Addition to protected members:
```cpp
esp_a2d_connection_state_t connection_state;
```

**BluetoothA2DPSink.cpp**
Function addition
```cpp
esp_a2d_connection_state_t BluetoothA2DPSink::get_connection_state() {

  return connection_state;
}
```

Addition to case switch in function 
```cpp
void  BluetoothA2DPSink::app_a2d_callback(esp_a2d_cb_event_t event, esp_a2d_cb_param_t *param)
``` 

```cpp
    case ESP_A2D_CONNECTION_STATE_EVT:
        ESP_LOGD(BT_AV_TAG, "%s ESP_A2D_CONNECTION_STATE_EVT", __func__);
        connection_state = param->conn_stat.state;
        app_work_dispatch(av_hdl_a2d_evt_2, event, param, sizeof(esp_a2d_cb_param_t));
        break;
```

rbuehlma, 2021-04-21:
Thanks. That does work perfectly fine.

pschatzmann, 2021-04-21:
I agree - It's on my to-to list...

pschatzmann, 2021-04-27:
I added the method get_connection_state();

closed 2021-04-27 by pschatzmann