pschatzmann/ESP32-A2DP #41: Volume state & control

rbuehlma, 2021-04-21:
When playing musing from my (Android) mobile, I'm able change the volume on the mobile. It would be great, if this was also possible on the ESP (as a sink).

I tried to implement a volume control by adding these two functions:

`void BluetoothA2DPSink::volumeUp(){
    executeAVRCCommand(ESP_AVRC_PT_CMD_VOL_UP);
}

void BluetoothA2DPSink::volumeDown(){
    executeAVRCCommand(ESP_AVRC_PT_CMD_VOL_DOWN);
}
`
Unfortunately, that does not work.

Is it possible to control and may be even receive the current volume?

pschatzmann, 2021-04-21:
Did you try esp_avrc_ct_send_set_absolute_volume_cmd(uint8_t tl, uint8_t volume)
as described in the docu https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/esp_avrc.html

rbuehlma, 2021-04-21:
Yes, I tried that as well using this code to set the volume to 0:

`esp_avrc_ct_send_set_absolute_volume_cmd(1, 0);`

Unfortunately, that did not work either.

JonRobo, 2021-06-02:
The reason is that esp_avrc_ct_send_set_absolute_volume_cmd(uint8_t tl, uint8_t volume) is only available in the ESP IDF 4.2.x distributions, Arduino's latest ESP32 package uses ESP IDF 3.3.5 which does not include this volume command...I also am interested in using pschatzmann's library with volume control on the ESP32 side. I'll keep digging

hotteshen, 2021-08-05:
I also tried `esp_avrc_ct_send_set_absolute_volume_cmd` on ESP-IDF v4.2.1 and compile success, but no effect when I clicking buttons. Tested both on Android and iPhone.

pschatzmann, 2021-08-05:
As a side note I just wanted to highlight that you could easily implement a "volume control" by applying a multiplier on the sound data in the stream reader callback that you set via set_stream_reader() e.g.
- multiplying by 0 is muting
- multiplying by 0.5 is giving a lower volume

hotteshen, 2021-08-05:
It seems like a good solution.
But I still couldn't implement it. I tried to multiply the volume factor to the sound data on somewhere of `set_stream_reader()` body.
The `callback` is probably called from ADF side, so I could not find the interception point.

```cpp
float volume = 0.5;

void BluetoothA2DPSink::set_stream_reader(void (*callBack)(const uint8_t*, uint32_t), bool is_i2s){
  // apply volume and get a new callback
  this->stream_reader = volumeAppliedCallBack;
  this->is_i2s_output = is_i2s;
}
```

pschatzmann, 2021-08-05:
You find some details in the readme.

// In the setup function:
a2dp_sink.set_stream_reader(read_data_stream);

// Then somewhere in your sketch:
void read_data_stream(const uint8_t *data, uint32_t length)
{
      int16_t *ptr16 = (int16_t*) data; 
      uint21_t len16 = length/2;
      for(int j=0;j<len16;j++){
           ptr16[j] = 0.5 * ptr16[j];
      }

}

hotteshen, 2021-08-05:
Perfectly worked.
Thank you!!!

alexandru-2016, 2021-08-10:
I have a version of the code where I integrated avrcp 1.4 absolute volume commands, so that the phone just sends the volume level and raw data, and esp32 is the only one that does the volume adjustment. I will send a pull request soon once I have the code cleaned up. With this it would also be possible for the phone to be notified when the volume changes.

tv4you2016, 2021-08-12:
https://github.com/tv4you2016/ESP32-A2DP

hotteshen, 2021-08-12:
> https://github.com/tv4you2016/ESP32-A2DP

Nice.
How about merge your work to this repository via "pull request"?

tv4you2016, 2021-08-12:
Hello @pschatzmann 

I want to thank you in advance for the excellent work in the dev of this repo.
I would like to know if you are interested in doing a pull with the following changes:

- add volume control

- add set volume control (setVolume(uint8_t volume);)

- add get volume control (getVolume())

- Remove BluetoothA2DPSource

- add i2s mclk (i2s_mclk_pin_select(const uint8_t pin);)

- add callback set_on_connected2BT

- add callback set_on_disconnected2BT

- add PIN CODE ESP_BT_IO_CAP_IN;

pschatzmann, 2021-08-13:
Absolutely, I think this is very interesting...

pschatzmann, 2021-08-16:
the additional functionality from tv4you2016 has been merged into the master branch

rbuehlma, 2021-08-23:
> I have a version of the code where I integrated avrcp 1.4 absolute volume commands, so that the phone just sends the volume level and raw data, and esp32 is the only one that does the volume [...]

@alexandru-2016 I assume this is the PR that has been merged about 7 days ago? I tried to use the get_volume function but it always returns 0 no matter which volume I have on the phone. Changing the volume using set_volume does actually work but the volume control seems to be independent of the one on the phone. Is this expected?

pschatzmann, 2021-08-24:
Just to double check: did you use it with the esp 2.0.0-rc2 version ?

tv4you2016, 2021-08-24:
> Just to double check: did you use it with the esp 2.0.0-rc2 version ?

this question is for me? @pschatzmann

rbuehlma, 2021-08-24:
This question was probably for me:

I'm using this commit https://github.com/espressif/arduino-esp32/commit/780588dce3d975e4a693cf4c596bd3cbf5a69aaa because the following commit breaks the build in Arduino. I will try to build with 2.0.0-rc2 and give you an update.

rbuehlma, 2021-08-24:
I now tried the current master of espressif (https://github.com/espressif/arduino-esp32/commit/1f59c5abecc17fd450a5806362d199c3aa4f6c25) and the volume control still seems to be completely independent to the one on my android device. If I set the volume to 0 on the source or sink, I cannot increase it on the other device.

rbuehlma, 2021-09-12:
This has been fixed for me with the current expressif.

closed 2021-09-12 by rbuehlma

HammerShankiz, 2022-10-17:
I am having a similar issue. I'm simply running the example a2dp sink code on a ESP32 board. I can pair it with my iPhone just fine and it plays music, but I cannot adjust the volume at all. It blasts at 100% volume no matter what, and every ~5 seconds the phone gets a signal saying the volume has been adjusted to 1 setting above the minimum. Adjusting volume buttons on the phone has no affect, and if you mute the phone's volume music will only stop for a few seconds - until the next volume change update comes in and the device is back to blasting 100% volume and displaying 1 click above minimum volume. I'm trying to troubleshoot, but this is my first project working with a microcontroller so I'm pretty confused.

tonyxforce, 2023-12-21:
I seem to have the same issue, the get_volume function only returns the value set by set_volume and seems completely disconnected from my phone
i have tried
```
//...
void volumeChange(int newVolume){
  Serial.println(newVolume);
}
void setup(){
  //...
  a2dp_sink.set_avrc_rn_volumechange(volumeChange);
  a2dp_sink.set_avrc_rn_volumechange_completed(volumeChange);
  a2dp_sink.set_on_volumechange(volumeChange);
 a2dp_sink.start("MyMusic", true);
//...
}
void loop(){
  //...
  Serial.println(a2dp_sink.get_volume());
  a2dp_sink.set_volume(random(0, 127));
  delay(1000);
  //...
}
```
but nothing is logged into serial except the change every second