pschatzmann/ESP32-A2DP #44: How can I reconnect  after end()

biglee991228, 2021-05-08:
my test skedue as bellow:
#include "BluetoothA2DPSink.h"

BluetoothA2DPSink a2dp_sink;

void setup() {
//a2dp_sink.start("MyMusic");  
  Serial.begin(115200);
}

void loop() {
  if(Serial.available()){
  command = Serial.readStringUntil('#');
  if(command.equals("on")){
   a2dp_sink.start("MyMusic",true);
   Serial.println("STARTING...");
    }
  if(command.equals("off")){
   a2dp_sink.end(false);
   Serial.println("STOPING...");
    }
  }
} 

I could start the A2DP by start() function,and stopped by end() as well,but when I restarted A2DP by start(),my mobil could not connect "MyMusic" anymore unless a hard reset.
How can I restart A2DP link in loop() without any hard reset?
thanks a lot for any help!

pschatzmann, 2021-05-08:
Yes, I did not find the root cause for this yet...

alexus2033, 2021-06-10:
Hi, I have the same issue. It seems that it has nothing to do with end() statement.
If I connect and disconnect to the same device, streaming doesn´t work anymore. In the log, everything looks ok...

`12:30:57.808 -> [W][BluetoothA2DPSink.cpp:304] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
12:30:57.808 -> [I][BluetoothA2DPSink.cpp:406] av_hdl_a2d_evt(): A2DP connection state: Connecting, [f0:e4:a2:44:ea:c6]
12:30:57.808 -> [I][BluetoothA2DPSink.cpp:437] av_hdl_a2d_evt(): ESP_A2D_CONNECTION_STATE_CONNECTING
12:30:57.930 -> [W][BluetoothA2DPSink.cpp:304] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
12:30:57.930 -> [I][BluetoothA2DPSink.cpp:465] av_hdl_a2d_evt(): a2dp audio_cfg_cb , codec type 0
12:30:57.930 -> [I][BluetoothA2DPSink.cpp:477] av_hdl_a2d_evt(): a2dp audio_cfg_cb , sample_rate 44100
**12:30:57.930 -> [I][BluetoothA2DPSink.cpp:488] av_hdl_a2d_evt(): configure audio player 21-15-2-35**`
... but the last message doesn´t appear anymore! Only a reset helps :-(

pschatzmann, 2021-06-20:
Merged correction provided by alexus2033

closed 2021-06-20 by pschatzmann