pschatzmann/ESP32-A2DP #45: Using my own .h data

pablofr918, 2021-05-10:
Hello! I am trying to use my own .h in the bt_music_sender_write to send the audio, but I'm not sure how to convert to h. I tried the steps below, but it only plays noise. I tried another app that converts directly to .raw, and then a bin2h converter. It converts the audio and it actually looks it can work, because it has the same format, but then it only sounds noise. Could you help me to get my mp3 files into h? :)

pschatzmann, 2021-05-10:
I am afraid that this does not work with MP3. You will need to export raw PCM data. In Audacity this is with Export -> Audio -> Uncompressed Files  RAW (header-less).
I you want to use MP3 on the ESP32 - you would need to run it thru a decoder (which converts MP3 to PCM. e.g. with https://github.com/earlephilhower/ESP8266Audio

Here is an example with an SD file but you can adapt the code to read it form memory:

https://github.com/pschatzmann/arduino-audio-tools/blob/main/basic-api/base-file_mp3-a2dp/base-file_mp3-a2dp.ino

pablofr918, 2021-05-10:
I already saw the problem! It was it did not use two channels in the audacity app, but it is solved after I could upload with both channels! Thank you

closed 2021-05-10 by pablofr918

pablofr918, 2021-05-10:
But this code works sending mp3 file via bluetooth? This is exactly what  I needed, I'm going to try and tell you.
Thank you very much!

pablofr918, 2021-05-10:
It works!! I'm so gratfeul. But I have one question, is it possible to connect the BT device with start, and then send the data in the moment I want? it's because it calls the callback when it connects, but I don't know how to send the data if it is not in the callback.