pschatzmann/ESP32-A2DP #50: HFP Support

CDX-24, 2021-05-15:
Hello.
Great work, it works very smoothly with a PCM5102 board.
Would it be possible for the library to be compatible with the HFP protocol to hear phone calls from a smartphone? 
Thanks !

ariknel, 2021-05-16:
Hi CDX-24, i am also using the PCM5102  board but with this code i can't get any audio coming trough. How do you have the i2s pins hooked up? For me it is 

GPIO22 --> LRCK
GPIO25 --> DATA
GPIO26 --> BCK

Also wondering the same question you asked btw ;)

pschatzmann, 2021-05-16:
.bck_io_num = 26,
            .ws_io_num = 25,  = LRCK
            .data_out_num = 22,

ariknel, 2021-05-16:
Aha! That's why it is not working for me ;) thanks! Big love from Belgium. Without your code i would be stuck at one volume with the default epsressif code!

CDX-24, 2021-05-16:
Hi @ariknel ! I wired my module to the ESP differently by changing the pins in the library code.

Module |  ESP 32 Dev Module
SCK  =>   N/C
BCK  => GPIO4
DIN   => GPIO 2
LCK  => GPIO 15

and I changed the library code in BluetoothA2DPSink.cpp at line 55 to the following : 

`pin_config = {
            .bck_io_num = 4,
            .ws_io_num = 15,
            .data_out_num = 2,
            .data_in_num = I2S_PIN_NO_CHANGE
        };
`

I know that a method can be called to change the pins in the main code without needing to change the library but I was in a hurry ^^.

Also I had to solder several pads on my modules including the bridge between SCK to GND (if think it is GND, not sure though 😶 ).
Take a look at the picture I attached ;) 
[image]
[image]

ariknel, 2021-05-16:
Okay thanks! I just had sck, DEMP, FLT, FMT tied to ground!

closed 2021-06-11 by pschatzmann