pschatzmann/ESP32-A2DP #51: Error compiling

AntonioBanditoz, 2021-05-22:
Hey! I faced such a problem, when I tried to compile the project, I get an error control reaches end of non-void function, also Error compiling for board ESP32 Dev Module. Please help in this matter. Thank you.

pschatzmann, 2021-05-23:
Unfortunately I could not reproduce your issue.
Which example is giving the error and if you check in the Arduino Boards Manager and select esp32: which version are you using ? 
The latest is 1.0.6

AntonioBanditoz, 2021-05-23:
The compilation error appears in any example and refers to library files. Yes, now it's the latest 1.0.6. I can download the error log
[image]

AntonioBanditoz, 2021-05-23:
At the moment all library files are in the same folder with the sketch.

Arduino: 1.8.15 (Mac OS X), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/antonegorov/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/antonegorov/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/antonegorov/Documents/Arduino/libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10815 -build-path /var/folders/9j/61twj27s4z3blr5y7hpsc8c80000gn/T/arduino_build_67832 -warnings=all -build-cache /var/folders/9j/61twj27s4z3blr5y7hpsc8c80000gn/T/arduino_cache_790706 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool_py.path=/Users/antonegorov/Library/Arduino15/packages/esp32/tools/esptool_py/3.0.0 -prefs=runtime.tools.esptool_py-3.0.0.path=/Users/antonegorov/Library/Arduino15/packages/esp32/tools/esptool_py/3.0.0 -prefs=runtime.tools.mkspiffs.path=/Users/antonegorov/Library/Arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=/Users/antonegorov/Library/Arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=/Users/antonegorov/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-97-gc752ad5-5.2.0.path=/Users/antonegorov/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0 -verbose /Users/antonegorov/Yandex.Disk-Banditozdeveloper.localized/RegisterMCU/ESP32-A2DP-master/examples/bt_music_receiver_simple/bt_music_receiver_simple.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/antonegorov/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/antonegorov/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/antonegorov/Documents/Arduino/libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10815 -build-path /var/folders/9j/61twj27s4z3blr5y7hpsc8c80000gn/T/arduino_build_67832 -warnings=all -build-cache /var/folders/9j/61twj27s4z3blr5y7hpsc8c80000gn/T/arduino_cache_790706 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool_py.path=/Users/antonegorov/Library/Arduino15/packages/esp32/tools/esptool_py/3.0.0 -prefs=runtime.tools.esptool_py-3.0.0.path=/Users/antonegorov/Library/Arduino15/packages/esp32/tools/esptool_py/3.0.0 -prefs=runtime.tools.mkspiffs.path=/Users/antonegorov/Library/Arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=/Users/antonegorov/Library/Arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=/Users/antonegorov/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-97-gc752ad5-5.2.0.path=/Users/antonegorov/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0 -verbose /Users/antonegorov/Yandex.Disk-Banditozdeveloper.localized/RegisterMCU/ESP32-A2DP-master/examples/bt_music_receiver_simple/bt_music_receiver_simple.ino
Using board 'esp32' from platform in folder: /Users/antonegorov/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6
Using core 'esp32' from platform in folder: /Users/antonegorov/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6
bash -c "[ ! -f \"/Users/antonegorov/Yandex.Disk-Banditozdeveloper.localized/RegisterMCU/ESP32-A2DP-master/examples/bt_music_receiver_simple\"/partitions.csv ] || cp -f \"/Users/antonegorov/Yandex.Disk-Banditozdeveloper.localized/RegisterMCU/ESP32-A2DP-master/examples/bt_music_receiver_simple\"/partitions.csv \"/var/folders/9j/61twj27s4z3blr5y7hpsc8c80000gn/T/arduino_build_67832\"/partitions.csv"
bash -c "[ -f \"/var/folders/9j/61twj27s4z3blr5y7hpsc8c80000gn/T/arduino_build_67832\"/partitions.csv ] || cp \"/Users/antonegorov/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6\"/tools/partitions/default.csv \"/var/folders/9j/61twj27s4z3blr5y7hpsc8c80000gn/T/arduino_build_67832\"/partitions.csv"
Detecting libraries used...
/Users/antonegorov/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0/bin/xtensa-esp32-elf-g++ -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX -I/Users/antonegorov/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/config -I/Users/antonegorov/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/app_trace -I/Users/antonegorov/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/app_update -I/Users/antonegorov/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/s [... cut ...]
/Users/antonegorov/Yandex.Disk-Banditozdeveloper.localized/RegisterMCU/ESP32-A2DP-master/examples/bt_music_receiver_simple/bt_music_receiver_simple.ino: In function 'void setup()':
/Users/antonegorov/Yandex.Disk-Banditozdeveloper.localized/RegisterMCU/ESP32-A2DP-master/examples/bt_music_receiver_simple/bt_music_receiver_simple.ino:24:28: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
   a2dp_sink.start("MyMusic");  
                            ^
exit status 1
control reaches end of non-void function [-Werror=return-type]

pschatzmann, 2021-05-23:
Oh, I see - you did not install the library correctly - it seems that it is not in the libraries folder. 

You can download the library as zip and call include Library -> zip library. Or you can git clone this project into the Arduino libraries folder e.g. with

cd  ~/Documents/Arduino/libraries
git clone pschatzmann/ESP32-A2DP.git

AntonioBanditoz, 2021-05-23:
now there is such a mistake

Arduino: 1.8.15 (Mac OS X), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/antonegorov/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/antonegorov/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/antonegorov/Documents/Arduino/libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10815 -build-path /var/folders/9j/61twj27s4z3blr5y7hpsc8c80000gn/T/arduino_build_298124 -warnings=all -build-cache /var/folders/9j/61twj27s4z3blr5y7hpsc8c80000gn/T/arduino_cache_790706 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=/Users/antonegorov/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-97-gc752ad5-5.2.0.path=/Users/antonegorov/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.mkspiffs.path=/Users/antonegorov/Library/Arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=/Users/antonegorov/Library/Arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.esptool_py.path=/Users/antonegorov/Library/Arduino15/packages/esp32/tools/esptool_py/3.0.0 -prefs=runtime.tools.esptool_py-3.0.0.path=/Users/antonegorov/Library/Arduino15/packages/esp32/tools/esptool_py/3.0.0 -verbose /Users/antonegorov/Documents/Arduino/libraries/ESP32-A2DP-master/examples/bt_music_receiver_simple/bt_music_receiver_simple.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/antonegorov/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/antonegorov/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/antonegorov/Documents/Arduino/libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10815 -build-path /var/folders/9j/61twj27s4z3blr5y7hpsc8c80000gn/T/arduino_build_298124 -warnings=all -build-cache /var/folders/9j/61twj27s4z3blr5y7hpsc8c80000gn/T/arduino_cache_790706 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=/Users/antonegorov/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-97-gc752ad5-5.2.0.path=/Users/antonegorov/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.mkspiffs.path=/Users/antonegorov/Library/Arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=/Users/antonegorov/Library/Arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.esptool_py.path=/Users/antonegorov/Library/Arduino15/packages/esp32/tools/esptool_py/3.0.0 -prefs=runtime.tools.esptool_py-3.0.0.path=/Users/antonegorov/Library/Arduino15/packages/esp32/tools/esptool_py/3.0.0 -verbose /Users/antonegorov/Documents/Arduino/libraries/ESP32-A2DP-master/examples/bt_music_receiver_simple/bt_music_receiver_simple.ino
Using board 'esp32' from platform in folder: /Users/antonegorov/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6
Using core 'esp32' from platform in folder: /Users/antonegorov/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6
bash -c "[ ! -f \"/Users/antonegorov/Documents/Arduino/libraries/ESP32-A2DP-master/examples/bt_music_receiver_simple\"/partitions.csv ] || cp -f \"/Users/antonegorov/Documents/Arduino/libraries/ESP32-A2DP-master/examples/bt_music_receiver_simple\"/partitions.csv \"/var/folders/9j/61twj27s4z3blr5y7hpsc8c80000gn/T/arduino_build_298124\"/partitions.csv"
bash -c "[ -f \"/var/folders/9j/61twj27s4z3blr5y7hpsc8c80000gn/T/arduino_build_298124\"/partitions.csv ] || cp \"/Users/antonegorov/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6\"/tools/partitions/default.csv \"/var/folders/9j/61twj27s4z3blr5y7hpsc8c80000gn/T/arduino_build_298124\"/partitions.csv"
Detecting libraries used...
/Users/antonegorov/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0/bin/xtensa-esp32-elf-g++ -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX -I/Users/antonegorov/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/config -I/Users/antonegorov/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/app_trace -I/Users/antonegorov/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/app_update -I/Users/antonegorov/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/asio -I/Users/antonegorov/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/bootloader_support -I/User [... cut ...]
/Users/antonegorov/Documents/Arduino/libraries/ESP32-A2DP-master/src/BluetoothA2DPSource.cpp:901:81: warning: suggest braces around empty body in an 'if' statement [-Wempty-body]
             ESP_LOGD(BT_APP_TAG, "=> len: %d / result_len: %d", len, result_len);
                                                                                 ^
cc1plus: some warnings being treated as errors
Using library ESP32-A2DP-master at version 1.0 in folder: /Users/antonegorov/Documents/Arduino/libraries/ESP32-A2DP-master 
exit status 1
Error compiling for board ESP32 Dev Module.

pschatzmann, 2021-05-23:
This is even stranger: You get __cc1plus: some warnings being treated as errors__
I don't even get any warnings in my environment!

AntonioBanditoz, 2021-05-23:
i have now even tried on another windows computer, same error

pschatzmann, 2021-05-23:
Can you check the following: Arduino->Preferences
Compiler Warnings: None or Default

I was not aware of this setting and I will try to fix the warnings in the next couple of days....

AntonioBanditoz, 2021-05-23:
excellent! compilation was successful!

closed 2021-06-11 by pschatzmann