pschatzmann/ESP32-A2DP #52: Humming sound with PAM8403/PAM8610

punj, 2021-05-27:
Hello,

I tried A2DP using PAM8403 and PAM8610 (Stereo Mini Class-D 3W+3W PAM8403 Audio Amplifier) to use esp32 Bluetooth to connect with mobile and play songs. But there is much humming sound in it. I tried adding capacitors onto PAM8403, but nothing work.

I am using an example code for now as now below,
```
void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  static const i2s_config_t i2s_config = {
    .mode = (i2s_mode_t) (I2S_MODE_MASTER | I2S_MODE_TX | I2S_MODE_DAC_BUILT_IN),
    .sample_rate = 44100, // corrected by info from bluetooth
    .bits_per_sample = (i2s_bits_per_sample_t) 16, /* the DAC module will only take the 8bits from MSB */
    .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,
    .communication_format = I2S_COMM_FORMAT_I2S_MSB,
    .intr_alloc_flags = 0, // default interrupt priority
    .dma_buf_count = 8,
    .dma_buf_len = 64,
    .use_apll = false
  };

  a2dp_sink.set_i2s_config(i2s_config);
  a2dp_sink.start("MyMusic");

}
```

Well, I am not sure if I can use a Class D Amplifier for the same purpose. 

If any advice or suggestion can help, I will be grateful to them.

TIA

Derek-K, 2021-06-18:
It could be a power issue, if you have access to a scope hook it up and check. And if you are using a breadboard, there could be cross talk as well.

BTW, the DAC inside ESP32 is only 8 bit, as far as I know, it is not going to be very 'Hi-Fi'. Use an external I2S DAC or just simply get a BT-Audio module if all you need is an A2DP sink.

punj, 2021-06-18:
> It could be a power issue, if you have access to a scope hook it up and check. And if you are using a breadboard, there could be cross talk as well.
>
> BTW, the DAC inside ESP32 is only 8 bit, as far [...]

I used MAX98357A but it has a limited 3W of Speaker capacity.

Is there a way to amplify capacity to 10W using MAX98357A?

TIA

pschatzmann, 2021-10-16:
I am happy that you could resolve the quality issue, but I am afraid that your last question is not within the scope of this library