pschatzmann/ESP32-A2DP #54: Expanding 16-bit PCM to 32-bit PCM

riraosan, 2021-05-31:
Hi. @pschatzmann. Thank you for your contribution.

A few years ago, I managed to modify the ESP-IDF a2dp_sink sample for an external DAC to get a very clean sound from the external DAC. 
I thought about applying the fix at that time to this library.
 And I forked your library and made some modifications.

I considered sending a pull request to your library, but couldn't figure out a good fix. 
Could you take a look at my fix? Please tell me even when you have time.

The modified part is BluetoothA2DPSink :: audio_data_callback ().
I used [i2s_write_expand ()](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2s.html#_CPPv416i2s_write_expand10i2s_port_tPKv6size_t6size_t6size_tP6size_t10TickType_t) to extend the input of 16-bit sampling to 32-bit sampling. Replaced i2s_write () with i2s_write_expand ().

However, I didn't understand the relationship between the source below and i2s_write ().
https://github.com/riraosan/ESP32-A2DP/blob/f93979c5e5212bd46fc9c806957d79f49f1f9384/src/BluetoothA2DPSink.cpp#L687-L698
This is [the i2s setting part](https://github.com/riraosan/ESP32_BT-A2DP_Receiver/blob/d86036ca8de77b6b107ec9fe5de2d39384e7498c/src/main.cpp#L61-L74
) of the application class. I have confirmed that this application can output sound from an external DAC.

Under what conditions should I write code that uses i2s_write () and i2s_write_expand () properly?

riraosan, 2021-05-31:
Is it possible to replace all i2s_write () with i2s_write_expand ()?

pschatzmann, 2021-05-31:
A2DP will in any case only provide 16 bit data, so I doubt that expanding it to 32 bit will have any impact on the sound quality. I guess this only makes sense if your DAC can only support 32 bit - but so far I haven't seen such a thing.

If you want to go ahead with this, my suggestion would be that you make it dependent on i2s_config_t bits_per_sample: if this is set  > 16 you would call the  i2s_write_expand. I am not sure however if this is will work since I never tried this myself....

riraosan, 2021-05-31:
Hi. @pschatzmann 

>If you want to go ahead with this, my suggestion would be that you make it dependent on i2s_config_t bits_per_sample: if this is set > 16 you would call the i2s_write_expand. I am not sure however if this is will work since I never tried this myself....

I only have a DAC dedicated to 32bit samples, so I can't check with a 16bit sample DAC. I'll give it a try anyway, maybe I can check with ESP's internal DAC.
Thank you for your suggestion!👍

pschatzmann, 2021-05-31:
Can you maybe send me a link to the spec of your DAC ? 
32 bit dacs usually also support 16 and 24 bits...

riraosan, 2021-05-31:
ES9038Q2M VR1.07 DAC Board(eBay item number:263908779821)

This is my DAC.

https://www.ebay.com/itm/263908779821?_ul=JP

Input support:
Coaxial: PCM44.1K-192K 24BIT / DOP DSD64
Fiber: PCM44.1K-192K 24BIT / DOP DSD64
IIS / DSD: PCM44.1K-384K 32BIT / DSD64, DSD128, DSD256
IIS / DSD pin definition:

Maybe IIS input fixed 32BIT.🤔

riraosan, 2021-06-02:
Thank you for your contribution.👍
I closed this issue.👋

closed 2021-06-02 by riraosan

pschatzmann, 2021-06-02:
Did you finally manage to make it work with 32 bit?
If so, I would be interested in a pull request...

riraosan, 2021-06-03:
Hi. @pschatzmann

Don't get your hopes up, just wait and see.
My work has temporarily stopped, so I'm taking a vacation tomorrow (the 4th).
I'm going to stay home and do some serious work on your library for my DAC. 
I'm addicted to a certain song that was recently distributed. I'd like to use your library to listen to it on my DAC and my home audio system. 👍
If I can fix it, please review it on my pull request.👍

Thanx!

riraosan, 2021-06-05:
Hi. @pschatzmann 

I would like to send a pull request for this issue. Please review it. It would be nice to get some pointers.
I have done [a simple unit test](https://github.com/riraosan/ESP32_BT-A2DP_Receiver/blob/master/test/test_main.cpp) with PlatformIO.
As it turns out, only 32-bit samples work with my DAC.
I would like you to confirm if it works with [16-bit samples](https://github.com/riraosan/ESP32_BT-A2DP_Receiver/blob/5349f02ec88b2b5ea2b91bca3ba38da28cec3ba5/test/test_main.cpp#L47-L55) or not. 
Thanks!👍