pschatzmann/ESP32-A2DP #57: BT Sink to ESP32 internal DAC just hear static?

Derek-K, 2021-06-11:
I have been trying to sink to internal DAC using "bt_music_receiver_to_dac.ino" I was able to connect my phone to ESP32 via BT, but I only hear a buzzing sound. I am using a simple LM386 for AMP, and I testes the AMP works when I use a simple dacWrite() function. (I am only using G26 at the moment)

BTW, isn't DAC output for ESP32 are G25/G26 (not G26/G27)?

I also added the code from "bt_music_receiver_with_metadata.ino",  a2dp_sink.set_avrc_metadata_callback(avrc_metadata_callback) and the avrc_metadata_callback() callback routine to verify indeed my phone is sending data after connection.

here is the debug output (sorry, it is a bit long) all the way to endless "audio_data_callback(): audio_data_callback" and I can't seem to find out what's wrong.

```
12:19:43.695 -> [D][BluetoothA2DPSink.cpp:151] start(): start
12:19:43.742 -> [I][BluetoothA2DPSink.cpp:156] start(): Device name will be set to 'MyMusic'
12:19:43.742 -> [D][BluetoothA2DPSink.cpp:717] init_nvs(): init_nvs
12:19:43.742 -> [D][BluetoothA2DPSink.cpp:731] get_last_connection(): get_last_connection
12:19:43.742 -> [D][BluetoothA2DPSink.cpp:217] init_bluetooth(): init_bluetooth
12:19:43.833 -> [I][BluetoothA2DPSink.cpp:222] init_bluetooth(): controller initialized
12:19:44.157 -> [I][BluetoothA2DPSink.cpp:231] init_bluetooth(): bluedroid initialized
12:19:44.250 -> [I][BluetoothA2DPSink.cpp:239] init_bluetooth(): bluedroid enabled
12:19:44.250 -> [D][BluetoothA2DPSink.cpp:321] app_task_start_up(): app_task_start_up
12:19:44.250 -> [D][BluetoothA2DPSink.cpp:859] app_task_handler_2(): app_task_handler_2
12:19:44.250 -> [D][BluetoothA2DPSink.cpp:832] app_task_handler(): app_task_handler
12:19:44.250 -> [D][BluetoothA2DPSink.cpp:294] app_task_handler(): app_task_handler
12:19:44.250 -> [D][BluetoothA2DPSink.cpp:246] app_work_dispatch(): app_work_dispatch event 0x0, param len 0
12:19:44.298 -> [D][BluetoothA2DPSink.cpp:278] app_send_msg(): app_send_msg
12:19:44.298 -> [D][BluetoothA2DPSink.cpp:301] app_task_handler(): app_task_handler, sig 0x1, 0x0
12:19:44.298 -> [W][BluetoothA2DPSink.cpp:304] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
12:19:44.298 -> [D][BluetoothA2DPSink.cpp:269] app_work_dispatched(): app_work_dispatched
12:19:44.298 -> [D][BluetoothA2DPSink.cpp:173] operator()(): av_hdl_stack_evt_2
12:19:44.298 -> [D][BluetoothA2DPSink.cpp:577] av_hdl_stack_evt(): av_hdl_stack_evt evt 0
12:19:44.298 -> [D][BluetoothA2DPSink.cpp:582] av_hdl_stack_evt(): av_hdl_stack_evt av_hdl_stack_evt BT_APP_EVT_STACK_UP
12:19:44.345 -> [D][BluetoothA2DPSink.cpp:591] av_hdl_stack_evt(): AVRCP controller initialized!
12:19:44.345 -> [D][BluetoothA2DPSink.cpp:610] av_hdl_stack_evt(): connect_to_last_device
12:19:44.345 -> [D][BluetoothA2DPSink.cpp:785] connect_to_last_device(): connect_to_last_device
12:19:44.345 -> [D][BluetoothA2DPSink.cpp:615] av_hdl_stack_evt(): esp_bt_gap_set_scan_mode(ESP_BT_SCAN_MODE_CONNECTABLE_DISCOVERABLE)
12:19:44.345 -> [D][BluetoothA2DPSink.cpp:869] app_a2d_callback_2(): app_a2d_callback_2
12:19:44.345 -> [I][BluetoothA2DPSink.cpp:190] start(): Output will go to DAC pins
12:19:44.392 -> [D][BluetoothA2DPSink.cpp:844] app_a2d_callback(): app_a2d_callback
12:19:44.392 -> [D][BluetoothA2DPSink.cpp:632] app_a2d_callback(): app_a2d_callback
12:19:44.392 -> [D][BluetoothA2DPSink.cpp:644] app_a2d_callback(): app_a2d_callback ESP_A2D_CONNECTION_STATE_EVT
12:19:44.392 -> [D][BluetoothA2DPSink.cpp:246] app_work_dispatch(): app_work_dispatch event 0x0, param len 16
12:19:44.392 -> [D][BluetoothA2DPSink.cpp:278] app_send_msg(): app_send_msg
12:19:44.392 -> [D][BluetoothA2DPSink.cpp:301] app_task_handler(): app_task_handler, sig 0x1, 0x0
12:19:44.392 -> [W][BluetoothA2DPSink.cpp:304] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
12:19:44.439 -> [D][BluetoothA2DPSink.cpp:269] app_work_dispatched(): app_work_dispatched
12:19:44.439 -> [D][BluetoothA2DPSink.cpp:636] operator()(): av_hdl_a2d_evt_2
12:19:44.439 -> [D][BluetoothA2DPSink.cpp:397] av_hdl_a2d_evt(): av_hdl_a2d_evt evt 0
12:19:44.439 -> [D][BluetoothA2DPSink.cpp:401] av_hdl_a2d_evt(): av_hdl_a2d_evt ESP_A2D_CONNECTION_STATE_EVT
12:19:44.439 -> [I][BluetoothA2DPSink.cpp:406] av_hdl_a2d_evt(): A2DP connection state: Connecting, [xx:xx:xx:xx:xx:xx]
12:19:44.439 -> [I][BluetoothA2DPSink.cpp:437] av_hdl_a2d_evt(): ESP_A2D_CONNECTION_STATE_CONNECTING
12:19:45.370 -> Not connected
12:19:47.234 -> [D][BluetoothA2DPSink.cpp:869] app_a2d_callback_2(): app_a2d_callback_2
12:19:47.234 -> [D][BluetoothA2DPSink.cpp:844] app_a2d_callback(): app_a2d_callback
12:19:47.234 -> [D][BluetoothA2DPSink.cpp:632] app_a2d_callback(): app_a2d_callback
12:19:47.281 -> [D][BluetoothA2DPSink.cpp:653] app_a2d_callback(): app_a2d_callback ESP_A2D_AUDIO_CFG_EVT
12:19:47.281 -> [D][BluetoothA2DPSink.cpp:246] app_work_dispatch(): app_work_dispatch event 0x2, param len 16
12:19:47.281 -> [D][BluetoothA2DPSink.cpp:278] app_send_msg(): app_send_msg
12:19:47.281 -> [D][BluetoothA2DPSink.cpp:301] app_task_handler(): app_task_handler, sig 0x1, 0x2
12:19:47.281 -> [W][BluetoothA2DPSink.cpp:304] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
12:19:47.281 -> [D][BluetoothA2DPSink.cpp:269] app_work_dispatched(): app_work_dispatched
12:19:47.330 -> [D][BluetoothA2DPSink.cpp:636] operator()(): av_hdl_a2d_evt_2
12:19:47.330 -> [D][BluetoothA2DPSink.cpp:397] av_hdl_a2d_evt(): av_hdl_a2d_evt evt 2
12:19:47.330 -> [D][BluetoothA2DPSink.cpp:461] av_hdl_a2d_evt(): av_hdl_a2d_evt ESP_A2D_AUDIO_CFG_EVT
12:19:47.330 -> [I][BluetoothA2DPSink.cpp:465] av_hdl_a2d_evt(): a2dp audio_cfg_cb , codec type 0
12:19:47.330 -> [I][BluetoothA2DPSink.cpp:477] av_hdl_a2d_evt(): a2dp audio_cfg_cb , sample_rate 44100
12:19:47.330 -> [I][BluetoothA2DPSink.cpp:488] av_hdl_a2d_evt(): configure audio player 21-15-2-35
12:19:47.330 -> 
12:19:47.330 -> [I][BluetoothA2DPSink.cpp:489] av_hdl_a2d_evt(): audio player configured, samplerate=44100
12:19:47.612 -> [D][BluetoothA2DPSink.cpp:869] app_a2d_callback_2(): app_a2d_callback_2
12:19:47.612 -> [D][BluetoothA2DPSink.cpp:844] app_a2d_callback(): app_a2d_callback
12:19:47.612 -> [D][BluetoothA2DPSink.cpp:632] app_a2d_callback(): app_a2d_callback
12:19:47.612 -> [D][BluetoothA2DPSink.cpp:644] app_a2d_callback(): app_a2d_callback ESP_A2D_CONNECTION_STATE_EVT
[... 82 lines cut ...]
12:19:48.505 -> [D][BluetoothA2DPSink.cpp:301] app_task_handler(): app_task_handler, sig 0x1, 0x2
12:19:48.505 -> [W][BluetoothA2DPSink.cpp:304] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
12:19:48.505 -> [D][BluetoothA2DPSink.cpp:269] app_work_dispatched(): app_work_dispatched
12:19:48.505 -> [D][BluetoothA2DPSink.cpp:359] operator()(): av_hdl_avrc_evt_2
12:19:48.553 -> [D][BluetoothA2DPSink.cpp:532] av_hdl_avrc_evt(): av_hdl_avrc_evt evt 2
12:19:48.553 -> [I][BluetoothA2DPSink.cpp:550] av_hdl_avrc_evt(): AVRC metadata rsp: attribute id 0x20, 
12:19:53.678 -> [D][BluetoothA2DPSink.cpp:869] app_a2d_callback_2(): app_a2d_callback_2
12:19:53.678 -> [D][BluetoothA2DPSink.cpp:844] app_a2d_callback(): app_a2d_callback
12:19:53.678 -> [D][BluetoothA2DPSink.cpp:632] app_a2d_callback(): app_a2d_callback
12:19:53.678 -> [D][BluetoothA2DPSink.cpp:648] app_a2d_callback(): app_a2d_callback ESP_A2D_AUDIO_STATE_EVT
12:19:53.678 -> [D][BluetoothA2DPSink.cpp:246] app_work_dispatch(): app_work_dispatch event 0x1, param len 16
12:19:53.726 -> [D][BluetoothA2DPSink.cpp:278] app_send_msg(): app_send_msg
12:19:53.726 -> [D][BluetoothA2DPSink.cpp:301] app_task_handler(): app_task_handler, sig 0x1, 0x1
12:19:53.726 -> [W][BluetoothA2DPSink.cpp:304] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
12:19:53.726 -> [D][BluetoothA2DPSink.cpp:269] app_work_dispatched(): app_work_dispatched
12:19:53.726 -> [D][BluetoothA2DPSink.cpp:636] operator()(): av_hdl_a2d_evt_2
12:19:53.726 -> [D][BluetoothA2DPSink.cpp:397] av_hdl_a2d_evt(): av_hdl_a2d_evt evt 1
12:19:53.726 -> [D][BluetoothA2DPSink.cpp:444] av_hdl_a2d_evt(): av_hdl_a2d_evt ESP_A2D_AUDIO_STATE_EVT
12:19:53.772 -> [D][BluetoothA2DPSink.cpp:664] audio_data_callback(): audio_data_callback
12:19:53.772 -> [I][BluetoothA2DPSink.cpp:446] av_hdl_a2d_evt(): A2DP audio state: Started
12:19:53.772 -> [D][BluetoothA2DPSink.cpp:664] audio_data_callback(): audio_data_callback
12:19:53.772 -> [D][BluetoothA2DPSink.cpp:664] audio_data_callback(): audio_data_callback
12:19:53.772 -> [D][BluetoothA2DPSink.cpp:664] audio_data_callback(): audio_data_callback
12:19:53.772 -> [D][BluetoothA2DPSink.cpp:664] audio_data_callback(): audio_data_callback
```

pschatzmann, 2021-06-11:
I confirm, the output is to GPIO25 (Channel 1) and GPIO26 (Channel 2).
I suggest to test with some simple earphones first.
If you see 'audio_data_callback' in the log you should get audible audio on those pins....

Derek-K, 2021-06-18:
Confirmed audio output on GPIO25 & 26 and it is working as indented.

closed 2021-06-18 by Derek-K