pschatzmann/ESP32-A2DP #61: ESP32 internal DAC play music with noise

wamk9, 2021-06-24:
Hi Everyone!

I'm using a DEVKIT V1 (ESP32 WROOM 32) to test my project, but when i start a BT connection with my Android using the internal DAC of ESP32 i hear a lot of noise when music played (of course, if they is not playing too).

My question is, this "issue" is caused by internal DAC or my "hardware"? 

I'm using exemple codes only, using a P2 jack removed on a motherboard and connecting on GPIO25, GPIO26 and GND pin, my board are powered with a 5v~1A power supply on VIN/GND pin and nothing more, I'm not using any amplifier, only my headphone connected on P2...

SlowLogicBoy, 2021-10-12:
Having, same issue, searching for solutions, I think it's isolation problem, it picks up noise of data that's going through the board.
https://newbedev.com/esp32-adc-not-good-enough-for-audio-music

pschatzmann, 2021-10-13:
I just did some testing today with a cheap china DEVKIT V1 from DOIT and I confirm that I am getting some noise as well, so I guess this is a generic problem coming from the radiation of the cpu and the bluetooth emitter.

The most reliable way is to use an external DAC: https://github.com/pschatzmann/ESP32-A2DP/wiki/External-DAC

0neday, 2022-04-15:
when I connect power suppy using 3.3v of esp32 for AMP,  and stop play music, there is no Common Ground noise，but when I play music, there have noise again. 
I guess, the noise is produced by esp32 bluetooth emmiter. so there no good solution for remove noise when you use internel DAC.
Maybe externel DAC could fix this

pschatzmann, 2022-04-15:
Yes it will!

dcerisano, 2022-06-05:
Try a different ground pin on the ESP32 for your sound stage.
Some ground pins definitely give more distortion than others, especially on cheap knockoff ESP32s.

clarkbremer, 2022-06-06:
> Try a different ground pin (there are several) on the ESP32 for your sound stage.
> Some ground pins definitely give more distortion than others, especially on cheap knockoff ESP32s, especially with BLE initialized.
> 

Wait... There are cheap knockoff esp32s out there?  Do you mean devkit boards, or the esp32 modules themselves?

dcerisano, 2022-06-06:
> Wait... There are cheap knockoff esp32s out there? Do you mean devkit boards, or the esp32 modules themselves?

I mean the boards. The cheaper ones create RF distortion on their common grounds. This can be minimized by using a piezo speaker wired in H-Bridge configuration on pins 25 and 26 and using a dual-GPIO-pin AC driver rather than the usual single-pin DC drivers.

pschatzmann, 2022-12-09:
The general recommendation is to use the internal DAC only for testing or if noise is acceptable. Otherwise use a proper I2S DAC!

closed 2022-12-09 by pschatzmann

fa1ke5, 2022-12-27:
Большинство усилителей имеют дифференциальные входы IN+ и IN-, если подключить их оба к GPIO то шумы питания должны взаимно компенсироваться, но это только теория.

mrhempman69, 2023-11-15:
You guys got improved audio quality by placing capacitors like this? (In "series" as the image shows...)
[image]

fa1ke5, 2023-11-16:
Это устранит проблему лишь частично, правильное решение подключать источник звука к дифференциальным выходам DAC

mrhempman69, 2023-11-17:
Sorry, my chinese is a bit rusty... How much for 3 dumplings?

Anyway... It turns out the DAC is fine at higher volumes, because the noise is a problem when it's at lower voltages. I get pretty good results with just a 2-5kiloOhm resistor and the volume at 2/3s or higher.

 The resistors quiet the actual sound that is output through the speakers (so that I can play at high DAC voltage outputs without burning my eardrums, and without it being bothered by the antenna noise).

dcerisano, 2023-11-23:
Adding inline caps and resistors is effectively creating highpass filters to reduce the RF interference.

dcerisano, 2023-11-23:
Adding inline caps and resistors is effectively creating highpass filters to reduce the RF interference. This comment "just use a proper I2C DAC" is kind of missing the point. The idea is to minimize BoC by using the internal DAC.

Also the commentator above is writing in cryllic, not hànzì (it's Russian not Chinese) and contained zero dumpling content. See examplar below:

У этого парня фетиш на пельмени
这家伙有饺子癖

dcerisano, 2023-11-23:
Adding inline caps and resistors is effectively creating highpass filters to reduce the RF interference. This comment "just use a proper I2C DAC" is kind of missing the point. The idea is to minimize BoC by using the internal DAC.

Also the commentator above is writing in cryllic, not hànzì (it's Russian not Chinese) and contained zero dumpling content. See examplar below:

У Хемпмана фетиш на пельмени
亨普曼喜欢饺子

bobisaperson1, 2024-06-25:
餃子を4個注文してもいいですか？