pschatzmann/ESP32-A2DP #62: using external bluetooth

priyansh-shankhdhar, 2021-06-30:
I am trying to use WiFi as well as A2DP library, but not able to use Bluetooth simultaneously. Therefore I used FreeRTOS but still both are not working there with I have to suspend one task and if not connected with Wifi or Bluetooth then resume the other. Therefore I wish to use external Bluetooth module with this a2dp library, so how do I use this? How can I configure external Bluetooth with  this library.

pschatzmann, 2021-06-30:
I can confirm that you can not use both at the same time.

The A2DP Bluetooth functionality is built into the ESP32 - so you can not use it with an external Bluetooth module. 
I think you might be more successful to use an external WiFi module instead which is a more generic functionality.

I have never tried this myself but you could try to use 2 ESP32 and connect them over I2S. One would be used for the Wifi communication the other for Bluetooth...

priyansh-shankhdhar, 2021-07-01:
Thanks a lot @pschatzmann

pschatzmann, 2021-07-03:
see https://github.com/pschatzmann/ESP32-A2DP/issues/19

closed 2021-07-03 by pschatzmann