pschatzmann/ESP32-A2DP #64: Webradio combined with Bluetooth/A2DP: strange memory problem

siy1, 2021-07-08:
Hi there,

I started a small webradio project based on an ESP32 and PCM5102 I2S Stereo DAC using the
code provided here: https://github.com/schreibfaul1/ESP32-audioI2S.

This works well and now I plan to extend this to a Bluetooth A2DP sink device.

First I tried your library stand-alone, this worked out of the box, I had no problems at all to get it running (**congratulations for this nice work!**).

Then I added it to the existing webradio sketch, which immediately failed. Maybe this is connected to the fact that I2S resources are allocated by both methods?

Next try: when I want to switch from one method to the other, I set a flag in NVS flash ram and perform a soft restart of the ESP32. Then the code will start with pure WiFI/webradio mode or pure Bluetooth/A2DP mode. This worked, but I experienced a significant loss of heap memory in webradio mode (around 40 - 50 kB). And this memory is needed to buffer the webradio stream.

Next step: I did not define the BluetoothA2DPSink object globally, I only defined a pointer to it:

`BluetoothA2DPSink* a2dp_sink_ptr;`

If Bluetooth mode is requested, I call the constructur and perfom the initialization (in the setup() function):

```
...
a2dp_sink_ptr = new BluetoothA2DPSink;
a2dp_sink_ptr->set_pin_config(i2s_pin_config);
a2dp_sink_ptr->start(bluetooth_ap);
...
```

This works well, but still the same amount of memory is missing when Bluetooth is NOT used (constructor is not called). So the simple presence of the `a2dp_sink_ptr = new BluetoothA2DPSink´ statement consumes memory.

This is not what I expected - do you understand what is happening here and do you have an idea how to mitigate this problem?

Thanks in advance and thanks for this nice piece of software!

pschatzmann, 2021-07-08:
The end method has an optional parameter (release memory)!

This will release all BT memory (by calling esp_bt_controller_mem_release), at the cost that you can not restart bluetooth any more.  

The destructor is only calling end(false), because in most cases, the user expect Bluetooth to start again if needed...

siy1, 2021-07-08:
Thanks for the quick reply!
>     * A pointer only is using very few memory.-
this is clear (I'm aware of that - I'm programming C++ for more than 20 years ...)
> 
>     * As soon as you call new BluetoothA2DPSink() the system reservers memory for the object.
Also clear. My problem is that in the webradio mode I DO NOT call this constructor. But somewhere a larger amount of memory is allocated, leaving not enough memory for webradio operation.
>
>     * When you call start(), a lot of memory gets allocated for the whole Bluetooth stack.
>     * calling end() releases the memory related to the Bluetooth stack:
Both methods are not called in webradio mode...
> 
> 
> In regard of the release of memory BluetoothA2DPSink:
> The end method has an optional parameter (release memory)!
> This will release all BT memory (by calling esp_bt_controller_mem_release), at the cost that you can not restart bluetooth any more.
> 
> The destructor is only calling end(false), because in most cases, the user expects Bluetooth to start again if needed...

Some memory statistics (free heap memory / bytes, webradio playing a 128kB MP3 stream):
  BluetoothA2DPSink not enabled at all: 126496
  BT enabled, "new BluetoothA2DPSink()" available, but not called: 48276
  BT enabled, "new BluetoothA2DPSink()" called, but not used: 48144
  BT enabled, "new BluetoothA2DPSink()" commented out: 89544

For me this looks quite strange. Simply making the BluetoothA2DPSink() class available (not instantiated) consumes a significant amount of heap memory. Maybe some functions of the bluetooth stack already use memory, even when they are not used. Or is program memory moved from flash to heap memory?

pschatzmann, 2021-07-08:
Hmm interesting - how are you measuring ?

siy1, 2021-07-09:
Getting the free heap memory on the ESP32 (and maybe also on the ESP8266) is easy:
I simply call ESP.getFreeHeap() and log the obtained value to the serial console every couple of seconds.

You can find more useful info about the ESP class here:
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/Esp.h

I also use ESP.restart() to enforce a soft restart when I switch from WiFi to Bluetooth mode ...

pschatzmann, 2021-07-09:
Cool, that’s what I have expected.
The restart is what others have been doing as well.
I guess the best thing I can do from my side is to update the documentation and privide some detailed information about the memory consumption.

closed 2021-10-16 by pschatzmann