pschatzmann/ESP32-A2DP #65: Using a External DAC (ADS1115)

wamk9, 2021-07-12:
Hi everyone!

Some days i open a issue about a noise in internal DAC, after a read some articles i decide to buy a external DAC to test in my project.

Well... I bought a ADS1115, a 16bit DAC.

[image]

My question is, how i connect this DAC on my project? Because on pinout of ADS1115 i have SCL and SDA, but on code i have BCK, WS, Data IN and OUT.

Follow my code on setup():
```
static const i2s_config_t i2s_config = {
    .mode = (i2s_mode_t) (I2S_MODE_MASTER | I2S_MODE_TX | I2S_MODE_DAC_BUILT_IN),
    .sample_rate = 44100, // corrected by info from bluetooth
    .bits_per_sample = (i2s_bits_per_sample_t) 16, /* the DAC module will only take the 8bits from MSB */
    .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,
    .communication_format = I2S_COMM_FORMAT_I2S_MSB,
    .intr_alloc_flags = 0, // default interrupt priority
    .dma_buf_count = 8,
    .dma_buf_len = 64,
    .use_apll = false
  };

  i2s_pin_config_t my_pin_config = {
    .bck_io_num = 26,
    .ws_io_num = 25,
    .data_out_num = 22,
    .data_in_num = I2S_PIN_NO_CHANGE
  };
  
  a2dp_sink.set_pin_config(my_pin_config);
  a2dp_sink.set_i2s_config(i2s_config);

  a2dp_sink.set_avrc_metadata_callback(avrc_metadata_callback);
```

And another question is, i need init my ADS1115 with some library on this code or ESP32-A2DP do this?

pschatzmann, 2021-07-13:
I am sorry to disappoint you: this is not a DAC. 

This is a ADC - which is just the opposite: it transforms analog signals to a digital signal. 

You will need a I2S Audio DAC (like https://www.adafruit.com/product/3678)

wamk9, 2021-07-14:
> 
> 
> I am sorry to disappoint you: this is not a DAC.
> 
> This is a ADC - which is just the opposite: it transforms analog signals to a digital signal.
> 
> You will need a I2S Audio DAC (like https://www.adafruit.com/product/3678)

OHHHHHHHHHHH... hahaha

Ok, let's go buy a correct now!

wamk9, 2021-07-14:
Ok, buying a DAC (now a correct and not a ADC, like a [PCM5102A](https://pt.aliexpress.com/item/33001714060.html)) [using this issue](https://github.com/pschatzmann/ESP32-A2DP/issues/50) to connect on my ESP32, anything else i need know?

pschatzmann, 2021-07-14:
HFP is not supported by my library - just A2DP. 

This DAC is do-able but a little bit tricky because you need to take care of the following additional pins

- FMT - Audio format selection : I2S (Low) / Left justified (High)
- XSMT - Soft mute control(1): Soft mute (Low) / soft un-mute (High)

pschatzmann, 2021-07-16:
I added a Wiki: https://github.com/pschatzmann/ESP32-A2DP/wiki/External-DAC

wamk9, 2021-08-22:
Thanks for support @pschatzmann! I will stop my project for some days and i will follow this page on wiki to continue then!

closed 2021-08-22 by wamk9

CursingThomas, 2024-01-06:
I have the same "problem", though i believe mine really is a DAC. Any thoughts?
"DAC" in question: https://www.dfrobot.com/product-1721.html @pschatzmann

pschatzmann, 2024-01-06:
Though this is a DAC, this is not (really) an audio DAC, so you will need to write a driver which

- merges the stereo signal into mono
- drives your dac with I2C making sure that you update with the right frequency

While this should not be too difficult to implement this with my audio-tools library, it would be much easier
to get a proper i2s DAC