pschatzmann/ESP32-A2DP #66: Help with UDA1334A Breakout Board

abrad1212, 2021-07-13:
I have [this breakout board](https://www.adafruit.com/product/3678) from Adafruit and I'm having a problem getting any audio out from it.

I have spent hours researching what the problem could be but have found no solution. I have tested [this library](https://github.com/schreibfaul1/ESP32-audioI2S/) with the breakout board and had no problems getting it to play audio.

Any time audio is playing through Bluetooth and outputted through the 3.5mm jack the only thing that is present is static.

I have tried to change pretty much every parameter in the ```i2s_config```. I even went so far as copying the ```i2s_config``` from the previously mentioned library yet still no luck.

Has anyone else experience this issue? It making me go crazy!

```
#include "BluetoothA2DPSink.h"

BluetoothA2DPSink a2dp_sink;

void setup() {
    static const i2s_config_t i2s_config = {
         .mode = (i2s_mode_t) (I2S_MODE_MASTER | I2S_MODE_TX),
         .sample_rate = 41000,
         .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT,
         .channel_format = I2S_CHANNEL_FMT_ALL_LEFT,
         .communication_format = (i2s_comm_format_t)I2S_COMM_FORMAT_I2S_MSB,
         .intr_alloc_flags = 0, // default interrupt priority
         .dma_buf_count = 8,
         .dma_buf_len = 64,
    };

    i2s_pin_config_t my_pin_config = {
        .bck_io_num = 27,
        .ws_io_num = 26,
        .data_out_num = 25,
        .data_in_num = I2S_PIN_NO_CHANGE
    };

    a2dp_sink.set_i2s_config(i2s_config);
    a2dp_sink.set_pin_config(my_pin_config);
    a2dp_sink.start("MyMusic");

}

void loop() {
}
```

pschatzmann, 2021-07-13:
That's difficult. I had a similar issue with another DAC and needed to connect ground to the mute pin to make it work. But this will unlikely help, if you get sound from another library.

The most frequent issue are the pin connections but the pin assignments look right (identical with your other library)!

For the I2S config you should use
            .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,
            .communication_format = (i2s_comm_format_t) (I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_MSB),

When you connect do not forget, that the volume on your mobile phone is controlling the volume of the output, so to be sure just put it to the maximum.

You can try to compile your sketch with the setting "Core Debug Level" verbose. This might give a hint if anything in the processing goes wrong...

abrad1212, 2021-07-13:
> For the I2S config you should use
.channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,
.communication_format = (i2s_comm_format_t) (I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_MSB),

I will retry this when I get home. If that doesn't help I will upload pictures with my connections.

abrad1212, 2021-07-13:
@pschatzmann Just got home to test what you suggested and no luck not even with another DAC I have on hand [from adafruit](https://www.adafruit.com/product/3006) (it's actually a I2S DAC + Amp)

pschatzmann, 2021-07-14:
Just found the following: 

For what it's worth, if the issue is that you get more noise than sound, it could be the bit order. To get my PT8211 to work, I had to change I2S_COMM_FORMAT_I2S_MSB to I2S_COMM_FORMAT_I2S_LSB, here:

I think this is worth a trial....

abrad1212, 2021-07-14:
@pschatzmann I will try this when I get the time today. Thank you for the suggestion.

abrad1212, 2021-07-15:
@pschatzmann YES! This works!! I had to unplug and replug the connections and it started working with the ```I2S_COMM_FORMAT_I2S_LSB```

If anyone has this problem in the future here is my code
```
#include "BluetoothA2DPSink.h"

BluetoothA2DPSink a2dp_sink;

void setup() {
    static const i2s_config_t i2s_config = {
         .mode = (i2s_mode_t) (I2S_MODE_MASTER | I2S_MODE_TX),
         .sample_rate = 41000,
         .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT,
         .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,
         .communication_format = (i2s_comm_format_t) (I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_LSB),
         .intr_alloc_flags = 0, // default interrupt priority
         .dma_buf_count = 8,
         .dma_buf_len = 64,
         .use_apll = false
    };

    i2s_pin_config_t my_pin_config = {
        .bck_io_num = 27,
        .ws_io_num = 26,
        .data_out_num = 25,
        .data_in_num = I2S_PIN_NO_CHANGE
    };

    a2dp_sink.set_i2s_config(i2s_config);
    a2dp_sink.set_pin_config(my_pin_config);
    a2dp_sink.start("MyMusic");

}

void loop() {
}
```

closed 2021-07-15 by abrad1212