pschatzmann/ESP32-A2DP #68: Sample application crashes on Lolin32 Lite board.

stenzing, 2021-07-22:
I tried most of the sample codes from the libe but all of them are crashing at boot time, with the following dump visible in the logs and no pins atached to anything on the board:

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
GGuru Meditation Error: Core  0 panic'ed (Double exception)
Core 0 register dump:
PC      : 0x400803c0  PS      : 0x00060d36  A0      : 0x80094b7d  A1      : 0x3ffe3a50  
A2      : 0x000000f0  A3      : 0x01a1b5bd  A4      : 0x00000500  A5      : 0x000000c6  
A6      : 0x000000c0  A7      : 0x00400000  A8      : 0x00000000  A9      : 0x3ffe3b00  
A10     : 0x000000d3  A11     : 0x00000001  A12     : 0x00000001  A13     : 0x00040000  
A14     : 0x000a9638  A15     : 0x3f400000  SAR     : 0x0000000a  EXCCAUSE: 0x00000002  
EXCVADDR: 0x00040004  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x400803c0:0x3ffe3a50 0x40094b7a:0x3ffe3b30 0x40094398:0x3ffe3b50 0x4009444d:0x3ffe3b70 0x40154e96:0x3ffe3b90 0x40083d50:0x3ffe3bd0 0x4008401c:0x3ffe3c20 0x40079247:0x3ffe3c40 0x400792ad:0x3ffe3c70 0x400792b8:0x3ffe3ca0 0x40079465:0x3ffe3cc0 0x400806da:0x3ffe3df0 0x40007c31:0x3ffe3eb0 0x4000073d:0x3ffe3f20

Rebooting...

pschatzmann, 2021-07-22:
Which board did you select ? I suggest that you also try it with the plain vanilla „ESP32 Dev Module“ setting.
It might also help if you use the ESP32 Exception Decoder to locate the stack trace where the system is crashing.

Unfortunatly I dont have access to any Lolin32 Lite to help you identify the root cause…

stenzing, 2021-08-04:
Hi!

Sorry for the long response time, i had to order a set of new boards to cross check. It seems that the issue was with the board itself, and with a new board it worked straight away.

Sorry for the noise.

closed 2021-08-04 by stenzing