pschatzmann/ESP32-A2DP #70: Volume control feature?

hotteshen, 2021-08-04:
I designed and made an application board using ESP32 and PCM5012.
And it works fine with this ESP32-A2DP library.

Button controls for play/pause, next/prev works fine, too.
But volume up/down are not.
How to add volume up/down feature?

[image]
[image]

Here's code I tried but not worked, in `BluetoothA2DPSink.cpp` file (of course defined headers on `.h` file)

```c
void BluetoothA2DPSink::vol_up(){
    execute_avrc_command(ESP_AVRC_PT_CMD_VOL_UP);
}
void BluetoothA2DPSink::vol_down(){
    execute_avrc_command(ESP_AVRC_PT_CMD_VOL_DOWN);
}
```

I got compilation error on Arduino like,
```
'ESP_AVRC_PT_CMD_VOL_UP' was not declared in this scope
'ESP_AVRC_PT_CMD_VOL_DOWN' was not declared in this scope
```

pschatzmann, 2021-08-04:
The reason is that esp_avrc_ct_send_set_absolute_volume_cmd(uint8_t tl, uint8_t volume) is only available in the ESP IDF 4.2.x distributions, Arduino's latest ESP32 package uses ESP IDF 3.3.5 which does not include this volume command...I also am interested in using pschatzmann's library with volume control on the ESP32 side. I'll keep digging

see https://github.com/pschatzmann/ESP32-A2DP/issues/41

closed 2021-08-04 by pschatzmann

hotteshen, 2021-08-04:
Thank you for your quick feedback.
I will consider using ESP-IDF.

But still hoping that we could have this feature on Arduino.