pschatzmann/ESP32-A2DP #77: Metadata not received (iPhone/Apple Watch)

GefaketHD, 2021-08-29:
Hi,
I was wondering if there is something wrong on my side or if receiving metadata from iPhone or Apple Watch is not supported (yet)?
I set the log level to verbose on the BluetoothA2DPSink.cpp and I can see all output like so:
```
...
[D][BluetoothA2DPSink.cpp:914] audio_data_callback(): audio_data_callback
[D][BluetoothA2DPSink.cpp:914] audio_data_callback(): audio_data_callback
[D][BluetoothA2DPSink.cpp:1188] app_a2d_callback_2(): app_a2d_callback_2
[D][BluetoothA2DPSink.cpp:1157] app_a2d_callback(): app_a2d_callback
[D][BluetoothA2DPSink.cpp:873] app_a2d_callback(): app_a2d_callback
[D][BluetoothA2DPSink.cpp:889] app_a2d_callback(): app_a2d_callback ESP_A2D_AUDIO_STATE_EVT
[D][BluetoothA2DPSink.cpp:360] app_work_dispatch(): app_work_dispatch event 0x1, param len 16
[D][BluetoothA2DPSink.cpp:392] app_send_msg(): app_send_msg
[D][BluetoothA2DPSink.cpp:415] app_task_handler(): app_task_handler, sig 0x1, 0x1
[W][BluetoothA2DPSink.cpp:418] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[D][BluetoothA2DPSink.cpp:383] app_work_dispatched(): app_work_dispatched
[D][BluetoothA2DPSink.cpp:877] operator()(): av_hdl_a2d_evt_2
[D][BluetoothA2DPSink.cpp:562] av_hdl_a2d_evt(): av_hdl_a2d_evt evt 1
[D][BluetoothA2DPSink.cpp:630] av_hdl_a2d_evt(): av_hdl_a2d_evt ESP_A2D_AUDIO_STATE_EVT
[I][BluetoothA2DPSink.cpp:632] av_hdl_a2d_evt(): A2DP audio state: Suspended
...
```
So starting and stopping audio works fine and using the ESP to start or stop the music via code works as well, but if I change volume or title on my phone or watch, I don't see anything in the logs. I am relatively new to the whole BT-stuff and protocols and limitations etc. but I would like to contribute to this project, if possible and learn on my way.
Maybe someone could give me a hint, where the problem might be and what I should add or change to make it work in the future.

Thank you in advance!

pschatzmann, 2021-08-29:
You only get metadata if the IPhone Application is providing metadata. I think having seen metadata information from Spotify on a IPad.
As far as the volume control is concearned things are more complicated: please check the info in the wiki: https://github.com/pschatzmann/ESP32-A2DP/wiki/Controlling-your-Phone-with-AVRC-Commands

pschatzmann, 2021-08-31:
You are right, the AVRC Metadata for the ESP32 release < 2.0.0 was broken when I introduced the support for 2.0.0. I have committed the correction and you can try again.

podaen, 2021-09-16:
May I ask how you did that?

> I set the log level to verbose on the BluetoothA2DPSink.cpp and I can see all output like so:

GefaketHD, 2021-09-16:
> You are right, the AVRC Metadata for the ESP32 release < 2.0.0 was broken when I introduced the support for 2.0.0. I have committed the correction and you can try again.

It works, thank you very much.
Regarding volume control, would you be fine if I try to add support to this project? As I already said, I‘m a newbie but I would love to try. If it is even possible…

GefaketHD, 2021-09-16:
> May I ask how you did that?
> 
> > I set the log level to verbose on the BluetoothA2DPSink.cpp and I can see all output like so:

You add the following to the top of the BluetoothA2DPSink.cpp file:
```cpp
#ifdef CORE_DEBUG_LEVEL
#undef CORE_DEBUG_LEVEL
#endif 

#define CORE_DEBUG_LEVEL 3
#define LOG_LOCAL_LEVEL ESP_LOG_VERBOSE
```
At least that’s how I did it :)

pschatzmann, 2021-09-16:
You can define the log level in the Arduino IDE via -> Tools -> Core Debug Level. E.g. verbose

podaen, 2021-09-16:
@GefaketHD Thanks for your feedback.

@pschatzmann Sorry to tell you, but I am working on Visual Micro and it is a bit different. I looked on the internet and could not find information about it. I finally found the setting in menu->vmicro->option 3 -> core debug level.

closed 2021-10-16 by pschatzmann