pschatzmann/ESP32-A2DP #79: Metadata not recieved

Uspizig, 2021-08-30:
Hi,
I tried the ESP-A2DP example sketch with the following environment
Arduino 1.8.15. ESP32 Core: 1.0.6
ESP32-A2DP

So far I have no success with metadata from any device:
Tried the following devices
Huawei Mate 10 Android 11(AVRCP 1.6)
- Huawei Media Player
- Samsung Media Player (known that this one transmits AVRCP Data correct)
- Hifi-Cast Player (known for proper AVRCP standard)
- Deezer

Samsung A3 2016
- Samsung Media Player

Huawei Media M5 lite Tab 10.1
- Huawei Music Player 

None of the above showed any Metadata.
I can control volume, I can skip up and down

Tried as well Arduino IDE 1.8.13 with ESP32 Core 1.0.4-> same result
Do you have any idea what I am doing wrong?
Sketch:
[bt_music_receiver_to_dac.txt](https://github.com/pschatzmann/ESP32-A2DP/files/7079231/bt_music_receiver_to_dac.txt)

Logs:
[verboselogs.txt](https://github.com/pschatzmann/ESP32-A2DP/files/7079228/verboselogs.txt)

pschatzmann, 2021-08-31:
You are right, the AVRC Metadata for the release < 2.0.0 was broken when I introduced the support for 2.0.0. I have committed the correction and you can try again.

Uspizig, 2021-08-31:
Thanks for the fast help and rework of your code.
I tried again and it works as expected:

21:05:01.377 -> [W][BluetoothA2DPSink.cpp:410] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
21:05:01.446 -> [I][BluetoothA2DPSink.cpp:752] av_hdl_avrc_evt(): AVRC passthrough rsp: key_code 0x4b, key_state 0
21:05:01.533 -> [W][BluetoothA2DPSink.cpp:410] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
21:05:01.580 -> [I][BluetoothA2DPSink.cpp:752] av_hdl_avrc_evt(): AVRC passthrough rsp: key_code 0x4b, key_state 1
21:05:01.878 -> [W][BluetoothA2DPSink.cpp:410] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
21:05:01.978 -> [W][BluetoothA2DPSink.cpp:410] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
21:05:01.978 -> [I][BluetoothA2DPSink.cpp:756] av_hdl_avrc_evt(): AVRC metadata rsp: attribute id 0x1, Million Years Ago
21:05:01.978 -> [W][BluetoothA2DPSink.cpp:410] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
21:05:02.032 -> [I][BluetoothA2DPSink.cpp:756] av_hdl_avrc_evt(): AVRC metadata rsp: attribute id 0x2, Adele
21:05:02.032 -> [W][BluetoothA2DPSink.cpp:410] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
21:05:02.032 -> [I][BluetoothA2DPSink.cpp:756] av_hdl_avrc_evt(): AVRC metadata rsp: attribute id 0x4, 25
21:05:02.032 -> [W][BluetoothA2DPSink.cpp:410] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
21:05:02.032 -> [I][BluetoothA2DPSink.cpp:756] av_hdl_avrc_evt(): AVRC metadata rsp: attribute id 0x8, 9
21:05:02.032 -> [W][BluetoothA2DPSink.cpp:410] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
21:05:02.032 -> [I][BluetoothA2DPSink.cpp:756] av_hdl_avrc_evt(): AVRC metadata rsp: attribute id 0x10, 11
21:05:02.105 -> [W][BluetoothA2DPSink.cpp:410] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
21:05:02.105 -> [I][BluetoothA2DPSink.cpp:756] av_hdl_avrc_evt(): AVRC metadata rsp: attribute id 0x20, 
21:05:02.936 -> Connected Lautstärke 1: 70
21:05:05.951 -> Connected Lautstärke 1: 70

closed 2021-08-31 by Uspizig