pschatzmann/ESP32-A2DP #80: Raw audio to DAC from existing framework for event

jordancrubin, 2021-09-01:
Does there exist within the current framework the ability to Push raw Audio data to the DAC as a notification such as an instance where:

SoundData *connectbt = new OneChannelSoundData((int16_t*)bluetoothConn_raw, bluetoothConn_raw_len/2);

where bluetoothConn is a raw audio file saying 'BLUETOOTH CONNECTED'

called upon by 
void connect_bt_callback() {
   a2dp_sink.writeData(connectbt);
}

just like a2dp_source?

pschatzmann, 2021-09-01:
You can use i2s_write see https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2s.html

The logic might look like this:

SoundData *connectbt = new OneChannelSoundData((int16_t*) bluetoothConn_raw, bluetoothConn_raw_len/2);

void connect_bt_callback() {
  int pos=0;
  int len=512;
  size_t written;
  uint8_t buffer[len];
  int available = 0;
  do {
    available = connectbt->get2ChannelData(pos, len, buffer);
    i2s_write(I2S_NUM_0, buffer, available, &written, portMAX_DELAY);
    pos+=available;
  } while(available>0);
}

jordancrubin, 2021-09-02:
I have used the I2S library as recommended, thanks.   In conjunction with my raw data, when I connect to my phone, I can hear 'BLUETOOTH CONNECTED' clearly on the output, I can then play audio through the phone into the esp32 thereafter.  However should I disconnect bluetooth, I loose audio from both the raw audio and the phone when I reconnect.  Everything seems to run just no audio output.  I need to reboot the ESP to regain audio.   Code provided below

#include <Arduino.h>
#include "audiodata.h"
#include "BluetoothA2DPSink.h"
#include "driver/i2s.h"
const int muxPin = 19;

const unsigned int bluetoothConn_raw_len = 100352;
const i2s_port_t I2S_PORT = I2S_NUM_0;
const unsigned char *ConnWavFile=bluetoothConn_raw;
BluetoothA2DPSink a2dp_sink;
esp_a2d_connection_state_t last_state;
unsigned const char* AudioData;

void connect_bt_callback() {
  i2s_set_sample_rates(I2S_PORT, 22000);
   uint32_t index = 0; 
   AudioData = ConnWavFile;
   size_t BytesWritten;
   const unsigned char *Data;
   while (index < bluetoothConn_raw_len) {
    Data=AudioData+index;
    i2s_write(I2S_PORT,Data,4,&BytesWritten,portMAX_DELAY); 
    index+=4;
  }
  i2s_set_sample_rates(I2S_PORT, 44100);
}

void setup() {
  Serial.begin(115200);
  pinMode(muxPin, OUTPUT);
  a2dp_sink.set_mono_downmix(1);
  a2dp_sink.set_on_connected2BT(connect_bt_callback);
  //a2dp_sink.set_on_disconnected2BT(disconnect_bt_callback);
  a2dp_sink.start("JORDANS SAPPHIRE IX AUDIO TEST"); 
  // put your setup code here, to run once:
}

void loop() {}

pschatzmann, 2021-09-02:
Hard to say why this is but I suggest the following: 
- deactivate the autoconnect and connect with a2dp_sink.start("JORDANS SAPPHIRE IX AUDIO TEST", false);
- Set the log level to verbose so that you can analyse the log
- Try to call i2s_set_sample_rates(I2S_PORT, i2s_config.sample_rate);  instead of i2s_set_sample_rates(I2S_PORT, 44100); The sample rate is automatically negotiated and I am not sure if you end up with the correct value if you hardcode it to 44100

jordancrubin, 2021-09-02:
This was it!!!! Thank you !!!I did not try the suppression with the parameter.  I dont know WHY it breaks it, but it does.
**_deactivate the autoconnect and connect with a2dp_sink.start("JORDANS SAPPHIRE IX AUDIO TEST", false);_**

As for the below.... perhaps using the default config, im not presented with an i2s_config. to get sample_rate from, it returns an undef reference.

Try to call i2s_set_sample_rates(I2S_PORT, i2s_config.sample_rate); instead of i2s_set_sample_rates(I2S_PORT, 44100); The sample rate is automatically negotiated and I am not sure if you end up with the correct value if you hardcode it to 44100
Check the return code of i2s_set_sample_rates

pschatzmann, 2021-09-02:
good point: you can call sample_rate() which will return i2s_config.sample_rate...

jordancrubin, 2021-09-02:
A closer evaluation found that this actually started working when the debug level became 4 or higher, after testing I found a 3 or lower this issue started, why I cannot say.  It was with the recommendation for verbose output that actually cleared the issue.

build_flags = -DCORE_DEBUG_LEVEL=4

jordancrubin, 2021-09-03:
I had attempted to add delays around the raw audio code to see if that would help and it did not at debug 3 and below.   Eventually I came up with the convention below, dirty but effective restart on disconnect.  Note that the sample rates you see are experimentation and should be disregarded.  Ive further modified the write code to stop the clicking at the end of the audio.

`
void playAudio(bool type) {
  AudioData = ConnRawFile;
  int len = bluetoothConn_raw_len;
  if (type == 0) {
    //AudioData = DisconnWavFile;
    //len = bluetoothDisconn_raw_len;
    return;
  }
  i2s_set_sample_rates(I2S_PORT, 11025);
  uint32_t index = 0; 
  AudioData = ConnRawFile;
  size_t BytesWritten;
  const unsigned char *Data;
  int rest;
  int byteSize = 4;
  while (index < len) {
    rest = len - index;
    if (rest < 4){byteSize = rest;}
    Data=AudioData+index;
    i2s_write(I2S_PORT,Data,byteSize,&BytesWritten,portMAX_DELAY); 
    index+=4;
  }
  i2s_set_sample_rates(I2S_PORT, 44100);
}

void connect_bt_callback() {
  playAudio(1);
}
void disconnect_bt_callback() {
  playAudio(0);
  ESP.restart();
}
`

jordancrubin, 2021-09-03:
I was not so fortunate when adding the disconnected audio, because the audio runs before the reboot.  Same issue.  I isolated the line below that causes the issue,  again only on debug level 3 or below.  The rate change line.

#include <Arduino.h>
#include "audiodata2.h"
#include "BluetoothA2DPSink.h"
#include "driver/i2s.h"
const int muxPin = 19;
const unsigned int bluetoothConn_raw_len = 43600;
const unsigned int bluetoothDisconn_raw_len = 55730;

const i2s_port_t I2S_PORT = I2S_NUM_0;
const unsigned char *ConnRawFile=bluetoothConn_raw;
const unsigned char *DisconnRawFile=bluetoothDisconn_raw;
BluetoothA2DPSink a2dp_sink;
esp_a2d_connection_state_t last_state;
unsigned const char* AudioData;

void playAudio(bool type) {
  AudioData = ConnRawFile;
  int len = bluetoothConn_raw_len;
  if (type == 0) {
    AudioData = DisconnRawFile;
    len = bluetoothDisconn_raw_len;
  }
  ///////////////////////////////////if (type ==0){return;}
  i2s_set_sample_rates(I2S_PORT, 11025);
  ///////////////////////////////////if (type ==0){return;}
  uint32_t index = 0; 
  size_t BytesWritten;
  const unsigned char *Data;
  int rest;
  int byteSize = 4;
  while (index < len) {
    rest = len - index;
    if (rest < 4){byteSize = rest;}
    Data=AudioData+index;
    i2s_write(I2S_PORT,Data,byteSize,&BytesWritten,portMAX_DELAY); 
    index+=4;
  }
  i2s_set_sample_rates(I2S_PORT, 44100);
}

void connect_bt_callback() {
  playAudio(1);
}
void disconnect_bt_callback() {
  playAudio(0);
  ESP.restart();
}