pschatzmann/ESP32-A2DP #81: A2DP And DF Mini Player

podaen, 2021-09-05:
Hardware:
Board: ESP-WROOM-32
Core Installation/update date: 01/SEP/2021
IDE name: Arduino IDE

Dear,

I have a project with a "DF mini player" that plays music from a SD card and output in a 24bit DAC format on an amp. 

[Documentation](https://picaxe.com/docs/spe033.pdf)

I am interested in your code and I am able to run your example without no problem. Now I want to implement it to my project.

How do I stitch those two projects together with only one amp? Do I connect the dac output of the "DF mini player" to the ESP32? And how do I handle that audio format data?

Kind regards
Dave

podaen, 2021-09-07:
I came to the conclusion that an audio mixing cable would be the best solution.

pschatzmann, 2021-09-07:
Yes but is really depends on what you want to achieve: This board is pretty cool because it already provides all the audio functionality itself

podaen, 2021-09-07:
True, soo it would be not that smart to send it back to the esp32 and converted back to digital. At first my solution was another sd boards, but that diden't give some satisfying results because of the lack of mp3 . Soo the best thing for me is a mixing cable or even an adapter.

pschatzmann, 2021-09-07:
You could try to use I2S input using the built in ADC mode to sample the output from your board.

podaen, 2021-09-14:
I prefere also the I2S communication, but the board I am using has only uart and it only can be used to send commands.

closed 2021-09-15 by podaen