pschatzmann/ESP32-A2DP #82: Right Justified 16bit

albertomario, 2021-09-05:
Hello, 
I'm struggle to get this mode to work.
I've tried any configuration but none seems to work.

    i2s_config_t i2s_config = {
        .mode = (i2s_mode_t) (I2S_MODE_MASTER | I2S_MODE_TX),
        .sample_rate = 44100, // corrected by info from bluetooth
        .bits_per_sample = (i2s_bits_per_sample_t) 16, /* the DAC module will only take the 8bits from MSB */
        .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,
        .communication_format = (i2s_comm_format_t) I2S_COMM_FORMAT_I2S_MSB,
        .intr_alloc_flags = 0, // default interrupt priority
        .dma_buf_count = 8,
        .dma_buf_len = 64,
        .use_apll = false,
    };

In espressif documentation on i2s i see this line,
https://github.com/espressif/esp-idf/blob/0736c91d68a0f4eff59effcf3fa832eb762d5851/components/soc/esp32c3/include/soc/i2s_struct.h#L106

Also i got info about ZephyrProject how have an constant named I2S_FMT_DATA_FORMAT_RIGHT_JUSTIFIED, but need to be flashed on esp32 and is a lot of work to do.

Any ideea about how can i get 16bit right justified output?

Thank you very much!

pschatzmann, 2021-09-06:
Which External DAC  are you using ?
Are you sure that it is not covered by the Wiki

pschatzmann, 2021-09-07:
I just double checked your link which is only relevant for to the ESP32C3. This is a single core RISC-V processor which is not supported by this framework. 

I am only supporting regular ESP32 where the official documentation is https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2s.html

The use of external DACs is documented in the Wiki which can be found at https://github.com/pschatzmann/ESP32-A2DP/wiki/External-DAC

albertomario, 2021-09-08:
I am trying to get output for my car radio (Volvo)
I know the radio want as input 16 bit right justified, but esp32 looks to output only left justified.

pschatzmann, 2021-09-08:
I would have expected that the setting described in the Wiki for Adafruit boards should work

    static const i2s_config_t i2s_config = {
         .mode = (i2s_mode_t) (I2S_MODE_MASTER | I2S_MODE_TX),
         .sample_rate = 41000,
         .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT,
         .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,
         .communication_format = (i2s_comm_format_t) (I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_LSB),
         .intr_alloc_flags = 0, // default interrupt priority
         .dma_buf_count = 8,
         .dma_buf_len = 64,
         .use_apll = false
    };

closed 2021-10-16 by pschatzmann