pschatzmann/ESP32-A2DP #83: Audiostate delay on audio data stop

jordancrubin, 2021-09-07:
Given the code below.  Should a connected device go from pause to play,  audiostate will go seemingly immediately from 0 to 2.  However Should you press stop, even as the debug also shows audio has stopped, it takes about 5 seconds to go from an audiostate of 2 back to 0.   Is there a setting that can shorten this transition from 2 to 0?

void loop() {
  esp_a2d_audio_state_t audiostate = a2dp_sink.get_audio_state();
  if (audiostate == 2){
    Serial.print("We have AUDIO OUTPUT")
  }
  else {
    Serial.print("NO AUDIO")
  }
  delay(500);
}

pschatzmann, 2021-09-07:
Not that I am aware of. You could activate the log to see if you might see what events are received and change the code acordingly...

jordancrubin, 2021-09-07:
I've looked through the code and found nothing to this end. the debug of 5 indicates that The audio status continues from the phone after pause.  It would seem that Android is a few seconds shorter than IOS.  Iphone about 5 or more and Android around 3.  Please close this as beyond scope.

closed 2021-09-08 by pschatzmann