pschatzmann/ESP32-A2DP #84: Trying to connect 24-bit PCM5102 PCM5102A Stereo

podaen, 2021-09-07:
I am stuck on this. I don't understand how to connect the board below. On the side of the esp32 is clear, but on the external dac SD, L/R, WS can not be found on the board. Also the BCK is in the colume ESP32. Could you please check this again. 

[External Dac 24-bit PCM5102 PCM5102A Stereo](https://github.com/pschatzmann/ESP32-A2DP/wiki/External-DAC#24-bit-pcm5102-pcm5102a-stereo-dac-digital-to-analog-converter-pll-voice-module-phat)

pschatzmann, 2021-09-07:
I updated the documentation can you try it out and let me know the results ?

podaen, 2021-09-07:
Thanks, that should do it. Only one thing you named XMT as XST.

pschatzmann, 2021-09-07:
corrected as well. Thanks for your feed back....

podaen, 2021-09-07:
Can it be that L/R to GND is gone?

podaen, 2021-09-07:
I found this [connection diagram for a raspberry pi.](https://goteborgsaventyrscenter.se/product/mevjpym68n31/i2s-pcm5102-dac-decoder-32-bit-player-module-than-es9023-pcm1794-for-raspberry-pi)

As you can see the FTL,DMP,SCL and FMT are connected to ground.

pschatzmann, 2021-09-07:
If I remember right, out of these only FMT is critical. I guess the others are connected to ground to have a defined state.
For DMP and FLT you could play around if setting it to high improves the quality.
I was really struggeling with this module and only when I started to set XMT to high it started to work for me.

podaen, 2021-09-07:
I diden't test it yet. I let you know. I am trying to compile without disable the warnings. But without succes for now.

pschatzmann, 2021-09-07:
I hope the warnings are not coming from my library. 
If so are you using the latest version ?

podaen, 2021-09-07:
I downloaded them a while a go. The 4th of may to be precese. Are they addapted?

podaen, 2021-09-07:
oh yes, I see now. Thanks

pschatzmann, 2021-09-07:
You might even consider to upgrade the ESP32 release to 2.0.0

podaen, 2021-09-07:
What do you mean by upgrading the ESP32 release? Updating the boards of Arduino IDE?

The board works with you suggested wiring!  

I just get noise when there is no music playing, even when the ESP32 is not powered on... Soo I think it is coming from my bad power supply I am currently using. The wiring is not solderd yet and the two GND of the power supply are currently not connected to each other. I will try it with my lineare regulated power supply. Would it help by setting the XSMT to low... This is what the datasheet tells me: Soft mute control : Soft mute (Low) / soft un-mute (High).

podaen, 2021-09-08:
I did the following...

a2dp_sink.set_volume(0);

digitalWrite(PinMute,LOW);

Bound the grounds together

The noise remains. The only thing I can do as far as I know is place a filter around the power supply cable. Or use an other powersupply and solder the wires.

podaen, 2021-09-08:
Is the sound muted when it's not connected?

pschatzmann, 2021-09-08:
In my code, the disconnect event is switching off I2S. 
The only other setting related to noise that I am aware of is in  

            i2s_config.tx_desc_auto_clear = true // avoiding noise in case of data unavailability

podaen, 2021-09-15:
> You might even consider to upgrade the ESP32 release to 2.0.0

You where right, there was a new release.

I followed the instruction

[instruction](https://support.arduino.cc/hc/en-us/articles/360016119519-How-to-add-boards-in-the-board-manager)

closed 2021-09-15 by podaen