pschatzmann/ESP32-A2DP #85: This library does not compile successfully on ESP32 Arduino 2.0.0.

qnxsgwy, 2021-09-07:
's_avrc_peer_rn_cap' function has been removed in the ESP32 Arduino 2.0.0 release, and some functions have been deprecated, like: 'I2S_COMM_FORMAT_I2S_MSB'
The following error is reported when compiling

C:\Users\hnbd\Documents\Arduino\libraries\ESP32-A2DP-main\examples\bt_music_receiver_to_dac\bt_music_receiver_to_dac.ino:29:31: warning: 'I2S_COMM_FORMAT_I2S_MSB' is deprecated [-Wdeprecated-declarations]
       .communication_format = I2S_COMM_FORMAT_I2S_MSB,

C:\Users\hnbd\Documents\Arduino\libraries\ESP32-A2DP-main\src\BluetoothA2DPSink.cpp:744:13: error: 's_avrc_peer_rn_cap' was not declared in this scope
             s_avrc_peer_rn_cap.bits = 0;
             ^~~~~~~~~~~~~~~~~~
C:\Users\hnbd\Documents\Arduino\libraries\ESP32-A2DP-main\src\BluetoothA2DPSink.cpp:744:13: note: suggested alternative: 'esp_avrc_tg_send_rn_rsp'

ESP32 Arduino 2.0.0 has been officially released, please fix it as soon as possible. thanks!

pschatzmann, 2021-09-07:
Are you sure that you use the latest version of the library ?
I am supporting 2.0 since middle of August and I do not get any errors...

qnxsgwy, 2021-09-09:
This problem has been fixed in the version two days ago, and now it can be compiled successfully. Thank you

closed 2021-09-09 by qnxsgwy