pschatzmann/ESP32-A2DP #86: Is arvc_tg supported?

fmvoxasoft, 2021-09-09:
Hey,

Normally I want use the ESP32 as a  audio source for headphones. So I tried to copy the arcv_tg needed function into the BluetoothA2DPSource library. But I got in platform.io couple of `undefined reference` to warining and a `error: ld returned 1 exit status`.
```
.pio\build\esp32dev\libe3d\libESP32-A2DP.a(BluetoothA2DPSource.cpp.o):(.literal._ZN19BluetoothA2DPSource21bt_av_hdl_avrc_tg_evtEtPv+0x20): undefined reference to `esp_avrc_tg_send_rn_rsp'
.pio\build\esp32dev\libe3d\libESP32-A2DP.a(BluetoothA2DPSource.cpp.o):(.literal._ZN19BluetoothA2DPSource19bt_av_hdl_stack_evtEtPv+0x2c): undefined reference to `esp_avrc_tg_init'
.pio\build\esp32dev\libe3d\libESP32-A2DP.a(BluetoothA2DPSource.cpp.o):(.literal._ZN19BluetoothA2DPSource19bt_av_hdl_stack_evtEtPv+0x30): undefined reference to `esp_avrc_tg_register_callback'
.pio\build\esp32dev\libe3d\libESP32-A2DP.a(BluetoothA2DPSource.cpp.o): In function `BluetoothA2DPSource::bt_av_hdl_avrc_tg_evt(unsigned short, void*)':
C:\Users\fmein\Documents\PlatformIO\Projects\esp32_i2s_to_ad2p/.pio\libdeps\esp32dev\ESP32-A2DP\src/BluetoothA2DPSource.cpp:1193: undefined reference to `esp_avrc_tg_send_rn_rsp'
.pio\build\esp32dev\libe3d\libESP32-A2DP.a(BluetoothA2DPSource.cpp.o): In function `BluetoothA2DPSource::bt_av_hdl_stack_evt(unsigned short, void*)':
C:\Users\fvox\Documents\PlatformIO\Projects\esp32_i2s_to_ad2p/.pio\libdeps\esp32dev\ESP32-A2DP\src/BluetoothA2DPSource.cpp:1193: undefined reference to `esp_avrc_tg_init'
C:\Users\fvox\Documents\PlatformIO\Projects\esp32_i2s_to_ad2p/.pio\libdeps\esp32dev\ESP32-A2DP\src/BluetoothA2DPSource.cpp:1193: undefined reference to `esp_avrc_tg_register_callback'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32dev\firmware.elf] Error 1
```

So I tried the it with BluetoothA2DPSink library, but there I need the `ESP_IDF_VERSION_MAJOR = 4` when I change it to 4, I got a lot of missing stuff.

So it is not ready to receive AVRC Commands from a device? 

Greeting

pschatzmann, 2021-09-10:
I think this is rather an question/issue for platformIO how to use a more recent version of ESP-IDF.
Just a couple of days ago Arduino has upgraded to ESP-IDF 4.4.

fmvoxasoft, 2021-09-15:
> 
> 
> I think this is rather an question/issue for platformIO how to use a more recent version of ESP-IDF.
> Just a couple of days ago Arduino has upgraded to ESP-IDF 4.4.
> 
> https://github.com/pschatzmann/ESP32-A2DP/wiki/Controlling-your-Phone-with-AVRC-Commands

I checked this out and tried the newer version in platform.io, but got a lot of trouble again. Will watch this later. But tahnks for the hint.