pschatzmann/ESP32-A2DP #87: Convert the packages to an analoge signal

podaen, 2021-09-10:
I want to convert the digtale signal thats recieved from the packages to an analoge signal. What's the best way to do that?

podaen, 2021-09-10:
It needs to be done withing the ESP.

podaen, 2021-09-11:
My idee is to analyse the audio signal, not use it to produce a sound with it. 

If I use the internal dac... the digital audio has passed the converter chip. Soo if I connect the internal dac to a gpio pin I have my signal to work with. Right... Or are there other way's to do that?

Will the I2S signal still be produced as the internal dac is in use?

pschatzmann, 2021-09-11:
Have a look at the documentation: Accessing the Sink Data Stream with Callbacks

podaen, 2021-09-11:
Can not make it work...

```
void setup() {
	//virtual void set_stream_reader(void (*callBack)(const uint8_t*, uint32_t), bool i2s_output=true);
	a2dp_sink.set_stream_reader(set_stream_reader);
}
void set_stream_reader(const uint8_t* Metadata, uint32_t audiodata) {
	Serial.printf("==> AVRC streamreader rsp: attribute id 0x%x, %s\n", Metadata, audiodata);
}
```

I think this will be wrong

Serial.printf("==> AVRC streamreader rsp: attribute id 0x%x, %s\n", Metadata, audiodata);

podaen, 2021-09-11:
Or maybe I can use this. But I don't know how.

 // Callback for music stream 
    virtual void audio_data_callback(const uint8_t *data, uint32_t len);

podaen, 2021-09-11:
OK. This is what I found in the documentation

> _Or you can access the packet:
> 
> // In the setup function:
> a2dp_sink.set_stream_reader(read_data_stream);
> 
> // Then somewhere in your sketch:
> void read_data_stream(const uint8_t *data, uint32_t length)
> {
>   // Do something with the data packet
> }
> In the a2dp_sink.set_stream_reader() method you can provide an optional parameter that defines if you want the output to I2S to be active or deactive - So you can use this method to e.g. to switch off I2S just by calling a2dp_sink.set_stream_reader(read_data_stream, false)_

This is how I do it now.

n = analogRead(MIC_PIN);                        // Raw from mic 

I want to replace analogRead(MIC_PIN) whit the datastream I recieve. Do I need to pull wires from the dac to a gpio or can I do it softwarematically. Note: the dac pins are allready in use by the decoder board.

pschatzmann, 2021-09-11:
Maybe I was misunderstanding what you tried to do. I thought you wanted to analyze the data that you receive via bluetooth. The data that you receive via read_data_stream() is an array with 2 channels of int16_t data.
So you could do the following:
```
size_t count = length / 2; // number of int6_t
int16_t *values =  (int16_t) data;
// process the values....
// first left value is values[0] - // first right value is values[1]
// last left value is values[length-2] - // last right value is values[length-1]

```

podaen, 2021-09-12:
Sorry I was not explaining myself so clear, It is because it is all new for me. Now I recieve sinus wave values, but there are some issues.

- The values are positive and negative (I could solve that if I know what the most upper and lower values is)
- Long after the music stops, I still recieve values. For my application it has to function as the analogread that is corresponding in time.
- After a whill it crashes and reboots. This is the messages I recieve:

> E (74842) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
> E (74842) task_wdt:  - IDLE0 (CPU 0)
> E (74842) task_wdt: Tasks currently running:
> E (74842) task_wdt: CPU 0: BtA2dSinkT
> E (74842) task_wdt: CPU 1: IDLE1
> E (74842) task_wdt: Aborting.
> abort() was called at PC 0x40155620 on core 0
> 
> ELF file SHA256: 0000000000000000
> 
> Backtrace: 0x4008eddc:0x3ffbf330 0x4008f059:0x3ffbf350 0x40155620:0x3ffbf370 0x4008d6ad:0x3ffbf390 0x400d3235:0x3ffe31a0 0x400d22e1:0x3ffe31c0 0x400d2542:0x3ffe31e0 0x400d2569:0x3ffe3200 0x400d2633:0x3ffe3220 0x400d1375:0x3ffe3240 0x400d1a45:0x3ffe3260 0x400d1ebb:0x3ffe32a0 0x400d1ec9:0x3ffe32c0 0x40129ac7:0x3ffe32e0 0x40129af9:0x3ffe3310 0x40129c4a:0x3ffe3330 0x4009010e:0x3ffe3360
> 
> Rebooting...

My serial monitor error:

>  Error
>    0x40155620 task_wdt_isr
>    0x4008eddc invoke_abort
>    0x4008f059 abort
>    0x4008d6ad _xt_lowint1
>    0x400d3235 uartWriteBuf
>    0x400d22e1 HardwareSerial
>    0x400d2542 Print
>    0x400d2569 Print
>    (inlined by) Print
>    0x400d2633 Print
>    0x400d1375 set_stream_reader(unsigned char const*, unsigned int)
>    0x400d1a45 BluetoothA2DPSink
>    0x400d1ebb BluetoothA2DPSinkCallbacks
>    0x400d1ec9 audio_data_callback_2
>    0x40129ac7 btc_a2d_data_cb_to_app
>    
>    0x40129af9 btc_a2dp_sink_data_ready
>    0x40129c4a btc_a2dp_sink_task_handler
>    :355 (discriminator 1):::0x4009010e:vPortTaskWrapper
>    : ?? ??:0
>    : ?? ??:0
>    : ?? ??:0
   
```
#include <SoundData.h>
#include <config.h>
#include <BluetoothA2DPSource.h>
#include <BluetoothA2DPCommon.h>
#include "BluetoothA2DPSink.h"
BluetoothA2DPSink a2dp_sink;
int8_t IntStat = 0;
int8_t IntConnected = 0;
void setup() {
	Serial.begin(115200);//57600
	while (!Serial);
	//a2dp_sink.set_i2s_config(i2s_config);
	//virtual void set_stream_reader(void (*callBack)(const uint8_t*, uint32_t), bool i2s_output=true);
	a2dp_sink.set_stream_reader(set_stream_reader);
	////set_avrc_metadata_callback(void(*callback)(uint8_t, const uint8_t*))
	//a2dp_sink.set_avrc_metadata_callback(avrc_metadata_callback);
	a2dp_sink.start("MyMusic");
	Serial.println("end setup");
}
void set_stream_reader(const uint8_t* data, uint32_t length) {
	Serial.println("new package");
	size_t count = length / 2; // number of int6_t
	int16_t* values = (int16_t*)data;
	// process the values....
	// first left value is values[0] -  first right value is values[1]
	// last left value is values[count-2] -  last right value is values[count-1]
	//mono
	for (uint32_t i = 0; i < count; i = i + 2) {
		if (values[i] !=0) {
			//Serial.print("i: ");
			//Serial.println(i);

			Serial.println(values[i]);
		}
	}
	Serial.println("");
	Serial.println("");
}
void avrc_metadata_callback(uint8_t id, const uint8_t* text) {
	Serial.printf("==> AVRC metadata rsp: attribute id 0x%x, %s\n", id, text);
}
void loop() {
	delay(10);
	if (a2dp_sink.isConnected() != IntConnected) {
		IntConnected = a2dp_sink.isConnected();
		if (IntConnected == 0) {
			Serial.println("Not Connected");
		}
		if (IntConnected == 1) {
			Serial.println("Connected");
		}
	}
	if (a2dp_sink.get_audio_state() != IntStat) {
		IntStat = a2dp_sink.get_audio_state();
		if (IntStat == 0) {
			Serial.println("Stop");
		}
		if (IntStat == 2) {
			Serial.println("Play");
		}
	}
}
```

podaen, 2021-09-13:
I tried something else... I have connected the jack of the external dac to an analog input. What I see is that the readings still coming in when I stop the music. The music stops come out of the spreakers but the stat is still playing and the analogeread is still reading values.

I also tried to leave the streamreader and the analoge readings out of the program. I just checked the state of the audio and I have the same results. the stop comes a few seconds the music stops with playing. This is very slow for a mcu that works on ms. Is there a solution for that?

This is the code where I check the state.

```
	if (a2dp_sink.get_audio_state() != IntStat) {
		IntStat = a2dp_sink.get_audio_state();
		if (IntStat == 0) {
			Serial.println("Stop");
		}
		if (IntStat == 2) {
			Serial.println("Play");
		}
	}
```

podaen, 2021-09-13:
I think that I know what is hapening. On this one the streamreader has equal packages of 1278 bytes. The state stops when the stream reader has finished the packages. But there is still something strange.

There are still coming packages in when the music actually stops. See here:

> new package
> i: 0
> -34
> i: 2
> -60
> i: 4
> -9
> i: 6
> -10
> i: 8
> -48
> i: 10
> -39
> i: 12
> -28
> i: 14
> -9
> i: 16
> -20
> i: 18
> -80
> i: 20
> -42
> i: 22
> 43
> i: 24
> -1
> i: 26
> -84
> i: 28
> -64
> i: 30
> -8
> i: 32
> -30
> i: 34
> -87
> i: 36
> -82
> i: 38
> -34
> i: 40
> -33
> i: 42
> -83
> i: 44
> -123
> i: 46
> -76
> i: 48
> 10
> i: 50
> 11
> i: 52
> -77
> i: 54
> -106
> i: 56
> -63
> i: 58
> -68
> i: 60
> -85
> i: 62
> -27
> i: 64
> 6
> i: 66
> -57
> i: 68
> -105
> i: 70
> -63
> i: 72
> -22
> i: 74
> -44
> i: 76
> -61
> i: 78
> -27
> i: 82
> -9
> i: 84
> -17
> i: 86
> -10
> i: 88
> -7
> i: 90
> -9
> i: 92
> -7
> i: 94
> -1
> i: 98
> -5
> i: 100
> -10
> i: 102
> -6
> i: 104
> -2
> i: 106
> -5
> i: 108
> -6
> i: 110
> -2
> i: 116
> -1
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> new package
> 
> 
> Stop

podaen, 2021-09-13:
The problem of crashes has been solved by puting the serial monitor up to 2M baud.

podaen, 2021-09-14:
How can I activate the log file?

pschatzmann, 2021-09-16:
Tools -> Core Debug Level and set it e.g. to Verbose

podaen, 2021-09-16:
Thanks, here is my log. Can you find something weird?

> Opening port
> Port open
> VMDPR_[  4057][D][BluetoothA2DPSink.cpp:195] start(): start
> [  4057][I][BluetoothA2DPSink.cpp:200] start(): Device name will be set to 'MyMusic'
> [  4058][D][BluetoothA2DPSink.cpp:995] init_nvs(): init_nvs
> [  4064][D][BluetoothA2DPSink.cpp:1011] get_last_connection(): get_last_connection
> [  4072][D][BluetoothA2DPSink.cpp:310] init_bluetooth(): init_bluetooth
> 
> Brownout detector was triggered
> 
> ets Jun  8 2016 00:22:57
> 
> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:1
> load:0x3fff0030,len:1100
> ho 0 tail 12 room 4
> load:0x40078000,len:12308
> load:0x40080400,len:3076
> entry 0x400805ec
> [��mum����2-hal-cpu.c:211] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
> VMDPR_[  4057][D][BluetoothA2DPSink.cpp:195] start(): start
> [  4057][I][BluetoothA2DPSink.cpp:200] start(): Device name will be set to 'MyMusic'
> [  4058][D][BluetoothA2DPSink.cpp:995] init_nvs(): init_nvs
> [  4064][D][BluetoothA2DPSink.cpp:1011] get_last_connection(): get_last_connection
> [  4072][D][BluetoothA2DPSink.cpp:310] init_bluetooth(): init_bluetooth
> [  4167][I][BluetoothA2DPSink.cpp:315] init_bluetooth(): controller initialized
> [  4499][I][BluetoothA2DPSink.cpp:324] init_bluetooth(): bluedroid initialized
> [  4576][I][BluetoothA2DPSink.cpp:332] init_bluetooth(): bluedroid enabled
> [  4576][D][BluetoothA2DPSink.cpp:427] app_task_start_up(): app_task_start_up
> [  4578][D][BluetoothA2DPSink.cpp:1179] app_task_handler_2(): app_task_handler_2
> [  4585][D][BluetoothA2DPSink.cpp:1146] app_task_handler(): app_task_handler
> [  4592][D][BluetoothA2DPSink.cpp:400] app_task_handler(): app_task_handler
> [  4599][D][BluetoothA2DPSink.cpp:352] app_work_dispatch(): app_work_dispatch event 0x0, param len 0
> [  4608][D][BluetoothA2DPSink.cpp:384] app_send_msg(): app_send_msg
> [  4614][D][BluetoothA2DPSink.cpp:407] app_task_handler(): app_task_handler, sig 0x1, 0x0
> [  4622][W][BluetoothA2DPSink.cpp:410] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
> [  4631][D][BluetoothA2DPSink.cpp:375] app_work_dispatched(): app_work_dispatched
> [  4638][D][BluetoothA2DPSink.cpp:217] operator()(): av_hdl_stack_evt_2
> [  4644][D][BluetoothA2DPSink.cpp:801] av_hdl_stack_evt(): av_hdl_stack_evt evt 0
> [  4652][D][BluetoothA2DPSink.cpp:806] av_hdl_stack_evt(): av_hdl_stack_evt av_hdl_stack_evt BT_APP_EVT_STACK_UP
> [  4662][D][BluetoothA2DPSink.cpp:815] av_hdl_stack_evt(): AVRCP controller initialized!
> [  4663][D][BluetoothA2DPSink.cpp:1199] app_gap_callback_2(): app_gap_callback_2
> [  4670][E][BluetoothA2DPSink.cpp:831] av_hdl_stack_evt(): esp_avrc_tg_set_rn_evt_cap failed
> end setup[  4685][D][BluetoothA2DPSink.cpp:852] av_hdl_stack_evt(): connect_to_last_device
> [  4685][D][BluetoothA2DPSink.cpp:1170] app_gap_callback(): app_gap_callback
> [  4695][D][BluetoothA2DPSink.cpp:1065] connect_to_last_device(): connect_to_last_device
> 
> [  4707][D][BluetoothA2DPSink.cpp:857] av_hdl_stack_evt(): esp_bt_gap_set_scan_mode(ESP_BT_SCAN_MODE_CONNECTABLE_DISCOVERABLE)
> [  4707][I][BluetoothA2DPSink.cpp:493] app_gap_callback(): event: 10
> [  4737][D][BluetoothA2DPSink.cpp:1189] app_a2d_callback_2(): app_a2d_callback_2
> [  4737][D][BluetoothA2DPSink.cpp:1158] app_a2d_callback(): app_a2d_callback
> [  4743][D][BluetoothA2DPSink.cpp:874] app_a2d_callback(): app_a2d_callback
> [  4747][D][BluetoothA2DPSink.cpp:902] app_a2d_callback(): app_a2d_callback ESP_A2D_AUDIO_CFG_EVT
> [  4755][D][BluetoothA2DPSink.cpp:352] app_work_dispatch(): app_work_dispatch event 0x4, param len 16
> [  4764][D][BluetoothA2DPSink.cpp:384] app_send_msg(): app_send_msg
> [  4770][D][BluetoothA2DPSink.cpp:407] app_task_handler(): app_task_handler, sig 0x1, 0x4
> [  4778][W][BluetoothA2DPSink.cpp:410] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
> [  4787][D][BluetoothA2DPSink.cpp:375] app_work_dispatched(): app_work_dispatched
> [  4794][D][BluetoothA2DPSink.cpp:878] operator()(): av_hdl_a2d_evt_2
> [  4801][D][BluetoothA2DPSink.cpp:554] av_hdl_a2d_evt(): av_hdl_a2d_evt evt 4
> [  4807][I][BluetoothA2DPSink.cpp:681] av_hdl_a2d_evt(): A2DP PROF STATE: Init Compl
> 
> [  4817][D][BluetoothA2DPSink.cpp:1199] app_gap_callback_2(): app_gap_callback_2
> [  4824][D][BluetoothA2DPSink.cpp:1170] app_gap_callback(): app_gap_callback
> [  4829][I][BluetoothA2DPSink.cpp:493] app_gap_callback(): event: 10
> [  4835][D][BluetoothA2DPSink.cpp:1199] app_gap_callback_2(): app_gap_callback_2
[... 1310 lines cut ...]
> [ 21358][D][BluetoothA2DPSink.cpp:1189] app_a2d_callback_2(): app_a2d_callback_2
> [ 21358][D][BluetoothA2DPSink.cpp:1158] app_a2d_callback(): app_a2d_callback
> [ 21361][D][BluetoothA2DPSink.cpp:874] app_a2d_callback(): app_a2d_callback
> [ 21367][D][BluetoothA2DPSink.cpp:890] app_a2d_callback(): app_a2d_callback ESP_A2D_AUDIO_STATE_EVT
> Stop
> [ 21376][D][BluetoothA2DPSink.cpp:352] app_work_dispatch(): app_work_dispatch event 0x1, param len 16
> [ 21388][D][BluetoothA2DPSink.cpp:384] app_send_msg(): app_send_msg
> [ 21392][D][BluetoothA2DPSink.cpp:407] app_task_handler(): app_task_handler, sig 0x1, 0x1
> [ 21400][W][BluetoothA2DPSink.cpp:410] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[... cut ...]
> [ 24070][I][BluetoothA2DPSink.cpp:493] app_gap_callback(): event: 13
> [ 24093][D][BluetoothA2DPSink.cpp:1199] app_gap_callback_2(): app_gap_callback_2
> [ 24095][D][BluetoothA2DPSink.cpp:1170] app_gap_callback(): app_gap_callback
> [ 24097][I][BluetoothA2DPSink.cpp:493] app_gap_callback(): event: 13
> [ 27298][D][BluetoothA2DPSink.cpp:1199] app_gap_callback_2(): app_gap_callback_2
> [ 27298][D][BluetoothA2DPSink.cpp:1170] app_gap_callback(): app_gap_callback
> [ 27301][I][BluetoothA2DPSink.cpp:493] app_gap_callback(): event: 13
> 
>

closed 2021-10-16 by podaen