pschatzmann/ESP32-A2DP #88: Volume issue

Alfred586, 2021-09-11:
Thankyou Phil for your excellent work.

I found an issue in the volume control.
When i change the volume in the source side (smartphone), there's no change in the sink side.

Code: bt_music_receiver_simple
Library version: 1.4.0
ESP32 2.0.0 arduino core

Alfred586, 2021-09-12:
I don't know how to make a pull request, so i decided to write what i found here, sorry.
I found that the flag "is_volume_used" is not been activated in the function "void BluetoothA2DPSink::volume_set_by_controller(uint8_t volume)".
If i put the flag in these function, the issue is corrected succesfully.
I made this correction in the file "BluetoothA2DPSink.cpp".

pschatzmann, 2021-09-16:
Thanks for your input. I have committed the correction. 
Can you double check and let me know if it is ok now ?

Alfred586, 2021-09-16:
Yes, now works perfect.
Thanks.
I will close de issue.

closed 2021-09-16 by Alfred586