pschatzmann/ESP32-A2DP #91: Add Output to "simulated" DAC

DrNeuroSurg, 2021-09-15:
After a (long) look at https://github.com/earlephilhower/ESP8266Audio , I think it would possible and a great option to make output as Sigma-Delata modulation, isn't it ? 
Take look in the above source ("AudioOutputI2SNoDAC.h + .cpp").
So one could use this library to build a very cheap BT "speaker" with only one transistor an only one resistor !

pschatzmann, 2021-09-16:
What would be the advantages of this compared to directly using the built in "internal DAC" of the ESP32 ?

DrNeuroSurg, 2021-09-16:
IMHO if you feed the output over a simple low-pass (RC) and/or a transistor (for driving a speaker) this should give a much more better sound than the 8-bit DAC - isn‘t it ?

pschatzmann, 2021-09-16:
I honestly don't know and I guess this would be worth an experiment. Do you have the necessary components ?

DrNeuroSurg, 2021-09-16:
The components aren't the problem :-( 
tried the example from "AudioOutputI2SNoDAC" and it looks (so far) fine .. Now I'm on the algorithm itself ..

pschatzmann, 2021-09-20:
Did you manage to integrate the logic from AudioOutputI2SNoDAC or do you need some help ?

DrNeuroSurg, 2021-09-20:
Sorry - I think, it‘s not worth the effort:
So if you convert the 8bit (where you can have 256 different values - you know ?) and you make a DeltaSigma value of that, you‘ll need (at least) 128bit … to achieve the same resolution, for the price of noise ..

closed 2021-09-20 by DrNeuroSurg