pschatzmann/ESP32-A2DP #97: Cant connect to esp32 receiver from source after restarting receiver

benjamin32561, 2021-10-04:
`
#include "BluetoothA2DPSink.h"

BluetoothA2DPSink a2dp_sink;

void read_data(const uint8_t *data, uint32_t len){
}

void setup() {
  Serial.begin(115200);
  a2dp_sink.set_stream_reader(read_data);
}

void loop() {
  a2dp_sink.start("ESP32_SPEAKER_TEST",false);
  Serial.println("started");
  delay(60000);
  if(a2dp_sink.isConnected()) a2dp_sink.disconnect();
  a2dp_sink.end();
  Serial.println("turned off");
  delay(10000);
}
`
After starting the Bluetooth receiver I can't connect to it from my phone.
The library works perfectly if I'm not stopping and then starting the Bluetooth receiver but i need it for my project, any fixes?

pschatzmann, 2021-10-05:
I just committed a correction, but I am not really happy with the solution. The only method that I could call w/o loosing the possibility to call start again was esp_avrc_ct_deinit() - which does not really release any memory.
I am afraid that there are only the following options:
- a2dp_sink.end() or a2dp_sink.end(false) -> subsequent start() supported but no memory release
- a2dp_sink.end(true) -> memory release but no subsequent start() 
The best option to release the memory is to call ESP.restart(); which restarts the processor

benjamin32561, 2021-10-06:
Restarting the ESP fixed it, but it is kind of a sketchy solution, hope you will be able to fix this issue in the library, other than that the library is amazing, thanks a lot!

clarkbremer, 2021-11-22:
I'm Running into the same thing (using v1.6.0).  If I take my phone out of range, then back again, I can manually re-connect from the phone.  But if I turn off the BT on my phone, then back on again, the only way I can re-connect is to reboot the ESP32.

jokubasver, 2021-12-10:
> Using a different ESP32 board (that supports the latest espressif code), and this problem has gone away.

Are you saying that some boards don't support latest espressif code? Haven't heard of this.

I am experiencing the same issue, which is quite annoying, as I'm using this as a bluetooth aux device for my car. The ESP32 is turned on after unlocking the car, and since the car's body acts like a faraday cage, the range is limited outside of the car. Therefore, I always have to manually connect the phone to the A2DP sink, as the phone is always out of range when the ESP32 boots up.

If the phone is in range of the Bluetooth signal before the ESP32 boots up, the phone connects to the ESP32 without problems.

I tried using https://github.com/platformio/platform-espressif32#feature/arduino-upstream together with https://github.com/espressif/arduino-esp32#2.0.1-RC1 and the issue is still there.

clarkbremer, 2021-12-10:
Yes, some arduino board types do not include the latest espressif code.  Not sure if it's a hardware limitation, or just that the provider of the board library is behind.  In my case, using a board library from Heltec, which included the code for the integrated OLED display, did not have the latest espressif code.  I then bought a "generic" board (Hiletgo?) and used the standard board library, and I was able to use the latest features.  Perhaps coincidence, but my reconnect issue went away when I switched boards.

jokubasver, 2021-12-13:
If I modify  `void BluetoothA2DPCommon::clean_last_connection()` in BluetoothA2DPCommon.cpp to: 
```
void BluetoothA2DPCommon::clean_last_connection() {
    ESP_LOGD(BT_AV_TAG, "%s", __func__);
    //esp_bd_addr_t cleanBda = { 0 };
    //set_last_connection(cleanBda);
}
```
and in my sketch I add:
```
unsigned long startMillis;
unsigned long currentMillis;
const unsigned long period = 10000; // Time after which we reboot if we are disconnected for 10 seconds

void loop() {
  //Get the current time since boot
  currentMillis = millis();

  // Check if we have connection
  if (a2dp_sink.get_connection_state() == ESP_A2D_CONNECTION_STATE_DISCONNECTED)
  {
    if (currentMillis - startMillis >= period)  //test whether the period has elapsed
    {
      startMillis = currentMillis;
      Serial.println("Connection state: disconnected - Rebooting ESP32 to try again");
      delay(50);
      ESP.restart();
    }
    return;
  }
```

Then the auto-reconnect issue when the phone is out of BT range, goes away.
This gives a 10 second window to attempt a reconnection, and if it fails - the ESP reboots and tries again. The time window could probably be reduced, but 10 seconds works fine in my case.

Seems like not clearing the last connection together with a reboot let's the phone reconnect at any given time. 

Just doing that restart after 10 seconds on it's own did not fix the issue, neither did the removal of `clean_last_connection();`
I had to use both in order to fix the issue (albeit it's a hacky, dumb and bad solution, but hey... it works)

I thought that by removing `clean_last_connection();` it wouldn't let a different device to connect to the A2DP sink, but it still works fine (tested with two phones)

Unfortunately, `clean_last_connection();` is a protected function, so it would be great if this function could be disabled in the sketch, without modding the library.

pschatzmann, 2021-12-13:
Thanks for your feed-back. I had a quick look when the clean_last_connection(); is called.  The logic is that it tries to reconnect for AUTOCONNECT_TRY_NUM times and when it fails it cleans the last connection.
I think this makes sense - however I also noticed that the default for AUTOCONNECT_TRY_NUM is 1.

I would propose that you just add e.g.
#define AUTOCONNECT_TRY_NUM 10000

I guess this would prevent that you need to reboot.

By the way  clean_last_connection() is virtual - so you can just create your own subclass of BluetoothA2DPSink and overwrite this method with your own implementation - w/o the need to change any code

pschatzmann, 2021-12-13:
After a second thought: the define in the Sketch might not be picked up - and you would need to change it in the source to test. If this is working, I can change this into a set method that can be called in the sketch

jokubasver, 2021-12-13:
I undid my changes and tried your suggestion. Setting #define AUTOCONNECT_TRY_NUM to 10000 in BluetoothA2DPSink.h seems to fix the issue as well. No need to remove clean_last_connection(); and no need to use ESP.restart().

I will do some more testing to be sure, but so far it seems to work. Thank you!

pschatzmann, 2021-12-13:
Cool, I also plan to increase the default value (maybe to 10 or 20)
Would you have any suggestion ?

jokubasver, 2021-12-13:
I tried previously a value of 10, but I was putting #define AUTOCONNECT_TRY_NUM in my sketch and it didn't fix the issue. I will try again with 10000 in the sketch, because I'm not sure if the value was too low, or if the define has to be set in the source of the library in order for it to work

jokubasver, 2021-12-13:
Okay, #define AUTOCONNECT_TRY_NUM 10000 in the sketch doesn't work. I will now try some sensible values in the source and I will let you know what default you could use.

jokubasver, 2021-12-13:
10 retries seem to work fine. I guess you could set the default to 20 to be on the safe side. But now I'm wondering - what's the advantage of having a limit on auto-connect retry count?

pschatzmann, 2021-12-13:
I think the idea was that the system tries to reconnect to the last used system. 
If it fails it stops  and you should be able to connect from e.g. another device

pschatzmann, 2021-12-13:
I moved the setting to config.h and increased it to 10
In addition I added the new method set_try_reconnect_max_count()

closed 2021-12-14 by pschatzmann

pschatzmann, 2021-12-14:
Sorry, the functionality has changed again:
https://github.com/pschatzmann/ESP32-A2DP/wiki/Auto-Reconnect

esteves67, 2021-12-15:
> I'm Running into the same thing (using v1.6.0). If I take my phone out of range, then back again, I can manually re-connect from the phone. But if I turn off the BT on my phone, then back on again, the only way I can re-connect is to reboot the ESP32.

Im having the same issue, after some disconnects (taking the smartphone out of range) the esp32 seems to crash (like if it its out of memory, i dunno), and so i can't connect anymore to the esp32 bluetooth, the only way to get it back to work is resetting the esp32

esteves67, 2021-12-15:
> I'm Running into the same thing (using v1.6.0). If I take my phone out of range, then back again, I can manually re-connect from the phone. But if I turn off the BT on my phone, then back on again, the only way I can re-connect is to reboot the ESP32.

@clarkbremer 

Im having the same issue, after some disconnects (taking the smartphone out of range) the esp32 seems to crash (like if it its out of memory, i dunno), and so i can't connect anymore to the esp32 bluetooth, the only way to get it back to work is resetting the esp32.

I will try a more recent esp32 board like you did, to see if the problem goes away.

esteves67, 2021-12-15:
@jokubasver Thank you so much for the sketch you provided, it worked like a charm for me using both the `ESP.restart();` and the `clean_last_connection();`, along with the sketch logic you wrote, thank you so much!

esteves67, 2021-12-15:
Just to clarify, my use case is that i want the esp32 to only connect to a specific device, and to accept frequently disconnections/connections, but my esp32 was crashing by many connections, and the ESP.restart() along with commenting the clean_last_connection(), worked very well for me :heart: @jokubasver you're my hero :+1: