pschatzmann/ESP32-A2DP #100: Redusc

podaen, 2021-10-16:


pschatzmann, 2021-10-16:
It is not quite clear to me to which memory you are referring to.
If you mean program storage space I suggest to change the Partition Scheme to Huge App...

With this I am getting:
- Sketch uses 969746 bytes (30%) of program storage space.
-Global variables use 30308 bytes (9%) of dynamic memory

podaen, 2022-09-10:
> It is not quite clear to me to which memory you are referring to. If you mean program storage space I suggest to change the Partition Scheme to Huge App...
> 
> With this I am getting:
> 
> * Sketch uses 969746 bytes (30%) of program storage space.
> * Global variables use 30308 bytes (9%) of dynamic memory

If I take this simple shetch

```
#include <BluetoothA2DPSink.h>

#define JackPin 34
#define DC_OFFSET  0  // DC offset in mic signal - if unusure, leave 0
#define NOISE     10  // Noise/hum/interference in mic signal
int lvlLeft = 10; // Current "dampened" audio level

BluetoothA2DPSink a2dp_sink;
//bool is_active = true;

int8_t IntStat = 0;
int8_t IntConnected = 0;
int16_t IntValues[40000];
int16_t IntBufferCounter = 0;

void setup() {
	Serial.begin(115200);// 115200);//57600//921600//2000000
	while (!Serial);
	a2dp_sink.start("MyMusic");
	Serial.println("end setup");
}

void loop() {
        Delay(50);
	PrintFreeGap();
}

void PrintFreeGap() {
	Serial.println("");
	Serial.print("xPortGetFreeHeapSize: ");
	Serial.println(xPortGetFreeHeapSize());//290116//Since ESP32 has 328 KiB of data RAM
	Serial.print("getFreeHeap: ");
	Serial.println(ESP.getFreeHeap());//290116//Since ESP32 has 328 KiB of data RAM
	Serial.print("getFreePsram: ");
	Serial.println(ESP.getFreePsram());//290116//Since ESP32 has 328 KiB of data RAM
	Serial.println("");
}
```

Then it uses with me...
> xPortGetFreeHeapSize: 4306255
> getFreeHeap: 151920
> getFreePsram: 4173267

An almost empty shetch gives me getFreeHeap: 317556
So I am thinking that everything from A2DP is using 317556 -151920 =  165636 of DRAM...

I using now A2DP now for a long time and my shetch (not the example). And I have additional things I have added over time, but I am getting short in space.  Is there something that can be done get more space?

pschatzmann, 2022-09-10:
The only thing I can think of is to use an ESP32 version with SPI PSRAM!
https://www.pschatzmann.ch/home/2022/05/30/esp32-and-psram/

podaen, 2022-09-10:
I have such a chip and it was turned on. If I turn it off I have a Free Heap of 138568 and compare it with the option on, I won only 13352 in the simple sketch...

pschatzmann, 2022-09-10:
Did you also monitor the PSRam ? 
What do you mean with you switched the option on ?

podaen, 2022-09-10:
What I mean is... You can see the uses of PSRAM too.

PSRAM on
xPortGetFreeHeapSize: 4306255
getFreeHeap: 151920
getFreePsram: 4173267

PSRAM off
xPortGetFreeHeapSize: 114200
getFreeHeap: 138568
getFreePsram: 0

Ps: I am using version 1.6.0

pschatzmann, 2022-09-10:
So you have 4MB free PS RAM to use. Just make sure that when you allocate memory you use PSRAM and not the heap

podaen, 2022-09-10:
I will do, but I don't have any large variables in my sketch, so I don't suspect I will gain a lot. Bluetooth in general takes a lot, I know, but I was hoping I could get more free hap. The other half is used by other libraries I have WiFI, MQTT,... (30000 + 48000 +70000). So those 3 need 148000 Allocated memory from the DRAM and there is not so much I can do with PSRAM.
I'm think I am stuck on this path. Are there ESP32's with more DRAM than I already have? If I understand it right there is a 4,8 and 16MB version, but that is SPIFLASH I think. Don't know what the DRAM is.

pschatzmann, 2022-09-10:
You can use  e.g. 

    const int limit = 500;
    heap_caps_malloc_extmem_enable(limit);

This will allocate all memory > 500 to psram. 
Just make sure that you allocate your objects with new after you executed this call.
I never tried if A2DP still works with this...

podaen, 2022-09-10:
It gave me an other 20000 of freeheap, I am glad with it. I'm going to see what I can do more in my sketch. Thanks for the help!

closed 2022-09-10 by podaen